HiddenRamblings/TagMo #1: Request: Port app to WP

godofwrath, 2016-09-21:
Could the app be ported to WP/UWP?, the alternatives for tagmo on a windows device aren't very versatile, that way it could possibly be compiled for use on a desktop pc or a windows 10 mobile.

HiddenRambler, 2017-01-10:
Unfortunately I don't have WP. Also i'm not quite sure WP would allow the necessary access to NFC to do this.

godofwrath, 2017-02-13:
There's an app that exists but it's as robust as tagmo for wp, it's called "amiigo", the base functions are all the same and it appears to work just fine.

Although use with an nfc reader on windows with the same features as tagmo would be fantastic, I guess it's not really worth making since the target audience for such a tool is quite small.

Thanks anyway!

humphries40, 2017-03-17:
If you're satisfied @godofwrath you should close the issue if you can. Helps keep things cleaner.

closed 2017-03-17 by godofwrath