HiddenRamblings/TagMo #2: Scanning causes force close on tablets(?)

AndrewIsOffline, 2016-10-08:
I have an ASUS TF103CE with NFC and even though the NFC feature works and is able to read Amiibos, whenever I press the "scan" button on this app, it force closes. Could it be a problem with tablets, or am I doing something wrong?

HiddenRambler, 2017-01-10:
Can you try this version: https://github.com/HiddenRambler/TagMo/releases/download/2.3.4/TagMo.2.3.4Beta.apk

AndrewIsOffline, 2017-01-22:
Nope, same problem.

AndrewIsOffline, 2017-01-22:
It can read saved .bin files and re-save them, but doing anything else crashes the app.

AndrewIsOffline, 2017-01-22:
I tried debugging the program on the tablet and at startup this gets thrown:

LLDBDriverException: failed to get reply to handshake packet

However, this doesn't (immediately) crash the app.

AndrewIsOffline, 2017-01-22:
Alright, I got the stack trace for whatever is causing this.
`E/AndroidRuntime( 3896): FATAL EXCEPTION: main
E/AndroidRuntime( 3896): Process: com.hiddenramblings.tagmo, PID: 3896
E/AndroidRuntime( 3896): java.lang.RuntimeException: Unable to start activity ComponentInfo{com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.NfcActivity_}: android.view.InflateException: Binary XML
 file line #31: Error inflating class ImageView
E/AndroidRuntime( 3896):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2298)
E/AndroidRuntime( 3896):        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2360)
E/AndroidRuntime( 3896):        at android.app.ActivityThread.access$800(ActivityThread.java:144)
E/AndroidRuntime( 3896):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1278)
E/AndroidRuntime( 3896):        at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 3896):        at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime( 3896):        at android.app.ActivityThread.main(ActivityThread.java:5221)
E/AndroidRuntime( 3896):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 3896):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime( 3896):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
E/AndroidRuntime( 3896):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)
E/AndroidRuntime( 3896): Caused by: android.view.InflateException: Binary XML file line #31: Error inflating class ImageView
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:763)
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.rInflate(LayoutInflater.java:806)
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.inflate(LayoutInflater.java:504)
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.inflate(LayoutInflater.java:414)
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.inflate(LayoutInflater.java:365)
E/AndroidRuntime( 3896):        at android.support.v7.app.AppCompatDelegateImplV7.setContentView(AppCompatDelegateImplV7.java:256)
E/AndroidRuntime( 3896):        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:109)
E/AndroidRuntime( 3896):        at com.hiddenramblings.tagmo.NfcActivity_.setContentView(NfcActivity_.java:51)
E/AndroidRuntime( 3896):        at com.hiddenramblings.tagmo.NfcActivity_.onCreate(NfcActivity_.java:42)
E/AndroidRuntime( 3896):        at android.app.Activity.performCreate(Activity.java:5933)
E/AndroidRuntime( 3896):        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1105)
E/AndroidRuntime( 3896):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2251)
E/AndroidRuntime( 3896):        ... 10 more
E/AndroidRuntime( 3896): Caused by: android.content.res.Resources$NotFoundException: File res/drawable-anydpi-v21/nfc_circle.xml from drawable resource ID #0x7f020047
E/AndroidRuntime( 3896):        at android.content.res.Resources.loadDrawableForCookie(Resources.java:2428)
E/AndroidRuntime( 3896):        at android.content.res.Resources.loadDrawable(Resources.java:2330)
E/AndroidRuntime( 3896):        at android.content.res.TypedArray.getDrawable(TypedArray.java:749)
E/AndroidRuntime( 3896):        at android.widget.ImageView.<init>(ImageView.java:146)
E/AndroidRuntime( 3896):        at android.widget.ImageView.<init>(ImageView.java:135)
E/AndroidRuntime( 3896):        at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:58)
E/AndroidRuntime( 3896):        at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:54)
E/AndroidRuntime( 3896):        at android.support.v7.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:95)
E/AndroidRuntime( 3896):        at android.support.v7.app.AppCompatDelegateImplV7.createView(AppCompatDelegateImplV7.java:938)
E/AndroidRuntime( 3896):        at android.support.v7.app.AppCompatDelegateImplV7.onCreateView(AppCompatDelegateImplV7.java:992)
E/AndroidRuntime( 3896):        at android.support.v4.view.LayoutInflaterCompatHC$FactoryWrapperHC.onCreateView(LayoutInflaterCompatHC.java:44)
E/AndroidRuntime( 3896):        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:725)
E/AndroidRuntime( 3896):        ... 21 more
E/AndroidRuntime( 3896): Caused by: java.lang.NumberFormatException: Invalid float: "1.1.9"
E/AndroidRuntime( 3896):        at java.lang.StringToReal.invalidReal(StringToReal.java:63)
E/AndroidRuntime( 3896):        at java.lang.StringToReal.initialParse(StringToReal.java:160)
E/AndroidRuntime( 3896):        at java.lang.StringToReal.parseFloat(StringToReal.java:323)
E/AndroidRuntime( 3896):        at java.lang.Float.parseFloat(Float.java:306)
E/AndroidRuntime( 3896):        at android.util.PathParser.getFloats(PathParser.java:178)
E/AndroidRuntime( 3896):        at android.util.PathParser.createNodesFromPathData(PathParser.java:59)
E/AndroidRuntime( 3896):        at android.graphics.drawable.VectorDrawable$VFullPath.updateStateFromTypedArray(VectorDrawable.java:1439)
E/AndroidRuntime( 3896):        at android.graphics.drawable.VectorDrawable$VFullPath.inflate(VectorDrawable.java:1421)
E/AndroidRuntime( 3896):        at android.graphics.drawable.VectorDrawable.inflateInternal(VectorDrawable.java:542)
E/AndroidRuntime( 3896):        at android.graphics.drawable.VectorDrawable.inflate(VectorDrawable.java:458)
E/AndroidRuntime( 3896):        at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:1095)
E/AndroidRuntime( 3896):        at android.graphics.drawable.Drawable.createFromXml(Draw
`

AndrewIsOffline, 2017-01-22:
I managed to track down the issue: https://code.google.com/p/android/issues/detail?id=202670

AndrewIsOffline, 2017-01-22:
Okay, got it! As it is, the current vector paths for some of the images are broken. Something with how they used to be generated was wrong and they need to be re-converted. I managed to get the app to scan NFCs by replacing NFC_circle.xml and NFC_bar.xml with some generic images (and it looks terrible), but for some reason SSB Data and View Hex still aren't working.

AndrewIsOffline, 2017-01-23:
Here's the stack trace for View Hex:
`E/AndroidRuntime( 8501): FATAL EXCEPTION: main
E/AndroidRuntime( 8501): Process: com.hiddenramblings.tagmo, PID: 8501
E/AndroidRuntime( 8501): java.lang.UnsatisfiedLinkError: com.android.tools.fd.runtime.IncrementalClassLoader$DelegateClassLoader[DexPathList[[dex file "/data/data/com.hiddenramblings.tagmo/files/insta
nt-run/dex/slice-support-annotations-23.1.1_915e977317a7b84fa14b4c07a73c866700b56539-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_9-classes.dex", dex
file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_8-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_7-classes.dex", dex file "
/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_6-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_5-classes.dex", dex file "/data/
data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_4-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_3-classes.dex", dex file "/data/data/c
om.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_2-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-slice_1-classes.dex", dex file "/data/data/com.hid
denramblings.tagmo/files/instant-run/dex/slice-slice_0-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-internal_impl-23.1.1_22c393fd68111a6fb5a7ec585a73ed2f86a
c0c07-classes.dex", dex file "/data/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-com.android.support-support-v4-23.1.1_85645c8c01be33e261e58f6700be07c7a7c745a5-classes.dex", dex file "/d
ata/data/com.hiddenramblings.tagmo/files/instant-run/dex/slice-com.android.support-appcompat-v7-23.1.1_f85d7fcf67cfbc62c60252ca804b7f4658498c8f-classes.dex", dex file "/data/data/com.hiddenramblings.t
agmo/files/instant-run/dex/slice-androidannotations-api-4.0.0_8dbc358617f6c920ab3d27acde08a8cd6b3481e5-classes.dex"],nativeLibraryDirectories=[/vendor/lib, /system/lib, /vendor/lib, /system/lib]]] cou
ldn't find "libamiitool.so"
E/AndroidRuntime( 8501):        at java.lang.Runtime.loadLibrary(Runtime.java:366)
E/AndroidRuntime( 8501):        at java.lang.System.loadLibrary(System.java:989)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.AmiiTool.<clinit>(AmiiTool.java:8)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.TagUtil.decrypt(TagUtil.java:93)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.EditorSSB.afterViews(EditorSSB.java:82)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.EditorSSB_.onViewChanged(EditorSSB_.java:96)
E/AndroidRuntime( 8501):        at org.androidannotations.api.view.OnViewChangedNotifier.notifyViewChanged(OnViewChangedNotifier.java:41)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.EditorSSB_.setContentView(EditorSSB_.java:52)
E/AndroidRuntime( 8501):        at com.hiddenramblings.tagmo.EditorSSB_.onCreate(EditorSSB_.java:42)
E/AndroidRuntime( 8501):        at android.app.Activity.performCreate(Activity.java:5933)
E/AndroidRuntime( 8501):        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1105)
E/AndroidRuntime( 8501):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2251)
E/AndroidRuntime( 8501):        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2360)
E/AndroidRuntime( 8501):        at android.app.ActivityThread.access$800(ActivityThread.java:144)
E/AndroidRuntime( 8501):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1278)
E/AndroidRuntime( 8501):        at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 8501):        at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime( 8501):        at android.app.ActivityThread.main(ActivityThread.java:5221)
E/AndroidRuntime( 8501):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 8501):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime( 8501):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
E/AndroidRuntime( 8501):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)
`
It seems to be looking for a "libamiitool.so"

azsde, 2017-03-09:
Issue fixed on my fork, look into it.

HiddenRambler, 2017-08-03:
I believe this is fixed in the latest version.

closed 2017-08-03 by HiddenRambler