HiddenRamblings/TagMo #3: Any News on NTAG216 compatibility?

m0nm0npalam0n, 2016-11-08:
Any news regarding this? I am unable to find a retailer of NTAG215 in my country :(

FinalDoom, 2017-01-05:
Can you not get them from aliexpress? They're more expensive at any in-country retailer for me. Shipping from China takes forever, but they work (just got mine, these: https://www.aliexpress.com/item/ISO14443A-NFC-Card-Passive-Contactless-ISO-RFID-Smart-Tag-1k-NTAG215-Chip-White-Card-200pcs-lot/32748565988.html)

gabrielklein, 2017-02-11:
Shouldn't you ask Nintendo instead? :)

humphries40, 2017-03-17:
I believe that Amiibo can only be NTAG215 so this issue isn't really relevant to TagMo and could/should probably be closed. @m0nm0npalam0n

meflakcannon, 2017-04-03:
NFC tags can actually have multiple partitions. Make a separate (second) partition on the NTAG216 that is exactly 348 bytes, and fill it with dummy data (FFs). This will then allow the writing via tagmo as if the card is an ntag215. 

IMO it would be easier to incorporate this feature into tagmo than doing what I'm currently doing. Especially since 216 cards and stickers are readily available and there seems to be a massive shortage of 215 tags without a multi week lead time shipping out of china.

jedahan, 2017-04-04:
I heard since NTAG21x has a partitioning scheme, if you partition a NTAG216 the same sizes as the 215 it will look like a 215. I have not been able to confirm this.

meflakcannon, 2017-04-04:
NTAG216 Partitioning as you call it is really just  NDEF Records. (It's critical to understand that). Writing out 348 bytes as FFs in a new NDEF record means you are left with a single NDEF of ideally an NTAG215 size. There is a thread over on gbatemp discussing this but there seems to be some misunderstanding as to whats actually happening. 

Writable space in ntag216 is 888bytes, ntag215 is 504. Even with amiibo dumps being 540 bytes. That means the space taken up with a an NDEF record filled with FFs should have been 384 bytes to perfectly emulate the 504 bytes of writable space we currently utilize in ntag215.

Instead a user is writing 348 bytes and leaving 540 bytes as writable space. This may be working (I haven't seen confirmation myself), but it means you are not so much emulating a NTAG215 as writing in a full 540byte amibo dump into an NTAG216 with all extra space filled up. If the amiibo reader is serial in nature and reads data sees the wrong tag but keeps reading then sees the right tag data it may work, but there are too many unknowns and no independent confirmation for such a feature to be added at this time.

HiddenRambler, 2017-08-03:
From everything we've see I don't think NTAG216 can be realisticly be supported. Closing this issue for now.

closed 2017-08-03 by HiddenRambler