HiddenRamblings/TagMo #5: Crash on Android 7

jimparis, 2017-01-10:
TagMo 2.3.3 just crashes on my phone (Nexus 5X, Android 7.1.1) when I try to write a tag or view hex.  However, the patched APK [described in this post](https://gbatemp.net/threads/release-tagmo-tool-for-creating-editing-amiibos.431670/page-3#post-6530204) worked great.  I don't have any further insight, but figured it was worth reporting here.

HiddenRambler, 2017-01-10:
Thank for the report. Can you try https://github.com/HiddenRambler/TagMo/releases/download/2.3.4/TagMo.2.3.4Beta.apk

jimparis, 2017-01-11:
Unfortunately no, seems no different than 2.3.3.  Still crashes on "Edit SSB Data", "View Hex", and "Write Tag (Auto)".

geemaan, 2017-01-13:
It still doesn't work on my nexus 5x with 7.1.1. Writing works with the version fixed [here](https://gbatemp.net/threads/release-tagmo-tool-for-creating-editing-amiibos.431670/page-3#post-6530204)

KevinCathcart, 2017-01-15:
@HiddenRambler:
The crash is caused by the vector graphics.

The vector graphics (or mote specifically, the nfc circle graphic) you are using contain technically invalid SVG paths. The problem is numbers like `1.1.9`. Older versions of android, web browsers, and svg editors all treat that as `1.1 0.9`, while newer versions of android freak out because `1.1.9` is not a valid decimal number, and thus the vector graphic is invalid. 

To fix this, edit "nfc_circle.xml" and find all occurrences of number with two decimal places and add a space before the second decimal point (`1.1.9` -> `1.1 .9`). That should fix the crash.

There might potentially be more required to get NFC to work though. The posted APK that users claim works for them also added three intent filters to the nfc activity. The actions are `android.nfc.action.TAG_DISCOVERED`, `android.nfc.action.NDEF_DISCOVERED`, and `android.nfc.action.TECH_DISCOVERED`; all three with a category of `android.intent.category.DEFAULT`. I do not have android 7.1 so I cannot if any or all of those is needed.

Hope this helps.

LinkofHyrule89, 2017-01-18:
I was trying to read ssb data and hex edit my wolf link amiibo on Android 7.1.1 specially Lineage OS 14.1 for Bacon and mine crashes as well.

LinkofHyrule89, 2017-01-26:
@KevinCathcart can you make a build with those changes I will test it.

LinkofHyrule89, 2017-01-26:
https://gbatemp.net/threads/release-tagmo-tool-for-creating-editing-amiibos.431670/page-4#post-7045437 sorry forgot the link.

l2775276, 2017-02-16:
I have the same issue.  Crashes on 2.3.3 and 2.3.4b.  Not limited to Android 7.  I can confirm the issue is on Android 5.0.1 and Android 6.0.1.  It crashes viewing hex as well as edit ssb data.  I don't know if I can write to a tag as I do not have any.

nathantschultz, 2017-02-22:
I'm seeing the same problem on android 5

bonovox400, 2017-02-22:
Is it normal after installing and running for the first time (2.3.4 beta) that it says "Keys missing - load keys from menu" and the top left shows "No Unfixed Key!" "No Locked key!" warnings?

Never used something like this before.  Have an unused Wolf Link Amiibo I wanted to test reading and possible heart tag flashing. (if someone has a tag/method in doing this)  I don't even care if it's less than 20 hearts. Just want something higher than the default 3 it starts with!   
 Phone is on Android 5.0.1.

damccull, 2017-03-02:
I tried the Google Play Store version and the beta linked above on Google Pixel running 7.1. Crash on HEX and SSB.

To be fair, I'm ridiculously new to amiibo and especially amiibo editing. I don't actually HAVE an amiibo yet in real life...it's on the way from Amazon as we speak. Wolf Link. Cause Switch...yeah...

Anyhow. Let me know if I can help in any way to get this fixed. Anyone know of a good tutorial or reference explaining how Amiibo files are structured and how to edit them? Is there a reference containing common amiibo fields anywhere? I understand they're encrypted with a retail key, but that's about as far as I've got.

Thanks,
David

azsde, 2017-03-09:
Try my fork, it will still crash with the SSB editor when loading a .bin file, but the SSB editor will work just fine when scanning a real amiibo.

JacobDB, 2017-03-18:
Any progress fixing this? I don't have access to an Android device pre 7.0, so I'm kind of stuck 🙁

ubergeek77, 2017-03-18:
It's crashing for me too. I just got some brand new NTAG215's, but they crash when I try writing them using AUTO. This issue doesn't seem to be exclusive to Android 7.0, either. This crash happens on my LG V20 on Android 7.0, and my Galaxy Note 4 on Android 6.0. I'm not sure what else to do about it. I can dump my own real Amiibo just fine, but I'm completely unable to copy them.

Can anyone help?

ryvaeus, 2017-03-22:
@ubergeek77 @JacobDB  The crashing is still present in the latest version 2.3.3 and the 2.3.4 beta, however either rolling back to 2.3.1 or using the modified APK listed in previous comments will work.

cifer1337, 2017-04-11:
On my Galaxy S7 edge with Android 7.0 the app crashes too also with the version 2.3.4 beta
Here is the stack trace
`E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
                                                 Process: com.hiddenramblings.tagmo, PID: 8552
                                                 java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.hiddenramblings.tagmo-1/base.apk"],nativeLibraryDirectories=[/data/app/com.hiddenramblings.tagmo-1/lib/arm64, /system/lib64, /vendor/lib64]]] couldn't find "libamiitool.so"
                                                     at java.lang.Runtime.loadLibrary0(Runtime.java:972)
                                                     at java.lang.System.loadLibrary(System.java:1530)
                                                     at com.hiddenramblings.tagmo.AmiiTool.<clinit>(AmiiTool.java:8)
                                                     at com.hiddenramblings.tagmo.TagUtil.decrypt(TagUtil.java:93)
                                                     at com.hiddenramblings.tagmo.TagUtil.patchUid(TagUtil.java:124)
                                                     at com.hiddenramblings.tagmo.TagWriter.adjustTag(TagWriter.java:81)
                                                     at com.hiddenramblings.tagmo.TagWriter.writeToTagAuto(TagWriter.java:41)
                                                     at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:176)
                                                     at com.hiddenramblings.tagmo.NfcActivity_.access$501(NfcActivity_.java:30)
                                                     at com.hiddenramblings.tagmo.NfcActivity_$6.execute(NfcActivity_.java:155)
                                                     at org.androidannotations.api.BackgroundExecutor$Task.run(BackgroundExecutor.java:405)
                                                     at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
                                                     at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                     at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:272)
                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
                                                     at java.lang.Thread.run(Thread.java:762)`

HiddenRambler, 2017-08-03:
Should be fixed in the latest version.

closed 2017-08-03 by HiddenRambler