HiddenRamblings/TagMo #6: Ability to edit Wolf Link hearts and stats?

LinkofHyrule89, 2017-01-18:
So I don't know exactly how this works since I don't own the twilight princess HD game but I bought a Wolf Link Amiibo. Apparently, when you use it with the Wii U version of tp you can level him up and he gets hearts and when the new Zelda comes out you can use him in it with his hearts. I don't own a Wii U so it would be nice if he app could edit the Amiibo so at hearts and such. Anyways thanks for the hard work I look forward to using this app.

possi, 2017-01-29:
+1
I've found the offset I think, so I'm looking to make a PR for that.

LinkofHyrule89, 2017-01-29:
This would be great if I can use it to edit my actual Amiibo and not just a
tag.

possi, 2017-01-29:
You can, that's the reason I build it: https://github.com/possi/TagMo/releases
But looks like you need to have saved at least one visit to the shadow cave on the amiibo before.

LinkofHyrule89, 2017-01-29:
I'll have to find someone with the game I guess.

fokusfocus, 2017-03-06:
Hi,

So I tried this with both NFC sticker and the actual Wolf Link amiibo (already visited shadow cave once) but I'm getting the following error when trying to Write Tag (Auto)

"Error while writing main data (stage 1) java.io.IOException: Transceive failed"

I tried it both with Nexus 6P and Pixel

Any idea?

moookayla, 2017-03-07:
I'm having the same error on a note two as @fokusfocus

sigboe, 2017-03-08:
@LinkofHyrule89 You can actually do this, find a dumped amiibo that has all the hearts, then load up that bin file in TagMo, then check the allow restore to different tag, then restore tag button. I tested it already.

LinkofHyrule89, 2017-03-08:
Can you direct me to a bin file? Did you do this on an actual Amiibo not a regular tag?

sigboe, 2017-03-08:
@LinkofHyrule89 I can't supply it since its copyrighted material. But I can say that was posted on the typical NFC sharing site that has the word Bank in the name, in the cheats section under the appropriate category for said amiibo file.

I did it on an ntag

LinkofHyrule89, 2017-03-08:
Well sadly that isn't going to work because as far as I've tried wth my tests if you flash the bins from that site they just corrupt the Amiibo and you have to reset it. If anyone else has treated it on an actual Amiibo feel free to correct me.

sigboe, 2017-03-08:
@LinkofHyrule89 What is the technical difference of using an sticker tag or a real amiibo in this regard?

LinkofHyrule89, 2017-03-08:
I think that the tags you write the entire Amiibo and the actual Amiibo you only want to write save data. I know Amiibo have a write locked area but I don't know much else about them other than when I tried to write bin files with the app a while back my Amiibo said it was corrupt and had to reset.

sigboe, 2017-03-08:
@LinkofHyrule89 I tried to write a blank wolf link first. Actually for the sake of transferring my Mii from the Wii U. Then I realized that it was almost useless to have a 3 heart wolf, and you could level it up. So I got the cheat save file bin and just updated the rewritable part. If I check the owner of the amiibo on my switch now it says not me anymore, it says the guy that uploaded the save file.

Remember I checked the box in tagmo that is called `Allow restore to different tag`

LinkofHyrule89, 2017-03-08:
I'll try again but as far as I've seen it won't let me change the save data because I never used the Amiibo on a Wii U because I don't own a Wii U.

LinkofHyrule89, 2017-03-08:
What are the exact steps you are doing to write the save data to the Amiibo? I have the bin marked 20 hearts I clicked allow restore different tags and load the tag. I click restore tag and it says done. I tap it on my switch and it tells me it's corrupt and has to fix it. After that it's still Wolf Link with 3 hearts.

sigboe, 2017-03-08:
Maybe the save you have is corrupt? The filename i used was `Wolf Link (20 Hearts, Cave of Shadows Beaten, B40).bin` It's MD5 sum is 3869383cdb414b92d00d9e51dbe8878f

LinkofHyrule89, 2017-03-08:
The one I'm using is the exact same file. This is a brand new Amiibo so the only thing on it is my name registration and Mii. I only have a Switch so I've never used it on a Wii U or with tphd. Maybe it's my key files what are the md5 of those?

sigboe, 2017-03-08:
Its over a year since I installed TagMo, but I have a file called secret key which I assume is the secret key, with md5 of `0ad86557c7ba9e75c79a7b43bb466333`

And I have a file called the info key which I assume is the public key, with MD5 of `2551afc7c8813008819836e9b619f7ed`

Technically speaking Sony lost in a court saying you cant copyright or trademark a key, but its maybe for the best we only share the md5 sum of those keys since I don't know what HiddenRambler or Github think about sharing the keys :P

LinkofHyrule89, 2017-03-08:
It's the same file. I'm thinking it's still that the Amiibo isn't fully initialized since I don't have access to Zelda tphd. Hopefully in the future they can make a way to initialize it in the app as well.

sigboe, 2017-03-08:
@LinkofHyrule89 I really want to test this hypothesis. Its easier if we could speak by email, facebook messenger, discord or slack? I have some ideas to test it.

LinkofHyrule89, 2017-03-09:
I'm on the BOTW Discord server. https://discord.gg/JMSwaM4 Link_of_Hyrule on there.

Kidel, 2017-03-14:
you can find the fully upgraded link amiibo dump online anyway.

Kidel, 2017-03-14:
Anyway, it has been made: https://github.com/Kidel/TagMo/commits/master

LinkofHyrule89, 2017-03-14:
If there is a way to use the dumps on an uninitialized Amiibo a lot of us would love to know.

Kidel, 2017-03-14:
I don't have any amiibo myself, so I can't test it :|

Kidel, 2017-03-14:
I don't have any Wolf Link amiibo myself, so I can't test it :|
However I don't think it's currently possible, since the application only writes the entire bin, and it would give an error on an amiibo (as on any already written NFC). But if you can do it, try and let us know. I'm not very expert about nfc, maybe it will only write the additional data and give no error. 

If by "unitialized amiibo" you mean an empty amiibo (do they exist?) then it should be as writing on an empty nfc215. Again, I'm no expert.

LinkofHyrule89, 2017-03-14:
No I mean an Amiibo that has not been used in Zelda TPHD. So there isn't a save created on the Amiibo.

Kidel, 2017-03-14:
the first half of the answer then

johnsmclay, 2017-03-16:
https://nfc-bank.com/bins.php?do=download&downloadid=1034

That one has all 20 hearts.

Batlh, 2017-03-19:
@LinkofHyrule89 tested it with a fresh buyed wolf link amiibo who has never be in TPHD because  i didn't have a wiiu ^^
so i tried to get a maxed out bin file for wolf link from @johnsmclay link to get restored to the amiibo via tagmo but the only result is you get an error on the console.
so i flashed a backup on the amiibo from the original state as it comes out of the box
and there no error on the console so i would say the amiibo must at least used once in TPHD to get the right function :/

jbienz, 2017-03-21:
I can confirm what @Batlh is saying above. I bought an actual Wolf Link Amiibo but I never owned a Wii U or TPHD and I felt it was unfair there was no way to increase the hearts. I was able to restore a MAXED OUT heart bin file to the Amiibo, but when I tried to use it in the game it showed an old WII style (white) dialog telling me that the Amiibo was corrupt. Luckily I had saved the original state and was able to restore it, but of course it only has 3 hearts.

It certainly does appear that something happens to the Amiibo when you play TPHD that these BIN files are not covering. It's also worth nothing that when I restore one of these BIN files to my Amiibo which has not been played on TPHD, I can no longer edit the hearts configuration in TagMo. When one of those files has been loaded to the Amiibo and then scanned back, tapping on 'EDIT TP DATA' doesn't do anything.

jbienz, 2017-03-21:
@sigboe I'm happy to help troubleshoot via e-mail too. Just send me a private message.

LinkofHyrule89, 2017-03-21:
I just ended up buying some NFC tags and writing the 20 heart link to it but after I use that once and save I can use my regular amiibo.

HuXu77, 2017-03-22:
Ok so based on @LinkofHyrule89's findings, you can write the bin to a tag and it work on BotW, after using the tag once and saving then any other Wolf Link Amiibo used after that point grants you the 20 hearts regardless of whats saved.

One exception is that @sigboe said he was able to pull it off successfully, if that is true then I would suggest maybe it is a phone and/or OS difference.

I have Wolf Link Amiibo and no tags.  I have an Samsung S7 and it reads from the Amiibo flawlessly.  I have backed up my Amiibo's tags, of which it writes a 532 byte file, but the bin that we are all using to restore is 540 bytes.

These are the two dumps hex values side by side.

[image]

I don't know anything about the tag formats other than briefly looking over the code that was committed here, but maybe someone more knowledgable can pick out what might make the difference.

I'm going to do some more research as well as attempt to restore the tag to my Amiibo to see if maybe an S7 will work.

If anyone has some findings to report do include the phone and OS you are using.

nopt, 2017-03-24:
It just seems the last 8 bytes are missing. I'm getting 532 byte files too, but I don't think it's that big of a deal. It's just 0s.
Did you try the most recent fork by masterchan-777 and the edit TP data feature ? https://github.com/masterchan-777/TagMo ?

sigboe, 2017-03-25:
@HuXu77 I never used TagMo on a real amiibo.

MattKimura, 2017-04-01:
If you guys want to max out wolf link with no effort whatsoever, there's a stupidly easy way to max out. You need fresh wolf link data, and a fresh Zelda TP HD save.

Make a save in TP, save the game and close the software. Now boot the game back up and it will ask you to register wolf link amiibo. Doing this automatically gives him max hearts for some reason. Now in BOTW he'll have 20 hearts.

Batlh, 2017-04-01:
@MattKimura how do you did that? have you used an emulator on the pc and a nfc usb reader/writer for it or has you used a wiiu for that problem?
because some people here has not even bought a wiiu since launch or bought TPHD ^^
so i bought me now nfc tags for it but with a price for 11 pieces of 11.88€ it's a little annoying -.-* that makes for one 1.08€ it's a little expensive and irritating because one solo has me cost when i had only bought one 2.58€ -.-*
any sugesstion where i can buy some cheaper nfc tags XD

so tried it out with the maxed out bin file for wolf link on the tag and it works just fine ^^
but had a little annoying problem with one nfc tag and the writing process that has cost me one of the nfc tags
so by the way the botw amiibo works just fine with an older version of tagmo version 2.3.2 that's the only one that can my phone use.
and yeah i had tried the other sources too like the one from azsde and other developer -.-*
so i had a little question for something else i tried out today ^^
wrote the ssb zelda amiibo bin file to one of my tags and now i'm a little curious about the droprate of the shadow bow Oo
the only bow i got from it is the royale bow and that more than once when i reload a manual save game befour use the tag and yeah there are other drops like rhodonit or topaz or saphir but i don't get it how the drop chance is ^^
and yeah i completet 3 dungeons in botw ^^ the only titan left is the one from the ori ^^ and i had my master sword befour i get to the first dungeon ;)

sry that was a little off topic ^^

MattKimura, 2017-04-01:
@Batlh  It's pretty off topic but needless to say it's useful information for what people want.
Yup I used a Wii U and Zelda TP HD, it's the only way. I'm not sure if the Wii U emulator is compatible with NFC adapters but if it is, that's another option. The method I posted is only good if you already have wolf link , but as someone mentioned above there's a pre-maxed out version of wolf link that you can use your tags with.

This is the cheapest I've found tags from: http://zipnfc.com/nfc-stickers/plain-white-plastic-nfc-sticker/nfc-sticker-38mm-bullseye-white-pvc-front-ntag215.html
If you live outside of Europe like me, you'll need to add it to your cart then go to the cart and use the shipping calculator to apply that to your cart. Else it'll charge you 49.99.

Tagmo 2.3.2 is the sweetspot for creating tags, it doesn't crash when writing a tag. Good luck!

CreateAPersonalAccount, 2017-04-14:
@MattKimura that didn't work. I started a new file in the third slot, skipped the opening cut scene, pressed down on the d-pad to bring up the menu, saved, didn't choose yes or no when it asked if I wanted to continue playing, hit home on the gamepad and closed the software, then started TP HD back up, chose the newly created third save file and got back into the game. At no point was I asked to register Wolf Link. Any idea why?

MattKimura, 2017-04-14:
@CreateAPersonalAccount  You need two things for this to work
1. A fresh Wolf Link amiibo (Delete all of it's data in system settings > Amiibo
2. A fresh Zelda TP save file (Never played before, no files created)

So once you have those two, repeat what you did and it should ask you to register the wolf link amiibo. Once you get to the title screen, you're golden and have 20 hearts. Worked for me and my friend.

CreateAPersonalAccount, 2017-04-14:
@MattKimura I'm still getting a 3 heart wolf Link after going through the process twice. The second time I unregistered the amiibo from both systems then deleted wolf Link's game data on both systems before deleting my Wii U's TP HD save data. It did ask me to register the amiibo both times but each time I started my Breath of the Wild file I had a 3 heart wolf Link.

Since every version of TagMo from 2.3 - 2.3.4beta all crash I was really hoping your method would work.

MattKimura, 2017-04-14:
@CreateAPersonalAccount  The method definitely works, but if it makes a difference I was using a tag and not the amiibo itself. 

Tagmo 2.3.4 and 2.3.3 do crash as you mentioned, but I find that 2.3.2 or even 2.1 work just fine. I recommend 2.3.2. And in that case, just create a 20 heart wolf link rather than doing this method. Me and my friend had tags with three heart wolf link, converted to 20 heart through that method. Sorry for making you delete your save data, I completely forgot that you can also just create a new user and that'll create fresh save data.

CreateAPersonalAccount, 2017-04-14:
@MattKimura my save data was all on a 128GB USB drive so no worries there. As for TagMo I've tried all the versions from 2.3.0 through 2.3.4beta as I said and they all crash. I have the keys and a bin that supposedly has 20 heart data as well as instructions on which values to change even if the bin doesn't work. It just doesn't matter though since it crashes every time.

jakerogers, 2017-04-15:
The most recent BotW patch has screwed with the way Wolf Link works. Simply having a TPHD save on the Amiibo is no longer enough. Is there a way to write a Wolf Link amiibo write the full Cave of Shadows data?

LinkofHyrule89, 2017-04-15:
Yeah my nfc tag is just a useless 3 heart wolf link now as well.

xxBigRonxx, 2017-04-15:
My wolf link is working fine with 20 hearts.

CreateAPersonalAccount, 2017-04-15:
@jakerogers if you can get a version of TagMo that doesn't crash, find keys and a bin, then yes.

jakerogers, 2017-04-15:
So far I've tried two Wolf downloads, one from the NFC Bank and one from elsewhere that specifically advertised having all 20 hearts, and both showed up in my game with three.

nopt, 2017-04-15:
Did you try masterchan-777 fork ? I've compiled it and it doesn't crash https://ufile.io/e8b92
Just Scan, Edit TP data (to completed + 20 hearts), then Restore Tag.

xxBigRonxx, 2017-04-15:
Is it allowed to post .apk?  I have a working one using the latest build.

sigboe, 2017-04-15:
@xxBigRonxx Yes its allowed to post .apk

Whats not allowed is to post keys or dumps of amiibos

xxBigRonxx, 2017-04-15:
Here is a link to the tagmo .apk I compiled on 04/11/2017.   Built with the latest source. 
https://ufile.io/e872p

CreateAPersonalAccount, 2017-04-15:
@nopt thank you so much! That made it a cinch to edit my amiibo's data. You've saved me a ton of time.

jakerogers, 2017-04-16:
@nopt I tried it on my Galaxy S5, but it throws an error when I try to edit the TP data. Says something about the amiibo being in an error state, even when I scan my real Wolf Link.

xxBigRonxx, 2017-04-16:
@JakeRogers. Make a new tag using an .apk compiled from the latest source and a working 20 wolf link. My wolf link is working fine after the update

Batlh, 2017-04-16:
i has no problem with my tag and a 20 heart wolf link and use the latest patch for botw ^^
by the way i think the problem ist something else then because i used the bin from nfcbank with wolf link and editet nothing on it ^^ and i had a second bin file with a maxed out wolflink with 20 hearts but i don't know if it will work if someone is willingly to test?

Barbaquez, 2017-05-19:
i found a download to a 20 heart bin that worked with my 215 tag i can give link if anyone wants

Bombastisch, 2017-08-03:
I guess this can be closed, right?   The editing is already possible!

HiddenRambler, 2017-08-03:
there was a request on gbatemp about the editor. i keept this open as a reminder to that.

[6 more replies]

closed 2017-12-19 by Bombastisch