HiddenRamblings/TagMo #8: Ability to update stats on existing amiibo?

Ekim0789, 2017-01-30:
Is there any way to update stats of an already programmed NFC chip or real amiibo? If not is this a planned future addition? Thanks.

LinkofHyrule89, 2017-01-31:
You should probably be more specific than this. The app already has ssb editing and a version for editing Wolf Link.

JimmyMultani, 2017-02-06:
@LinkofHyrule89, I think @Ekim0789 is asking whether you can write directly onto the amiibo.

Bombastisch, 2017-08-05:
If you want to edit the SSB data of a SSB-amiibo, you can do that easily. Just scan the amiibo, click on "Edit SSB", edit the attributes and then restore the changes to your existing amiibo.

Did this answer your question?

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch