HiddenRamblings/TagMo #9: App Crashes on Nexus 7 Gen 2

ClonedDog, 2017-02-18:
When ever you tap "WRITE TAG (AUTO)" after selecting your Amiibo bin the App Crashes this Bug is on "2.3.4 Pre-Release" and "2.3.3" but the App works just fine with "2.3.2" my Android Version is 6.0.1 Stock Please Fix this Bug :(

williamclarkmcbride, 2017-02-20:
I can confirm the same issue on a Galaxy S4 running Android 4.4.4 and an Honor 8 running Android 7.0

freythman, 2017-02-24:
I can also confirm the same is happening with a Galaxy S6 Edge+, running Android 6.0.1, stock. 2.3.3 and 2.3.4 exhibit this issue while 2.3.2 works.

exploitagency, 2017-02-24:
Just saw this issue in my watch list.  Are vector graphics broken still?

https://github.com/HiddenRambler/TagMo/issues/5#issuecomment-272720137
https://github.com/HiddenRambler/TagMo/issues/2#issuecomment-274370934

Should be an easy fix...

esmith13, 2017-02-28:
Confirmed to happen on Nexus 7 (2013) and Nexus 6P - both running stock Android Marshmellow. Issue happens on 2.3.3 and 2.3.4Beta. The fixed version on GBATemp Thread referenced in the 2.3.4Beta DOES work fine though on both Nexus devices without crashing!!

bergarces, 2017-03-08:
I had the issue with my 7.0 OnePlus 3T on all versions (2.3.2, 2.3.3 and 2.3.4Beta).

My Nexus 9 with Android 7.1 stock had the issue on 2.3.3 and 2.3.4, but it works with 2.3.2.

If you need any information or help with testing I'm happy to help.

azsde, 2017-03-09:
Please try my forked version, you will have to compile it yourself since I don't have the authorization to publish an apk.

esmith13, 2017-03-09:
azsde - I tried to build your with the included bat file but it failed. If you can help me figure out the problem I'd be happy to test for you.

azsde, 2017-03-09:
Do you have the Android SDK and NDK installed ?

qrohlf, 2017-03-23:
Same issue, here's the stack trace

```
--------- beginning of crash
03-22 18:33:50.878  8304  8396 E AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
03-22 18:33:50.878  8304  8396 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 8304
03-22 18:33:50.878  8304  8396 E AndroidRuntime: java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.hiddenramblings.tagmo-2/base.apk"],nativeLibraryDirectories=[/data/app/com.hiddenramblings.tagmo-2/lib/arm, /system/lib, /vendor/lib]]] couldn't find "libamiitool.so"
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.lang.Runtime.loadLibrary0(Runtime.java:972)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.lang.System.loadLibrary(System.java:1530)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.AmiiTool.<clinit>(AmiiTool.java:8)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.TagUtil.decrypt(TagUtil.java:93)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.TagUtil.patchUid(TagUtil.java:124)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.TagWriter.adjustTag(TagWriter.java:81)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.TagWriter.writeToTagAuto(TagWriter.java:41)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:176)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity_.access$501(NfcActivity_.java:30)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity_$6.execute(NfcActivity_.java:155)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at org.androidannotations.api.BackgroundExecutor$Task.run(BackgroundExecutor.java:405)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:428)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:272)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
03-22 18:33:50.878  8304  8396 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:761)
03-22 18:33:50.881   846  4124 W ActivityManager:   Force finishing activity com.hiddenramblings.tagmo/.NfcActivity_
03-22 18:33:50.886   846  4124 W ActivityManager:   Force finishing activity com.hiddenramblings.tagmo/.MainActivity_
```

jmarler, 2017-04-07:
I built the latest from azsde and can confirm that it works on a Nexus 7. I was able to load a saved tag, it was identified perfectly, and was able to Auto write it to a blank tag. Thank you very much! Built on Mac OS Sierra 10.12.4 w/ Android Studio and SDK 23, 24, and 25. I used my own signing certificate to sign the APK.

HiddenRambler, 2017-08-03:
I believe this is fixed in the latest version.

closed 2017-08-03 by HiddenRambler