HiddenRamblings/TagMo #10: App crashes after pressing Write - tested every release

minim8, 2017-03-05:
Galaxy Note 3, Android 5. 
Like the title says, I tested every release of this git and it always crashes when I try to write. Even 2.32, which seem to work for some people, crashes. Can anyone help?

shad00m, 2017-03-08:
Galaxy Note 2 Android 4.4.2 also crashes on Writing or restoring :(

azsde, 2017-03-09:
Please try with my forked version.

RoXuS, 2017-03-09:
@azsde do you have a released version of your fork ?

azsde, 2017-03-09:
No, I do not have the authorization to distribute it, you'll have to compile it yourself.

RoXuS, 2017-03-09:
Ok thx @azsde, I have the following error on your fork when I write:
```
03-09 17:47:13.924 8756-8985/? D/NfcActivity: Error
                                              java.lang.Exception: Error while writing main data (stage 1)
                                                  at com.hiddenramblings.tagmo.TagWriter.writeToTagAuto(TagWriter.java:49)
                                                  at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:169)
...
```

azsde, 2017-03-09:
Are you sure your tag is blank ?

If so, can you try placing it at different positions ?

Finally, if it still doesn't work, can you try using another file explorer when loading your .bin file ?

RoXuS, 2017-03-09:
@azsde ... Ok it's my fault, I took the only tag that looked corrupt among the 10 that I received ...

It is ok !! Thx.

Can we rewrite a tag that was written or not?

RoXuS, 2017-03-09:
@azsde it is ok with your version !

I have tested in game and it works (I have a wolf in BOTW on switch !!!).

Thx really.

Ok ok so I have to buy several others tags because I have only 9 tags.

zburke911, 2017-03-10:
@azsde,
This may be a stupid question but how would I go about compiling your fork?

DarkCodedDragon, 2017-03-14:
Roxus do you have compile of Azsde work? if not I can get to it when I have time just at work received my tags and  wanted to try write one.

Thanks

DarkCodedDragon, 2017-03-14:
I am getting this same crash with HTC M8 android 4.4.4 also tried complying AZSDE version inside of Android studio with gradle and still must be doing something wrong as I end up with 3 apks and all of them fail.

Satin181, 2017-03-20:
@azsde 
Hey, how do I compile your fork? The default tagmo isn't working for me.

DarkCodedDragon, 2017-03-20:
Look at comments on issue 13 Kidel goes into the steps on how to compile the fork.

Mikeferaz, 2017-04-03:
Hello, If anyone can help Id apreciate it. I have a Xperia Z5, so far Ive tried EVER version listed here https://github.com/HiddenRambler/TagMo/releases

none has worked, the last 3 versions crash when trying to write auto my tag

the bottom 3 versions like new hope didnt crash but said  "error while writing main data (stage 1) java.io.IOException: tranceive failed﻿"

Scanning an amiibo works fine, when scanning a tag I get "transceive failed"

suTrust, 2017-05-31:
Hello! Looking for instructions on compiling an APK... 
This page seems relevant:
https://stackoverflow.com/questions/35014071/how-can-i-compile-an-apk-from-a-github-source

Bombastisch, 2017-08-03:
Could you check the latest version (2.5) an see if your problems are solved?

Bombastisch, 2017-09-08:
Main topic of the issue should be fixed. If you @Mikeferaz still have a problem on the latest version, please open a new issue thread.

closed 2017-09-08 by Bombastisch