HiddenRamblings/TagMo #13: App crashes when trying to write or scan tag

Kidel, 2017-03-10:
Galaxy S4 (tested with 2.3.2, 2.3.4beta and @azsde branch)

here is the log:
```
03-10 15:37:55.266: I/Timeline(17256): Timeline: Activity_launch_request id:com.hiddenramblings.tagmo time:1641079
03-10 15:37:55.366: E/PathParser(17256): error in parsing "c0 1.1.9 2 2 2"
03-10 15:37:55.366: D/AndroidRuntime(17256): Shutting down VM
03-10 15:37:55.376: E/AndroidRuntime(17256): FATAL EXCEPTION: main
03-10 15:37:55.376: E/AndroidRuntime(17256): Process: com.hiddenramblings.tagmo, PID: 17256
03-10 15:37:55.376: E/AndroidRuntime(17256): java.lang.RuntimeException: Unable to start activity ComponentInfo{com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.NfcActivity_}: android.view.InflateException: Binary XML file line #31: Error inflating class ImageView
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2702)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2767)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread.access$900(ActivityThread.java:177)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1449)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.os.Handler.dispatchMessage(Handler.java:102)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.os.Looper.loop(Looper.java:145)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread.main(ActivityThread.java:5951)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.reflect.Method.invoke(Native Method)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.reflect.Method.invoke(Method.java:372)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1400)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1195)
03-10 15:37:55.376: E/AndroidRuntime(17256): Caused by: android.view.InflateException: Binary XML file line #31: Error inflating class ImageView
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:770)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:813)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.inflate(LayoutInflater.java:511)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.inflate(LayoutInflater.java:415)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.inflate(LayoutInflater.java:366)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.app.AppCompatDelegateImplV7.setContentView(AppCompatDelegateImplV7.java:256)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:109)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at com.hiddenramblings.tagmo.NfcActivity_.setContentView(NfcActivity_.java:51)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at com.hiddenramblings.tagmo.NfcActivity_.onCreate(NfcActivity_.java:42)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.Activity.performCreate(Activity.java:6289)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1119)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2655)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	... 10 more
03-10 15:37:55.376: E/AndroidRuntime(17256): Caused by: android.content.res.Resources$NotFoundException: File res/drawable-anydpi-v21/nfc_circle.xml from drawable resource ID #0x7f020047
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.content.res.Resources.loadDrawableForCookie(Resources.java:3735)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.content.res.Resources.loadDrawable(Resources.java:3603)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.content.res.TypedArray.getDrawable(TypedArray.java:762)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.widget.ImageView.<init>(ImageView.java:151)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.widget.ImageView.<init>(ImageView.java:140)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:58)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:54)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:95)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.app.AppCompatDelegateImplV7.createView(AppCompatDelegateImplV7.java:938)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v7.app.AppCompatDelegateImplV7.onCreateView(AppCompatDelegateImplV7.java:992)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.support.v4.view.LayoutInflaterCompatHC$FactoryWrapperHC.onCreateView(LayoutInflaterCompatHC.java:44)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:732)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	... 21 more
03-10 15:37:55.376: E/AndroidRuntime(17256): Caused by: java.lang.NumberFormatException: Invalid float: "1.1.9"
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.StringToReal.invalidReal(StringToReal.java:63)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.StringToReal.initialParse(StringToReal.java:160)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.StringToReal.parseFloat(StringToReal.java:323)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at java.lang.Float.parseFloat(Float.java:306)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.util.PathParser.getFloats(PathParser.java:178)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.util.PathParser.createNodesFromPathData(PathParser.java:59)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.VectorDrawable$VFullPath.updateStateFromTypedArray(VectorDrawable.java:1439)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.VectorDrawable$VFullPath.inflate(VectorDrawable.java:1421)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.VectorDrawable.inflateInternal(VectorDrawable.java:542)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.VectorDrawable.inflate(VectorDrawable.java:458)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:1150)
03-10 15:37:55.376: E/AndroidRuntime(17256): 	at android.graphics.drawable.Drawable.createFromXml(D
03-10 15:37:58.829: I/Process(17256): Sending signal. PID: 17256 SIG: 9
```

Kidel, 2017-03-10:
Nevermind, I fixed it commenting out all the ImageView parts inthe xml and in the code. I suggest you replace it with a basic shape since it causes issues to everybody apparently.

Flecktarnfarbe, 2017-03-11:
@Kidel I have the same problems with my S4. Can you tell me please please which xml files you changed?

Kidel, 2017-03-11:
activity_nfc.xml and the NfcActivity class. 
I'll probably do a fork with a better fix, since I don't like simply commenting out stuff. UI problems shouldn't cause the application to stop completely.

Flecktarnfarbe, 2017-03-11:
Thx. I have no idea where to commenting out the view parts in the NfcActivity class -.- that is all new, never changed an app before. where can I find your fork when published?

Kidel, 2017-03-11:
Working on it right now. I'll let you know asap.

Kidel, 2017-03-11:
Here it is: https://github.com/Kidel/TagMo

Kidel, 2017-03-11:
I've also merged with @azsde 's branch

Flecktarnfarbe, 2017-03-11:
Great. Now i have to compile the code into an apk?

Kidel, 2017-03-11:
Yes

Flecktarnfarbe, 2017-03-11:
Ok lets try 😁

glenlivet99, 2017-03-11:
Will you post a compiled APK for others that don't have the means to do so?

Kidel, 2017-03-11:
I don't think I have the authority to do it. However you basically open the project with Android Studio, download the sdk for your Android version, apply every fix that Gradle suggests and then go to Build > Build APK

Flecktarnfarbe, 2017-03-11:
App is working. Thanks so much. Now i am waiting for my tags. Great work 👍

Kidel, 2017-03-11:
here is the pull request if you want to review it:
https://github.com/HiddenRambler/TagMo/pull/14

sxywalrus, 2017-03-12:
Not sure where I went wrong. I made all the changes suggested by Gradle, did Build > Build APK and got 3 outputs names: app-fat-armeabi-v7a-debug.apk, app-fat-universal-debug.apk, and app-fat-x86-debug.apk. I ran them all and none of them could write, still received the transceive failed (stage 1) java error when trying to write to a blank tag.

pyrostic, 2017-03-13:
Same, a compiled version would be greatly appreciated.

azsde, 2017-03-13:
Stop asking for a compiled version, we can't provide one unless @HiddenRambler says we are allowed to

pyrostic, 2017-03-13:
Sorry, didn't know it wasn't allowed I'll try and figure it out

glenlivet99, 2017-03-13:
This is open source, isn't that the point? You don't need permission

pyrostic, 2017-03-13:
Kinda what I figured, but maybe something with GNU licensing for a
releasable apk.

Kidel, 2017-03-13:
Somebody forked my version and  made a release anyway :/
He re on github.

closed 2017-03-13 by Kidel

sxywalrus, 2017-03-13:
well at least post the link before closing this.

Kidel, 2017-03-14:
I was going to, but then your attitude made me change my mind. Please understand that we are not your employees, nor we own you something. At least say "please" next time.

sxywalrus, 2017-03-14:
I didn't mean to sound like I had attitude. I should have said please I agree, but reading text on a message board is very different then speaking to someone. I can understand that it sounded forceful and rushed, when I was asking for the working app. 
Everyone stated it was okay to release it, so I didn't understand why you didnt release it, but still closed the topic. That to me was unwarranted, and showed that you didn't care about helping us who needed it and asked for it. I do appreciate the help don't get me wrong, this is way beyond anything I know. Thank you.

Kidel, 2017-03-14:
I closed the issue (that I opened in the first place) because the issue is solved. And as you can see it's not preventing you from replying. The fact that everyone says that it's ok for me to do a wrong thing doesn't make it less wrong and I gave you my reasons for not releasing an apk (that, again, you can make it yourself since I already made the code for you). And I actually didn't even have to give you any reason to be honest.
Also I don't get something, was that meant to be an apology? You said "it showed that you didn't care about helping us who needed it". To whom? Because as far as I know you're the only one who's complaining and everythng about this repository is about free help to others. 

Sorry, it's not getting any better.

Kidel, 2017-03-14:
Plus, you even did compile the apk by yourself, and the error that gives you, if I'm not wrong, is not a crash, but an handled exception, meaning you're doing something wrong. An apk compiled by me won't fix anything for you, sorry.

sxywalrus, 2017-03-14:
Those who requested the forked APK, or your compiled APK I guess. I know I did something wrong, that's why I asked for detailed instructions or a compiled version of the APK. I said before, this is all new for me, my first time working with code, building and compiling APKs, hence why I am thankful for your knowledge.
I missed the apology, while I didn't mean for it to have attitude, I didn't ask nicely and I apologize for that. It's my fault and because of my desperation to get this working, I rushed a reply and it came out the way it did. I am sorry for that.
I also did not realize that there are rules for releasing forked APK's and misunderstood that open source doesn't necessarily mean you can release your own APK. I was taking two people's word as law, when I honestly have no clue how this place works.
After watching several videos I got the compile correct and was able to write 6 ntags215s with your code, thank you for that.

Kidel, 2017-03-14:
good to hear :)

sxywalrus, 2017-03-14:
Yep, I did notice that after testing, when I scanned a tag they would not write.

sxywalrus, 2017-03-14:
But when I wrote instead of scanning first, they worked
Ah, no way to edit posts from mobile.