HiddenRamblings/TagMo #16: Crash with android 6

Rossomak, 2017-03-14:
As the title implies, I'm having issues.
I have "NFC enabled," "Unfixed key OK," "Locked key OK" and a common tag loaded, ready to be written onto brand new ntag215 cards. I'm using a LG G3.
However, when I press "Write Tag (Auto)" and press my card to the phone, I get the error "Unfortunately, Tagmo has stopped" error. I've made sure the app is up to date. Other things that crash the app are "View Hex," and "Edit SSB Data." Although I don't know what it does, I've also tried "Write Tag (Raw)," where I get the message "Source UID does not match the target!" "Scan Tag" and "Save Tag" both work fine. "Restore Tag" crashes.
Also, when I "Scan Tag" the blank cards, it says "Mario SSB" is loaded, however when I scan with amiiqo, it does not say this. Idk what's going on. Is there perhaps something wrong with my keys? When I tap the blank card to my phone without any apps, it confirms it is a blank card.
My phone is currently running android 6. I checked for an update manually but there is none for my phone yet.
Any solutions?

Kidel, 2017-03-14:
check my branch or @azsde 's

binpas, 2017-03-17:
@Kidel I'm having the same problem.  However, I'm not well versed in building android apps.  Do you have a build of your branch available? @Kidel

Batlh, 2017-03-19:
That error what rossomak has i know what he means have the same problem the only version i can get to work by my LG G3 is version 2.3.2 without a fore close crash.
so i get there a error with slot 1 or something like that when i try to write a tag -.-*
and versions over this keeps crashing after i try to write something or click hex edit or ssb stats edit.
in 2.3.2 the only time it crashs in force close when i try to edit ssb stats -.-*
and the other versions of this app
i can't try because dunno get the data complilled to a apk ^^ would be nice if it would there any link for downloading

DarkCodedDragon, 2017-03-19:
Kidel posted the steps to compile in one of the other issues . Compile his build and it will work. They can't release a apk with out permission.

Batlh, 2017-03-19:
ok i will search for it ^^

DarkCodedDragon, 2017-03-20:
Issue 13 has the steps

Batlh, 2017-04-01:
so the version 2.3.2 works fine for lg g3 so if anyone have problem with tagmo with a g3 try this version ^^

Bombastisch, 2017-08-03:
Seems like there are no more problems.
Also this problem is fixed in the lastest version (2.5)

--Closing this issue

closed 2017-08-03 by Bombastisch