HiddenRamblings/TagMo #17: App crashes on S5 with android 5

a7mag3ddon, 2017-03-14:
As title, when I click WRITE or READ tag app crashes.

Kidel, 2017-03-14:
check my branch or @azsde 's

a7mag3ddon, 2017-03-14:
Sorted. Thank k you :-)

johnsmclay, 2017-03-15:
It took me over an hour to load the Android Studio and all the dependencies so I could build @azsde 's version. It worked great, but I thought others might appreciate a built APK (zipped, because Github requirement)
[TagMo_azsde_41807e7_20170315.zip](https://github.com/HiddenRambler/TagMo/files/846095/TagMo_azsde_41807e7_20170315.zip)

DarkCodedDragon, 2017-03-16:
What are you using for phone etc . Apk doesn't work for my HTC m8 on android 4.4.4

johnsmclay, 2017-03-16:
I grab the "Universal" APK it built. I have an axon 7 running Android 7.0.

BSmith237, 2017-03-16:
I'm not having any luck with 2.3.2, 2.3.4 or John's build (btw, thanks for doing that).  I'm running 6.0.1 on an unrooted S6.  It'll read my Amiibo fine, seems to load keys and tags fine, crashes on "Edit SSB data", and gives me the "Transceive Error" when I try to write a tag using the "Auto" button.  Any suggestions?

BSmith237, 2017-03-16:
I also tried the universal version of Kidel's latest branch from another issue and have used multiple cards.  Still no luck.

Kidel, 2017-03-16:
Does it crash or you get an on screen error?

DarkCodedDragon, 2017-03-16:
With johns build I get a java error when I go to write the tag. something java exception I think it was posted on another issue thread. Left my tags at home so cant write to one atm.

Kidel, 2017-03-16:
does the app close or not?

DarkCodedDragon, 2017-03-16:
If I am not writing a tag currently application will stay open and wait for tag. If tag is applied I believe it just states the java error , on other builds it says tagmo has stopped and crashes that requires relaunch of the application.  I will  be off and back home @ around 5cst so I can post full message then.

Kidel, 2017-03-16:
if the app doesn't close but gives an on screen error (aka an handled exception), then you're doing something wrong, like using an nfc tag that is already written or not nfc215.

DarkCodedDragon, 2017-03-16:
Thank you Kidel , I will check few more tags when I get home.  I got ten or well 9 more I will check to see if anything is loaded onto it and they should be 215 tags but never know. I know I did check before I tried to load anything on it first time and it was blank. Since then I have probably tried every build of tagmo with the same chip? Is it possible to corrupt a tag?

Kidel, 2017-03-16:
Unlikely.

johnsmclay, 2017-03-16:
If you are getting "Error while writing main data (stage 1) java.io.IOException: Tranceive failed"  I only get that if I try to re-write a locked tag.  If you wanna check out your tag, get NXP's Taginfo app.  After scanning  go over to the tech tab.  In memory content you should have a bunch of lines w/  " *  00 00 00 00 |****|".  If the lines start with "+r" or just "r" then it is write-protected.

DarkCodedDragon, 2017-03-16:
I wrote to the tag that was the issue thank you

closed 2017-03-17 by a7mag3ddon