HiddenRamblings/TagMo #18: Makes NFC tags read only?

gummby8, 2017-03-17:
I got a few NTAG215s and their site shows I should be able to overwrite the tags multiple times. But after writing an amiibo to one, it becomes write locked. I can't format the tags even with other apps, they all say it is now write protected.

humphries40, 2017-03-17:
Normally NTAG215s are rewritable, however with Amiibos it write locks them. You can't re-use them. 1 Tag = 1 Amiibo.

andycjw, 2017-03-27:
I was having the same question, looking into the NTAG 215 spec and 3dbrew Amiibo data structure answered my curiousity

specifically, this two line
0x83	0x1	0x20C	0x4	No	Standard NTAG215: CFG0. Must match raw binary "00 00 00 04".
0x84	0x1	0x210	0x4	No	Standard NTAG215: CFG1. Must match raw binary "5F 00 00 00".

5F means it'll write locked, sharing this here if anyone else is also interested in this

https://www.nxp.com/documents/data_sheet/NTAG213_215_216.pdf
https://www.3dbrew.org/wiki/Amiibo

cardonator, 2017-04-11:
Out of curiosity, how does Nintendo write back to the Amiibo if they are write locked? Like the wolf link?

andycjw, 2017-04-13:
the write lock only locks the user configuration, there are app data pages that is still writable, but those pages doesn't include the data that determines what model of amiibo, those are in locked page

jordanbtucker, 2017-04-18:
So, is it impossible to overwrite an amiibo tag, or are there workarounds?

servel333, 2017-05-20:
From what I understand, for an Amiibo to work the part of the NTAG215 that says what kind of Amiibo it is must be write locked. Other memory in the tag can be rewritten but this is just data and not what identifies the Amiibo.

jmh363905, 2017-08-02:
Just throwing it out there, but this can be filed under the "Things that are supposed to happen" column, and I'd close this issue.

closed 2017-08-02 by HiddenRambler

llakssz, 2017-08-26:
Interesting, it sounds like if we patched the nfc module of the 3ds/wii u to not check for that '5F' writelock, then we could write amiibos with that lock off, and it would work (with hacks) and be able to be reused.

From a software point of view, adding an option to patch the amiibo data written to not be locked should be easy.

The tricky part is the 3ds/wiiu patch.

flarn2006, 2020-12-06:
I think there should be a warning of this in the app before you write a tag. I wrote the same amiibo to two stickers because I stuck the wrong one somewhere and figured I'd just reprogram the other one instead. If I had gotten a warning to that effect, I wouldn't have done that.

Also, does anyone make NFC tags that follow the NTAG215 spec, with the exception that they don't implement the function of the locking bits? So you could set the config that the systems want, while keeping the ability to reprogram it.

jmh363905, 2020-12-06:
> I think there should be a warning of this in the app before you write a tag. I wrote the same amiibo to two stickers because I stuck the wrong one somewhere and figured I'd just reprogram the other [...]

You literally can't write any tag and it not be read-only.  PowerSaves allows this, but that's a whole different beast.  By nature they must be write protected or they won't pass authentication on the console and therefore won't scan.  What you're asking for is a popup to tell you what must happen to even produce a useable tag on your console.  What you said is like saying: I turned on my monitor at night and it brightened up the room, if I had known it was going to do this, I wouldn't have turned the monitor on.

Bombastisch, 2020-12-06:
Hm,
I get what flarn2006 wants.
Maybe a final warning on the writing-process.
It's already stated in the ReadMe, but who reads those...

jmh363905, 2020-12-06:
> Hm,
> I get what flarn2006 wants.
> Maybe a final warning on the writing-process.
> It's already stated in the ReadMe, but who reads those...

Honestly, that would mean a popup on every....single.....write....  That would be more of a nuisance IMHO to have a pop up tell you what should already be known by the user on every...single...write....  Who wants to dismiss a message every time a piece of software is functioning as it is supposed to?  What if someone is fixing to write 26 tags back to back (the "Zelda" pack for BotW for example)--a "warning" is going to get annoying very very fast.  Meanwhile when I save a Word document to a flash drive, I don't get a popup that my available space is going to decrease...this is something that's supposed to happen.  :-)

Bombastisch, 2020-12-06:
Not talking about a pop-up. Everyone hates pop-ups.
I was thinking of just putting a warning text on the screen when you are about to write a tag.

jmh363905, 2020-12-06:
> Not talking about a pop-up. Everyone hates pop-ups.
> I was thinking of just putting a warning text on the screen when you are about to write a tag.

That would be better, indeed.  Maybe some flashing red text under the write button.

Golem642, 2024-03-29:
> I was having the same question, looking into the NTAG 215 spec and 3dbrew Amiibo data structure answered my curiousity
>
> specifically, this two line
> 0x83	0x1	0x20C	0x4	No	Standard NTAG215: CFG0. Must [...]

Currently making a N3DS IPS patch for this. Are there any other bytes I should be aware of ? Because i saw two other lines mentionning some kind of lock :
- On NFC page 0x0 : Standard NTAG215: 9-byte serial-number, "internal" u8 value, then the two lock bytes which must match raw binary "0F E0".
On NFC page 0x82 : Standard NTAG215: first 3-bytes are dynamic lock bytes. Must match raw binary "01 00 0F".

Do I need to modify those as well ?