HiddenRamblings/TagMo #20: Unable to load native libs

setheclark, 2017-03-19:
It seems that something changed with the most recent builds (2.3.4 and 2.3.3) that isn't including the native libs in the apk.  When I try to write a tag I end up with the following stacktrace:
```
E/AndroidRuntime( 2082): FATAL EXCEPTION: main
E/AndroidRuntime( 2082): Process: com.hiddenramblings.tagmo, PID: 2082
E/AndroidRuntime( 2082): java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.hiddenramblings.tagmo-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]] couldn't find "libamiitool.so"
E/AndroidRuntime( 2082): 	at java.lang.Runtime.loadLibrary(Runtime.java:366)
E/AndroidRuntime( 2082): 	at java.lang.System.loadLibrary(System.java:988)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.AmiiTool.<clinit>(AmiiTool.java:8)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.TagUtil.decrypt(TagUtil.java:93)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.EditorSSB.afterViews(EditorSSB.java:82)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.EditorSSB_.onViewChanged(EditorSSB_.java:96)
E/AndroidRuntime( 2082): 	at org.androidannotations.api.view.OnViewChangedNotifier.notifyViewChanged(OnViewChangedNotifier.java:41)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.EditorSSB_.setContentView(EditorSSB_.java:52)
E/AndroidRuntime( 2082): 	at com.hiddenramblings.tagmo.EditorSSB_.onCreate(EditorSSB_.java:42)
E/AndroidRuntime( 2082): 	at android.app.Activity.performCreate(Activity.java:5990)
E/AndroidRuntime( 2082): 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
E/AndroidRuntime( 2082): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2278)
E/AndroidRuntime( 2082): 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
E/AndroidRuntime( 2082): 	at android.app.ActivityThread.access$800(ActivityThread.java:151)
E/AndroidRuntime( 2082): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
E/AndroidRuntime( 2082): 	at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 2082): 	at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime( 2082): 	at android.app.ActivityThread.main(ActivityThread.java:5254)
E/AndroidRuntime( 2082): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 2082): 	at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime( 2082): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
E/AndroidRuntime( 2082): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
```

Looking at these APKs in the AS analyzer shows that the native libs are missing:
[image]

Compare this with version 2.3.2 which does not crash when writing a tag:
[image]

setheclark, 2017-03-19:
This issue would be fixed once #11 is merged.

azsde, 2017-03-23:
Use my branch :)

setheclark, 2017-03-23:
I had already fixed it myself. Noticed you probably fixed it when I was about to submit mine.

HiddenRambler, 2017-08-03:
Fixed in latest version.

closed 2017-08-03 by HiddenRambler