HiddenRamblings/TagMo #21: Crash on Android 5

Raknell, 2017-03-21:
I have a Samsung Galaxy Note 3 with Android 5 and my apk crash when I try to scan or write.
What version of the app do I need? or what do I have to do to make it work? Do I need to have the phone rooted? 
I need help ! D:

DarkCodedDragon, 2017-03-21:
Version you need cant be released atm because hiddenmaster has not given any one permission to release the apk. Phone does not have to be rooted. Compile the Code with steps listed by Kidel in issue #13 from Kidel or Azde branch that will solve the crashing issue.

Bombastisch, 2017-08-01:
Try the newest version. Your problem should be fixed! :)

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch