HiddenRamblings/TagMo #24: Crashes when attempting to Write

genekri7, 2017-03-24:
Using S7Edge android v6.0.1 tried 2.3.3 and 2.3.4 apks.

Anyone else similar or know a fix?

LordFrz, 2017-03-26:
Same issue on the S7Edge, unable to write.

LordFrz, 2017-03-26:
IT seems to be an issue with the S7Edge, as my old axiom works, but not my new S7.

theDevelopper, 2017-03-28:
Not working on Pixel as well. Tried 2.3.4 beta as well. Simply crashes

DarkCodedDragon, 2017-03-29:
See issue 13

LordFrz, 2017-03-29:
azsde's branch while not crashing, still gives an error.

Mikeferaz, 2017-04-03:
Hello, If anyone can help Id apreciate it. I have a Xperia Z5, so far Ive tried EVER version listed here https://github.com/HiddenRambler/TagMo/releases

none has worked, the last 3 versions crash when trying to write auto my tag

the bottom 3 versions like new hope didnt crash but said  "error while writing main data (stage 1) java.io.IOException: tranceive failed﻿"

Scanning an amiibo works fine, when scanning a tag I get "transceive failed"

johnaho, 2017-04-04:
@Mikeferaz Have you tried version 2.3.1?  I have a sony xperia Z3v and it worked with that version.  I selected 'Allow Restore to different tag' and used Write-Auto.   

I also couldn't get any of the releases newer than 2.3.1 to work writing tags.

Mikeferaz, 2017-04-04:
@johnaho ibe tried every release in the link I outted in my comment ,  2.3.1 is the one that sais permisison strikes back lol, but no ot dodnt work, perhaps your version is diferent? If you still have the Apk fule roaming in your smarthphone can you send ir to me? Ans the keys too? maybe a long shot bur who knows, ive tried everything

johnaho, 2017-04-04:
@Mikeferaz Well, I just used the files ending in *.apk   Unless you want to compile your own version of this application you don't need the source code zips.

I just used the 2.3.1 release listed on the page.

I can't send you the keys, that is something you have to find yourself. When you do have the keys loaded it will display "Unfixed key OK and Locked key OK" on the upper left side of the screen. 

Technique wise, I put my sticker tag down and when I was ready to write I put the phone down on the tag and moved it slowly until it recognized the tag and when it was finished it said "done" and went back to the main menu screen.

Mikeferaz, 2017-04-04:
@johnaho  I just tried with the sticker laid down like you said, still the same prolem, also Im using 2.3.1, I do have the keys ON (green) however I asked if I could use your files because Im not really a programer nor anything close, and at this point Im just trying with everything I find. I still got the same error stage 1 message, so far nothing, what could I be posiblely doing wrong, for everyone in youtube works like a charm!

raindog14, 2017-04-04:
being unable to make the edits and compile, I went and installed the 2.3.1 and after crashing on my xperia m4 aqua on lollipop using the first 3, that one worked (pressed button to write, no crash, put on tag, it wrote!)

scottjl, 2017-04-18:
Ditto for me, LG V20 (LG-H918) running Android 7.0.

Going back to 2.3.1 works for me. I can read and write, but the "Edit SSB Data" option will crash on a .bin, works on a real Amiibo.

vallieres, 2017-05-10:
Pixel XL, crashing on 2.3.4b and 2.3.3 but 2.3.2 works. Names of scans are wrong though.

quantumfizix, 2017-05-10:
@vallieres Are you trying to load and write newer amiibos? The library currently isn't complete, so you may see a generic load profile, even though the amiibo is the correct code.  The library was built out more in 2.3.4+

vallieres, 2017-05-10:
@quantumfizix old and new. Makes no difference, I get the app stopped working and the NFC card is useless. I did notice the weird names so I think I might just fork the repo and cherry pick the commits that refer to the up to date library.

quantumfizix, 2017-05-10:
@vallieres I am working on a Samsung Galaxy S4, and on version 2.3.2, and most of the tags I populate come back as Mario (with the correct type).  Didn't know if it was the same issue that I was working with.

vallieres, 2017-05-10:
@quantumfizix Yes, the name is wrong.  Like Mario (LOZ) for Zelda amiibos and I think this is expected since 2.3.2 has an incomplete library. But the tag written to the NFC card is valid.

quantumfizix, 2017-05-10:
@vallieres sounds like we are experiencing the same issue then.

I'm trying to work on the code to separate the hashlists into an XML and allow just that to be updated, so you don't need to get a new version of the code every time, just the updated XML.  Will try to commit if I get it working.

zshaheen, 2017-06-01:
2.3.2 works on my S7 Edge SM-G935T. I'm on 7.0.

Bombastisch, 2017-08-01:
Try the newest release. Your problems should be fixed.

scottjl, 2017-08-02:
Tried to install 2.5 but my phone keeps telling me that the apk is corrupt. I was able to install 2.4.3 successfully though.

HiddenRambler, 2017-08-03:
@scottjl  can you uninstall the old version and install the latest? that worked for another user.

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch