HiddenRamblings/TagMo #25: Error while writing main data (stage 1)

LordFrz, 2017-03-26:
I am getting an error saying tag was lost, any idea on what I am doing wrong?

LordFrz, 2017-03-26:
Was able to run it on my old Samsung Axiom, must be an issue with new version of android or the S7 Edge.

Zuryan, 2017-04-04:
I had the same error on my Galaxy S5 SM-G911F Android 6.0.1

demalmar, 2017-04-20:
Same here, Note 4, Android 5.1.1. I can't re-write a written card. Any help on this?

LordFrz, 2017-04-20:
@demalmar I don't think you can rewrite a written card. Not with this program anyway, as it's for amiibo which can NOT be overwritten.

Bombastisch, 2017-08-05:
Could you descrbe your problem more specific?   Did you want to write on a already existing amiibo or did you uave blank Ntag215s?

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch