HiddenRamblings/TagMo #26: [SUGGESTION] QR code Amiibo?

PokeAcer549, 2017-03-27:
I have a suggestion that may be helpful if one obtains their NFC BINs from a source that's not their own amiibo 😉 
The idea is to put Amiibo in a QR code one can scan and thus eliminate the need of waiting for *those* sites todo their 10esc countdown :3

Bombastisch, 2017-06-29:
If you would create a qr-code of the Hex values an then program something that converts it back to a .bin wouldn't be so hard I guess. Just tried creating a qr-code of the hex values and it worked!

Bombastisch, 2017-06-29:
But the QR codes are of that size....  

http://q-r-code.fr/qrcodes/67e238a2ed9ded893ac0273c3260ee1c.png?1498758214952

Bombastisch, 2017-08-01:
Yay! This feature was added in the newest release. Check it out!

closed 2017-08-01 by HiddenRambler