HiddenRamblings/TagMo #27: NFC not Supported?

Keyento, 2017-03-31:
I have an LG Escape 3. It is a Cricket(Carrier) phone. When I download the tagmo onto my phone, it says that it is not NFC supported. What is strange is that my phone is NFC supported, and I even searched up on the internet to find out if it is. I also can't find any NFC settings inside the phone settings. Is there a glitch, a defect, or my phone actually doesn't have NFC?

Yawrf, 2017-04-01:
I think the NFC is actually part of the battery on my phone (Either that or they have stupid labeling systems), so it's possible that your phone /could/ do it, but doesn't have the hardware currently

Bombastisch, 2017-08-05:
The easiest way to verify if your phone has NFC is to use the NXP TagInfo app and try to scan a amiibo or any other NFC tag: https://play.google.com/store/apps/details?id=com.nxp.taginfolite

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch