HiddenRamblings/TagMo #28: Crash on Galaxy S6 Edge plus

xXtheguy52Xx, 2017-03-31:
crashes every time i tap edit ssb Data

lozamiibo, 2017-04-05:
Same issue here on s7 edge..

joshbgosh10592, 2017-06-28:
Same here. Read/writes no problem, but tapping on the Edit SSB data crashes without loading any other screen.

Bombastisch, 2017-06-28:
It works! But you can't edit SSB data if there is no SSB data! Nake sure you scan a SSB Amiibo first with SSB data on it, then you can edit its datas. Trying to edit the binary files won't work, because there is no SSB data.

joshbgosh10592, 2017-06-28:
Looking into this more, I was trying an Amiibo-coded NFC215 chip, but it was fresh - never used in SSB (Toon Link, Windwaker).
I used another chip that I had used before in SSB (Mega Yarn Yoshi), and it was able to load into the modifier.
Just to confirm my findings, I loaded the "failing" chip into SSB, created save data, and tried again. This time, Toon Link: WW, was able to be modified.
Maybe @HiddenRambler could just add a catch when that button is pressed. Instead of crashing in an unexpected handler, display an error saying, "This Amiibo tag does not yet have save data, or is not supported in SSB."

HiddenRambler, 2017-06-28:
Can confirm. I ran into this myself. Will put in error handling for this.

Bombastisch, 2017-09-08:
Fixed but probably forgot to close xD .   If there a still any problems feel free to open a new issue thread. :)

closed 2017-09-08 by Bombastisch