HiddenRamblings/TagMo #29: Crashes when I try to Edit SSB Info

Yawrf, 2017-04-01:
Running Android 6.0.1 and the latest full release, every time I hit the Edit SSB Info button, it crashes. Anyone know why? (Or if it's known and being fixed, I didn't see it among the issues, but I may just be blind)

lozamiibo, 2017-04-05:
Same issue here...

azsde, 2017-04-11:
It only works when scanning a real amiibo, not when loading a .bin file.

lozamiibo, 2017-04-11:
Thanks for the info!

Bombastisch, 2017-05-14:
Still doesn't work....   (Nexus 5X,  Lineage OS, Android 7.0.1)

I'm really interested in this feature but it just won't work!

Bombastisch, 2017-05-15:
Okay, my amiibo was kinda broken? It works now, no problems anymore

smokeandlights, 2017-05-23:
I am also having the same problem. I can scan and load just fine, but if I try to scan an amiibo and edit ssb data or view hex, the app crashes. I haven't tried to write tags yet. Android 6.0.1, Samsung Galaxy S5 (G900V).

smokeandlights, 2017-06-19:
This seems to be fixed in 2.4.0 . Thanks!

closed 2017-06-26 by HiddenRambler