HiddenRamblings/TagMo #30: Support for Datel PowerTags

daleriechert, 2017-04-06:
I'd like to request investigation into the support of Datel's rewritable NFC PowerTags. This would circumvent the need for single-use tags, and support from this app would also circumvent Datel's proprietary PowerSaves USB NFC writer and software. A single tag is bundled with new packages of the PowerSaves writer, or separately in 3 and 10 packs here:
http://www.ebay.com/itm/252829545695
http://www.ebay.com/itm/252828341082

The tags seem to arrive blank, all FF bytes. I attempted to write SSB Mario to one, and data is written to the tag, but it is not valid in-game. Notably bytes 01D5-01DB remain FF after the tag has been written to.

Rafcin, 2017-04-09:
If I knew what FFFF... represents then I could write it, same issue

Rafcin, 2017-04-09:
*Update* I changed the values of the FFFF... still wont work, in Botw it just makes a beep.

Bombastisch, 2017-08-26:
Is there a good description, what this PowerTag is made for and how it works? I can't find a good one.

daleriechert, 2017-08-26:
I believe this tool has recently added some Powertag support which could lead to clues about its functionality.
https://gbatemp.net/threads/tools-amiibomb.467990/
https://github.com/AcK77/AmiiBomb-uino

Bombastisch, 2017-08-26:
As far as I unsderstood that article if you have the program called "power save" (and the nfc card reader) and a power tag you can create amiibos out of it.

Could you try to do it the legitimate way and tell me if that worked?

daleriechert, 2017-08-26:
Why wouldn't the legitimate way work? I do not have the Powersave NFC base and proprietary software, so I cannot confirm.

Bombastisch, 2017-08-26:
Ok,so we need to investigate in a other way. Please wipe you power tag using this app: 

https://www.google.de/url?sa=t&source=web&rct=j&url=https://play.google.com/store/apps/details%3Fid%3Dcom.nxp.nfc.tagwriter%26hl%3Dde%26referrer%3Dutm_source%253Dgoogle%2526utm_medium%253Dorganic%2526utm_term%253Dnfc%2Btagwriter%26pcampaignid%3DAPPU_1_yuWhWfqCEIiqUdy8iegE&ved=0ahUKEwj6j8bk6fXVAhUIVRQKHVxeAk0Q8oQBCB0wAA&usg=AFQjCNEc-GWARw-ILZziPpwYP2t2pdwk4A

Then try to write to the Tag with the newest version of TagMo.

Then use :    https://www.google.de/url?sa=t&source=web&rct=j&url=https://play.google.com/store/apps/details%3Fid%3Dcom.nxp.taginfolite%26hl%3Dde%26referrer%3Dutm_source%253Dgoogle%2526utm_medium%253Dorganic%2526utm_term%253Dnfc%2Btaginfo%26pcampaignid%3DAPPU_1_C-ihWaX-JYSBUcfhhaAI&ved=0ahUKEwjlr-336_XVAhWEQBQKHcdwAYQQ8oQBCB8wAQ&usg=AFQjCNFRo1GsbEIs6QFFhEQwdLxb0LuVLA

To see if the Powertag is identified as a NTag215

Please post your results.

Bombastisch, 2017-08-26:
Oh, links dont work

Bombastisch, 2017-08-26:
Just look in the playstore for:

#1: NXP TagWriter

#2NXP TagInfo

Bombastisch, 2017-08-27:
So I still need you to do the steps mentioned about.

After further invetigation on my own, i found out that where is no UID written to the PowerTag.    
(a ID that is used to tell apart two amiibos of the same type).

Please send me your results afterwards. :)

HiddenRambler, 2017-08-27:
@daleriechert  01D5-01DB  represent the UID of the tag which is supposed to be unique to each NFC tag. as @Bombastisch mentioned could you give the output of the nxp tag info app?

It might be that we could simply write to those pages (not possible for legitimate tags). If i made a custom app would you be willing to try?

daleriechert, 2017-08-28:
@Bombastisch @HiddenRambler The tags are not identified as NTAG215, but NTAG203(F). Erasing my written tag using the first app is not at all effective. TagMo still identifies it as the Zelda Guardian Amiibo, so I guess my last Mario write to it was ineffective. Thus because it is already identified as an Amiibo TagMo will not write to it again. This is while unchecking NTAG215 validation of course.

Here is the dump of this Guardian tag using the second app: [https://pastebin.com/raw/DAR7sRam](https://pastebin.com/raw/DAR7sRam)

As I had two additional blank tags (all FF's), I wrote Mario to one using TagMo. Here is the dump of it, similar results: [https://pastebin.com/raw/ciXdFC2w](https://pastebin.com/raw/ciXdFC2w)

Thus I have one more blank tag, and these two written ones. I'm happy to help however I can.

HiddenRambler, 2017-08-28:
Do you by any chance have a samsung phone? The NTAG203 identification is most likely an unrelated phone issue. Would it be possible to get the dump from a different phone (of a friends perhaps)?

daleriechert, 2017-08-28:
I have a OnePlus 3 running Lineage OS. I have access to a fairly wide variety of devices if necessary.

HiddenRambler, 2017-08-28:
Thanks. Please post a dump from a different phone. We've had NTAG203 identification issue which was mostly due to phone hardware.

daleriechert, 2017-08-28:
I'm getting the same results scanning from a Nexus 6 and Google Pixel.

https://pastebin.com/raw/shGUAZh4

daleriechert, 2017-08-29:
Popped open the plastic housing and the tag PCB is entirely unmarked.

HiddenRambler, 2017-08-29:
I get the feeling that the device does not implement the tag identification method. leading the app to believe its ntag 203. I think a modified version of tagmo might work unless it needs special commands.

I've ordered a powertag. will see if anything can be done when I get it. I didn't buy a writer though so it might end up being a waste.

Bombastisch, 2017-09-11:
Any progress on the powertag? @HiddenRambler ? :) If U can implement support for it I'd would be a great improvement I think!

HiddenRambler, 2017-09-11:
[PowerTag Reset Dump.zip](https://github.com/HiddenRamblings/TagMo/files/1294195/PowerTag.Reset.Dump.zip)

HiddenRambler, 2017-09-11:
I have not released a build as of yet, a wiki page has been created with the basic usage. https://github.com/HiddenRamblings/TagMo/wiki/PowerSaves-Support

daleriechert, 2017-09-12:
On attempting to write to my two already written tags from above I get the error "No available key for P10_ID". I don't have the PowerSaves base, so will I not be able to wipe them? If PowerTag support is not enabled by default I could see others running into the same issue.

Regardless, I am able to successfully write and rewrite the same and different Amiibo interchangeably to the third tag I have without issue. I can't thank you enough for taking additional time to add this support!

HiddenRambler, 2017-09-12:
@daleriechert I'll add support for this scenario (and your two tags) very soon.

HiddenRambler, 2017-09-12:
@daleriechert please check the latest release (2.6.1)

daleriechert, 2017-09-16:
I checked and it works great on all the tags, thanks again!

closed 2017-09-16 by HiddenRambler

DaCrazyGamer, 2020-04-02:
> I checked and it works great on all the tags, thanks again!

So can I confirm Tagmo works fine on power tags with latest Update