HiddenRamblings/TagMo #31: Wont Write Tag (just doesn't do anything)

randomliam, 2017-04-10:
So I load my data and everything, then when I write tag auto, and place my nfc tag, I hear one noise and then nothing... and literally nothing happens no matter how long I wait, Please help!!

xxBigRonxx, 2017-04-12:
Try an older version.

ALTLOVESF4, 2017-07-20:
i have been trying to get this to work all day i tried all the versions i could find.
everything is setup and ready to go. but when i click wright tag (auto) it acts like its scanning but nothing happens other then a sound when the tag is under the phone i have tried a galaxy s3 and s4

Bombastisch, 2017-09-08:
Closing this issue, because it doesnt seem like I will get any response soon. If there are still any issues, open a new issue thread. Thank you! :)

closed 2017-09-08 by Bombastisch