HiddenRamblings/TagMo #32: Seems not compatible with Felica (NFC in Japan phone) ?

DenisLin, 2017-04-11:
I have a Sony Z3 compact purchased from Japan. None of the version reads ntag215.
Is this a hardware issue or software one?

morimori007, 2017-04-13:
I also tried it on two Japan mobile (Fujitsu Arrows) but could not read ntag215
Perhaps a hardware problem

Johnathan1218, 2017-05-18:
Same here, getting crash from 2.3.2 to 2.3.4
2.3.1 ok but getting back Error message saying it is not NTAG215 when I scan my amiibo
It should not be hardware problem since other app could read amiibo
My phone is Fujitsu Arrows F02G

Bombastisch, 2017-09-08:
Could you post a logcat? Just try to write, then click on the three dots at the top right corner and press "Dump Logcat". Then send the file here on github so we can try to fix it :)

Bombastisch, 2017-11-12:
Closing due to a lack of feedback, if the problem still occurs please open a new issue thread! ;)

closed 2017-11-12 by Bombastisch