HiddenRamblings/TagMo #34: TagMo doesn't see tag?

shadraak, 2017-04-17:
Trying on my nephews Samsung mega.  Keys load.  Bin loads. When I hit write auto it says to lay the phone on the tag.  And that's as far as I get.  Thought it was just his phone.  Tried on my sisters note 3. And exact same issue.  I've tried 2.3.2 and 2.3.3.  Any ideas?

xxBigRonxx, 2017-04-17:
There is a known Samsung issue.  Try an older build or build/find an .apk with the latest source.

shadraak, 2017-04-17:
I tried the two oldest ones I could find.   Guess I'll try the beta?

shadraak, 2017-04-17:
Still same issue.  Just keeps saying place phone on tag

xxBigRonxx, 2017-04-17:
Try this one.  
https://uploadfiles.io/e872p

shadraak, 2017-04-17:
What is that?

xxBigRonxx, 2017-04-17:
A tagmo .apk using the latest source.  It has the fix for Samsung.

shadraak, 2017-04-17:
Says can't open file

dtb1987, 2017-04-17:
Uninstall it from your phone and try again

shadraak, 2017-04-17:
no i mean the download wont install

shadraak, 2017-04-17:
app-fat-universal-debug.apk   Is the name and not opening.

xxBigRonxx, 2017-04-17:
Not sure why.  Works great for me.  You can watch some YouTube videos and create your own .apk from the source.

shadraak, 2017-04-17:
his samsung mega is on i believe android 4.4.2 andd i download that and try
to install and it says that it can't.

shadraak, 2017-04-17:
But is this the known Samsung issue?  I don't get any errors or anything.  It just doesn't see any tags at all

xxBigRonxx, 2017-04-17:
Yes it is a known issue. Check out the following:
https://github.com/Kidel/TagMo

shadraak, 2017-04-18:
Nothing there. Weird.

xxBigRonxx, 2017-04-18:
Look in desktop mode

ElForever, 2017-04-18:
Hmm that also happened to me, i tried with 2 different phones a Galaxy S4 and a HTC One M7, NFC obviously activated, when i tried to scan tags it didn't scan them and when i tried to write a tag it just says done without even putting the NTAG.. Wish i knew what's the issue :/

Bombastisch, 2017-09-08:
Should be fixed by now. Try to use the newest release! If there are any issues, please open a new issue thread

closed 2017-09-08 by Bombastisch