HiddenRamblings/TagMo #35: TagMo Crash on LG G5 when writing

cerealfordinner, 2017-04-17:
Just got my NTAG215 tag in the mail today and i was really exited to write some amiibos to it but so far i havent been able to write using any version of TagMo (from 2.3.0-2.3.4) or any amiibo file. When i attempt to use write(auto) on the newer versions of TagMo the app just crashed and when i use the older versions like people have advised i get a javascript error "Error while writing main data".

Any help would greatly appreciated thank you.

scottjl, 2017-04-18:
Ditto for me, LG V20 (LG-H918) running Android 7.0, stock, T-Mobile.

reno777, 2017-04-20:
Same thing happens for me on Moto z force Android 7.0 Going to try 2.3.1

jayd189, 2017-04-20:
I have an LG G5 and 2.3.1 worked for me.

smilzo, 2017-04-21:
Same with HUAWEI P9 lite with Android 7.0 TagMo 2.3.3. Don't work the "Show HEX" and "Edit SSB Data"

charedj, 2017-04-24:
Likewise, ASUS Zenfone 2 5.5" running Android 5.0.

DJTheodorson, 2017-05-02:
Newest version crashes on S6 Active. Going to try 2.3.1

xaviertidus, 2017-05-03:
Crashes also on Huawei Nexus 6P and Sony Xperia Z2 & Z3. Attempted using all 2.3.x versions

xaviertidus, 2017-05-03:
Just tested Kidel's code changes on all devices, code changes do indeed fix this issue.

https://github.com/HiddenRambler/TagMo/pull/14/commits/18548177312a3836472b9fa40bc875a72dc9c265

Bombastisch, 2017-09-08:
Should be fixed. Use the newest version!  If there are still any problems, feel free to open a new issue thread :)

closed 2017-09-08 by Bombastisch