HiddenRamblings/TagMo #38: Nothing happens when writing tag.

snq5, 2017-05-02:
When I tap "Write Tag", it says "Done" at the bottom of the screen before I scan my NFC tag. When I do scan it, it just sends me back to the main screen, leaving the NFC tag empty. I have tried the beta, the current version and the 2 versions before that.

xaviertidus, 2017-05-03:
Can you please advise your phone make, model, etc...

snq5, 2017-05-03:
Huawei P8 Lite ALE-L21

Build number is ALE-L21C432B594

Android 6.0

lindsve, 2017-05-05:
I have the same behavior on a Huawei phone I borrowed from work.

Huawei VNS-L31
Build number VNS-L31C432B130
Android 6.0

PeggerWed, 2017-05-05:
Same model, Same Phone, Same Problem

ALTLOVESF4, 2017-07-20:
i am dealing with the something galaxy s3 and s4 but i get a sound when i put the tag under the phone

HiddenRambler, 2017-07-21:
@lindsve  @PeggerWed so the latest version (2.4.3) does not work?

@ALTLOVESF4 I presume you checked both 2.3.1 and 2.4.3?

PeggerWed, 2017-07-21:
@HiddenRambler 
Sorry, since 2.3.1 worked good i didn't update to try 2.4.3
So, i did now update to try it out:
- Works perfect on all NFC phones i own :)

Thanks for fixing it, sorry for not trying it out and letting you know :/

ALTLOVESF4, 2017-07-26:
Hello first of all thank you to all you guys who made this app!!
And I got it working it ended up being a error on my part

ALTLOVESF4, 2017-07-26:
2.4.1 working on Galexsy s3

HiddenRambler, 2017-07-26:
@ALTLOVESF4 Could you tell us what caused the error so that it might help others with the same issue?

ALTLOVESF4, 2017-07-26:
Well I was going down the list of downloads and nothing seemed to work.
I ended up getting annoyed and tried one more time and 2.4.1 one just worked. So I  assume I skipped it or i had a bad download or install. If you are having  trouble getting this app to work be aware that in most cases your nfc Antenna is on the battery of your phone. And if you replaced the your battery you may not have gotten one with a Antenna. This seems to be why the  Galaxy s4 is still not working for me

closed 2017-07-28 by HiddenRambler