HiddenRamblings/TagMo #39: NFC goes wonky after installation

NuclearDeLorean, 2017-05-05:
Hello! I have a HTC One Remix with NFC running android 4.4.2 and when I load TagMo with NFC enabled/on and fully functional.  As soon as any version of TagMo gets installed, NFC now cycles on and off and I cannot use it nor turn it off unless I do a factory reset.  Then NFC returns to normal. I have installed now every version of TagMo on the site. Any thoughts?

Bombastisch, 2017-08-04:
Could you try the latest version and check if your problems are solved!

LoyalUser, 2017-09-02:
Having the exact same problem with HTC One Mini 2 also running on android 4.4.2.

Bombastisch, 2017-09-03:
Could you open the app, enable nfc and just run the app for a few seconds. then click on the three dots upper right and click "dump logcat" and post the file in here?   logcat.txt can be found in the tagmo folder of the phones base directory

Bombastisch, 2017-09-08:
I still need sonone of you to send me a logcat, since I cant replicate that bug. If we have a logcat we can most likely fix the problem easily

Bombastisch, 2017-11-12:
Closing this due to a lack of feedback.
If the issue still appears please open a new issue thread. :)

closed 2017-11-12 by Bombastisch