HiddenRamblings/TagMo #42: Crashes when writing on tag using Nexus 5X

TonySpegel, 2017-05-13:
After trying to write a bin file onto a tag the app crashes. Another app displays that there weren't any bytes written.

Bombastisch, 2017-05-14:
I have the same problem with my Nexus 5X!  In my case I use a Lineage ROM with Open Gapps, Android 7.0.1  . But if you compile some of the Pull requests the app works perfectly! But on the other side all Pull requests miss the feature of editing SSB data.

TonySpegel, 2017-05-14:
Version 2.3.2 worked for me w/o crashing :3

Bombastisch, 2017-05-14:
Have you tried to edit the SSB data of real Amiibo? This feature always crashes... Does it work on somebodys phone?

TonySpegel, 2017-05-15:
I don't own any Amiibos - I just had problems writing bin file onto Tags.

Bombastisch, 2017-05-15:
On 2.3.2 SSB data editing works perfectly! But on the newer versions SSB fails all the time...(app crashes)  Dunno what to do, the pull requests miss that feature entirely. I really hope it will be added or fixed. I love that feature!

HiddenRambler, 2017-08-01:
is this still an issue with the latest?

Bombastisch, 2017-08-01:
No more problems. App works perfectly fine!

closed 2017-08-01 by HiddenRambler