HiddenRamblings/TagMo #44: I need help with keys

Raknell, 2017-05-19:
Hi
I try to write an amiibo but the app crash, i have a Samsung Galaxy note 3. The version 2.3.2 worked with other amiibo (Super Mario SSB) but i try now and dont work nothing, when i load or scan tag all time load the Super Mario amiibo or app crash. I install 2.3.4 but i dont have the keys, Please i need help and i dont have the new keys for 2.3.4.

Bombastisch, 2017-05-23:
There are no new keys for 2.3.4! Just use the old ones. And 2.3.4 often crashes on devices. Just go back and use 2.3.2, thats the version I currently use, because the newer ones crash all the time. You can also try to compile the Pull requests, they work perfectly, but HiddenRambler didn't update them yet.

HiddenRambler, 2017-07-31:
Closing as this is a old ticket. if you still have problems with the app please open a new ticket.

closed 2017-07-31 by HiddenRambler