HiddenRamblings/TagMo #45: Crashes when I try to view hex

smokeandlights, 2017-05-23:
I can scan and load just fine, but if I try to scan an amiibo and view hex or edit ssb data, the app crashes. I haven't tried to write tags yet. Android 6.0.1, Samsung Galaxy S5 (G900V).

Bombastisch, 2017-05-23:
That problem has been discussed so often now. Version 2.3.4 & 2.3.3 are unstable and crash very often.
Go back and use 2.3.2, this version works perfectly or try to use the Pull requests, these are improved versions and are pretty bug free. HiddenRambler just didn't merge them into the master branch yet.

smokeandlights, 2017-05-24:
I'm new here, didn't know. I didn't see anything in the issues, so I posted one. 

Anyway, thanks, that fixed my "view hex" crash. Still having trouble viewing/editing ssb data.

Bombastisch, 2017-05-24:
Make sure you use a real Amiibo or Ntag, which was once used in Super Smash Bros, because the Amiibos/ NTags need to be formatted in SSB in someway. Otherwise the is no SSB-data and the app crashes.

YiweiYang, 2017-05-25:
I try to debug with this issue, found that is cause with **couldn't find "libamiitool.so"*
After fix JNI build issue, I can use my Sony Z5 to view and edit hex
And I create this patch, please try it.
["Can not found "libamiitool.so" Patch File"](https://goo.gl/vYrdSc)

smokeandlights, 2017-05-26:
Ah, that might be my issue. I don't think my amiibo has ever gotten to play with ssb.

HiddenRambler, 2017-06-16:
Could you try now with the latest version?

smokeandlights, 2017-06-18:
Just tried it. Viewing Hex works fine.

Sent from TypeApp<http://www.typeapp.com/r?b=9849>

closed 2017-06-26 by HiddenRambler