HiddenRamblings/TagMo #48: Write Tag says "done" before card touched to phone

hgelpke, 2017-06-03:
On an Axon 7 with NFC enabled. I can get up to the point where I write the tag but when I do it, it just says "done" before I do anything.

kwohlford, 2017-06-05:
Had the same problem on a nexus 6p, did a bit of messing around to try to figure it out. 

First, you can ignore the "done" message, it doesn't actually mean anything (looks like it was left in from some previous debugging). The problem is actually that it isn't detecting the NFC tag.

What fixed the issue for me was to clone this pull request: https://github.com/HiddenRambler/TagMo/pull/14
Then go into the build.gradle and change the minSdkVersion to 18. Reinstall the apk and give it a try. Hope it works for you.

HiddenRambler, 2017-06-29:
Can you tell if the latest release (2.4.1) fixes this issue?

HiddenRambler, 2017-07-14:
Closing due to the lack of feedback. If you still have the issue please try the latest build.

purpldevl, 2017-09-23:
I know this is closed, but I also have an Axon 7. It says "Done", but the prompts are translated poorly from Chinese- what it should say is "ready" and then you can hold the top of the phone to the NFC chip. The NFC antenna is right above the camera.

Bombastisch, 2017-09-23:
I think the "Done" shows up on all phones. It does on mine too.  It's probably just left over from older code.

The "Done" should show up after the Tag was succesfully written,  maybe a dev can fix it so it doesn't lead to further confusion of new users! ;)

Bombastisch, 2017-09-26:
@North101  Do you onow how to fix this?

HiddenRambler, 2017-09-26:
Just need to remove this line i think. But I don't have access to my pc this week unfortunately.

https://github.com/HiddenRamblings/TagMo/blob/49a6ae705ae9b49f52d3f9200aaa4d7eea34ecbc/app/src/main/java/com/hiddenramblings/tagmo/NfcActivity.java#L94

closed 2017-10-01 by Bombastisch