HiddenRamblings/TagMo #49: Is it possible to rewrite/format used NFC215 tags?

lnkpdr, 2017-06-08:


Bombastisch, 2017-06-08:
Nope, because if certain sectors on the Ntag storage are written, they ar not rewrite/ereasable. If you have once written a Amiibo .bin to a Ntag it is not rewriteable.

lnkpdr, 2017-06-09:
Thanks! Back to amazon to buy some more T_T

Bombastisch, 2017-06-09:
Yep, thats the thing...   But it has to be like this, because otherwise your real amiibos would be ereasable too!   I still hope for a phone emulation feature, with HCE (host card emulation), but it would need someone with knowledge in this development field.

HiddenRambler, 2017-06-16:
As @Bombastisch this is not possible due to the way Amiibo's are supposed work.

closed 2017-06-16 by HiddenRambler

alancnet, 2019-06-28:
So why is it an option in the app?

Bombastisch, 2019-06-28:
There isn't. You probably mean "Restore" which allows you to write to the data section of amiibo.

But you can't change the amiibo type