HiddenRamblings/TagMo #52: "Scan tag" won't return anything

VerpinZal, 2017-06-21:
Not sure if pre-releases are eligible for issue posting, and this is part issue / part feedback, anyway. Apologies in advance if this is not the right place.

Phone: Galaxy S4 mini with stock UI
Tagmo version 2.4.0

I previously installed 2.3.1 and non other until 2.4.0. I used to test the tags I wrote by scanning the tag and see if the name came up correctly on 2.3.1. On 2.4.0 this function doesn't work for me, just a heads up.

Bombastisch, 2017-06-21:
I don't have selfmade Amiibos lying around here. But scanning real Amiibos outputs a file! Please make sure you granted Tagmo write access to your storage and be sure where the files will be saved.

Obusmydealz, 2017-06-26:
I encountered this problem, too. In all forks, i.e. @masterchan-777 , @possi, @azsde , @Kidel  I cannot scan tags. And I also cannot scan tags from here since version 2.3.4. Version 2.3.3 is working.

For example with my Wolf Link I proceeded as follows:
I installed Version 2.3.3, scanned it, uninstalled this version and installed another version from the forks or the new 2.4.0 to load the scanned tag and to edit the TP-Link data. This worked.

So it has to be the small change which was done from version 2.3.3 to version 2.3.4.

My device is a "Asus Google Nexus 7 32 GB WiFi" with Slimkat installed (Android 4.4.4).

HiddenRambler, 2017-06-26:
Could you get a logcat output when the error occurs?

If you don't know how to take a logcat this would help:
https://www.xda-developers.com/guide-sending-a-logcat-to-help-debug-your-favorite-app/

If none of this is possible let me know and I will put in some error reporting functionality, but this will take a bit of time.

jmh363905, 2017-06-28:
LOGCAT.txt: 

06-28 11:54:15.661 D/KeyguardUpdateMonitor(2657): handleKeyguardVisibilityChanged(1)
06-28 11:54:15.721 D/AbsListView(16851): onDetachedFromWindow
06-28 11:54:15.726 D/KeyguardUpdateMonitor(2657): sendKeyguardVisibilityChanged(true)
06-28 11:54:15.731 D/KeyguardUpdateMonitor(2657): handleKeyguardVisibilityChanged(1)
06-28 11:54:15.731 D/PointerIcon(2491): setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
06-28 11:54:15.731 D/PointerIcon(2491): setMouseCustomIcon IconType is same.101
06-28 11:54:15.756 E/ViewRootImpl(16851): sendUserActionEvent() mView == null
06-28 11:54:15.756 W/InputMethodManagerService(2491): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@42c737f0 attribute=null, token = android.os.BinderProxy@427bb2d0
06-28 11:54:15.991 D/KeyguardUpdateMonitor(2657): sendKeyguardVisibilityChanged(true)
06-28 11:54:15.991 D/KeyguardUpdateMonitor(2657): handleKeyguardVisibilityChanged(1)
06-28 11:54:15.991 I/SurfaceFlinger(1776): id=1222 Removed QopupWindow (15/19)
06-28 11:54:15.996 I/SurfaceFlinger(1776): id=1222 Removed QopupWindow (-2/19)
06-28 11:54:16.001 D/KeyguardUpdateMonitor(2657): sendKeyguardVisibilityChanged(true)
06-28 11:54:16.001 D/KeyguardUpdateMonitor(2657): handleKeyguardVisibilityChanged(1)
06-28 11:54:16.626 D/InputReader(2491): Input event: value=1
06-28 11:54:16.626 D/InputReader(2491): !@notifyKey(172), action=0
06-28 11:54:16.626 D/InputManager-JNI(2491): !@interceptKeyBeforeQueueing(172), action=0
06-28 11:54:16.626 D/InputManager-JNI(2491): !@handleInterceptActions(172), action=0
06-28 11:54:16.636 D/RecentTasksLoader(2657): preloadRecentTasksList
06-28 11:54:16.636 D/STATUSBAR-RecentTasksLoader(2657): loadTasksInBackground() mstate:CANCELLED
06-28 11:54:16.636 D/STATUSBAR-RecentTasksLoader(2657): loadThumbnailsAndIconsInBackground()
06-28 11:54:16.636 D/STATUSBAR-RecentTasksLoader(2657): loadThumbnailsAndIconsInBackground-doInBackground()
06-28 11:54:16.641 D/STATUSBAR-RecentTasksLoader(2657): loadTasksInBackground -doInBackground()
06-28 11:54:16.656 D/STATUSBAR-RecentTasksLoader(2657):     recentTasks.size():7
06-28 11:54:16.681 D/STATUSBAR-RecentTasksLoader(2657): loadTasksInBackground-onProgressUpdate()
06-28 11:54:16.681 D/STATUSBAR-RecentTasksLoader(2657):  error - mRecentsPanel == null
06-28 11:54:16.686 V/STATUSBAR-RecentTasksLoader(2657): Loaded bitmap for task 32: android.graphics.Bitmap@425deb40 isPrivate? false PrivateSetting:false
06-28 11:54:16.691 D/STATUSBAR-RecentTasksLoader(2657): loadTasksInBackground-onProgressUpdate()
06-28 11:54:16.691 D/STATUSBAR-RecentTasksLoader(2657):  error - mRecentsPanel == null
06-28 11:54:16.696 V/STATUSBAR-RecentTasksLoader(2657): Loaded bitmap for task 32: android.graphics.Bitmap@435773d0 isPrivate? false PrivateSetting:false
06-28 11:54:16.696 D/STATUSBAR-RecentTasksLoader(2657): loadThumbnailsAndIconsInBackground-onProgressUpdate()
06-28 11:54:16.696 D/STATUSBAR-RecentTasksLoader(2657):     !isCancelled()
06-28 11:54:16.706 V/STATUSBAR-RecentTasksLoader(2657): Loaded bitmap for task 17: android.graphics.Bitmap@434c9510 isPrivate? false PrivateSetting:false
06-28 11:54:16.706 D/STATUSBAR-RecentTasksLoader(2657): loadThumbnailsAndIconsInBackground-onProgressUpdate()
06-28 11:54:16.706 D/STATUSBAR-RecentTasksLoader(2657):     !isCancelled()
06-28 11:54:16.746 D/dalvikvm(2657): GC_CONCURRENT freed 756K, 62% free 11122K/28776K, paused 5ms+7ms, total 48ms
06-28 11:54:16.746 D/dalvikvm(2657): WAIT_FOR_CONCURRENT_GC blocked 35ms
06-28 11:54:16.756 D/InputReader(2491): Input event: value=0
06-28 11:54:16.756 D/InputReader(2491): !@notifyKey(172), action=1
06-28 11:54:16.756 D/InputManager-JNI(2491): !@interceptKeyBeforeQueueing(172), action=1
06-28 11:54:16.756 D/InputManager-JNI(2491): !@handleInterceptActions(172), action=1
06-28 11:54:16.761 V/STATUSBAR-RecentTasksLoader(2657): Loaded bitmap for task 33: android.graphics.Bitmap@437d80a0 isPrivate? false PrivateSetting:false
06-28 11:54:16.761 D/STATUSBAR-RecentTasksLoader(2657): loadThumbnailsAndIconsInBackground-onProgressUpdate()
06-28 11:54:16.761 D/STATUSBAR-RecentTasksLoader(2657):     !isCancelled()
06-28 11:54:16.761 D/VoIPInterfaceManager(2491): isVoIPRinging()...
06-28 11:54:16.761 D/VoIPInterfaceManager(2491): isVoIPRunningAndDeregi()...
06-28 11:54:16.761 D/VoIPInterfaceManager(2491): Not exist call session
06-28 11:54:16.766 I/WindowManager(2491): startDockOrHome
06-28 11:54:16.766 D/PowerManagerService(2491): [api] [s] userActivity : event: 0 flags: 0 (uid: 1000 pid: 2491) eventTime = 939136931
06-28 11:54:16.766 D/RecentTasksLoader(2657): cancelPreloadingRecentTasksList
06-28 11:54:16.766 D/STATUSBAR-RecentTasksLoader(2657): cancelLoadingThumbnailsAndIcons()
06-28 11:54:16.766 D/STATUSBAR-RecentTasksLoader(2657): mTaskLoader.cancel()
06-28 11:54:16.766 D/STATUSBAR-RecentTasksLoader(2657): mThumbnailLoader.cancel()
06-28 11:54:16.771 D/STATUSBAR-AirplaneModeQuickSettingButton(2657): onReceive : android.intent.action.CLOSE_SYSTEM_DIALOGS
06-28 11:54:16.776 W/ActivityManager(2491): mDVFSHelper.acquire()
06-28 11:54:16.786 D/PointerIcon(2491): setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
06-28 11:54:16.786 D/PointerIcon(2491): setMouseCustomIcon IconType is same.101
06-28 11:54:16.791 D/WfcActivityStateReceiver(2935): onReceive(): packageName=com.pluscubed.matlog, activityState=pause
[... cut ...]
06-28 11:54:28.241 I/InputReader(2491): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.807 ] when=939148409638000
06-28 11:54:28.246 I/InputDispatcher(2491): Delivering touch to: action: 0x4, toolType: 1
06-28 11:54:28.246 I/InputDispatcher(2491): Delivering touch to: action: 0x0, toolType: 1
06-28 11:54:28.301 I/InputReader(2491): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=939148467166000
06-28 11:54:28.301 I/InputDispatcher(2491): Delivering touch to: action: 0x1, toolType: 1

jmh363905, 2017-06-28:
Device_Info.txt: 

build.board: universal3470
build.bootloader: T399NUVSAQA1
build.brand: samsung
build.cpu_abi: armeabi-v7a
build.cpu_abi2: armeabi
build.device: gardalteMetroPCS
build.display: KOT49H.T399NUVSAQA1
build.fingerprint: samsung/gardalteMetroPCS/gardalteMetroPCS:4.4.2/KOT49H/T399NUVSAQA1:user/release-keys
build.hardware: universal3470
build.host: SWDD3916
build.id: KOT49H
build.manufacturer: samsung
build.model: SGH-T399N
build.product: gardalteMetroPCS
build.radio: unknown
build.serial: 300461d62f164299
build.tags: release-keys
build.time: 1483526291000
build.type: user
build.user: dpi
version.codename: REL
version.incremental: T399NUVSAQA1
version.release: 4.4.2
version.sdk_int: 19

jmh363905, 2017-06-28:
That is a recording from the moment I pressed the play button (after having cleared previous logs), then opened tagmo, scanned without any info being scanned in, and back to matlog to press pause and send the log.  I'm a total android noob, so if there was a more streamlined way to give this data to you, I'm sorry.  But I had the exact same issues as the other posters.  After 2.3.3, it won't scan. Wont crash, but acts like it did nothing at all.

jmh363905, 2017-06-28:
https://youtu.be/r1LQmcmke1I for a video of the process with actual amiibo.

HiddenRambler, 2017-06-28:
Thanks for that detailed report.

From the look of the LogCat it looks like the tag was in fact scanned:

06-28 11:54:23.316 D/TagWriter(16930): 047BD225F2484D8077480FE0F110FFEEA5000100A1A4AC09DC818899BA4EF850F81882D57ED29ED7DEB766D6B9A5191F9C7F0713966A7156ACCC9F7E09A9AE41D3DE88B74AA96C0AD3BEA0BB732D6419896F3F0D01010000035609020D12C61776268F10F9FAD28C37C9FFA24F37FD1F092A27E1F7F8D96EF198DABA965720E078FA19214E50551B003E6797BCB4319212DF62786B580093C636393D98389F82546FE3D42BF0935BC90EB3EAD299E60F6B9578D3E0E0647C94A5E5CDA1CCA9E9F6BCADE30020706C8A0401E5E4ACAA8A66F84B01F30329DB18B1EECF93BD6F3C93DB08F25E9181799E0BF2DF27558CE5549B37E3739E9A3364502BAB4D3F9C3A6F296F48125CB51538631E58E24621DEE07AD39D8F265A1B67DD3F72264C61E03993D733169CE5BE87CC9925C849D740A59968CF42262DCF6AAE4790CC32650137A66DEF6A7D4415FCAC76C1791A443D2E552821E24B268915FBC183A2D44B11685D8C73071EFE8DEA30DA1A71A367C4D129259402A9735D7D1D6DA2997B5EF412B9689DB8A66565DA8684E3301D33AEE78531E9EE78A40EE62C31A5B674A5DA508377E15BCAE726BE510D583E5A3BFA1E56297D2FF5F755D8C0D5CE4083078E8CD2F2AE5C1EEFED79E40B070B0FD3F1A9E9DCBC94EDF47CA69B8FE1A94ACF4DEECE336F6DE3DB715445E7E769C8B46589D9F84CB28756AFDDC01182C968CE3ADC8BEA09F986284801000FBD000000045F000000

Thats the actual data from your tag. But for some reason the app didnt go through process of saving it. It might be that turning the phone around and keeping it on the desk put the phone into sleep mode/keyguard mode. Could you try to do it without while holding phone with screen up and making sure to not cover the proximity sensor (near the ear-spearker).

If that doesn't work i'll build a version with more logging to help track down the error.

jmh363905, 2017-06-28:
I only put it face down because was holding phone in other hand. When I do it properly I still get same results. But I can if you need to see it to believe it. Will be a little later as that was on my lunch break and I'm not at home at the moment. Anything else I can help with? Video or log of 2.3.3 ? (last one that worked)

jmh363905, 2017-06-28:
https://youtu.be/vrXtt8O6QB8 There ya go.  Three different amiibos.  Didn't lay phone down.

HiddenRambler, 2017-06-29:
Thanks for that. Looks like the scan activity is not passing the message back to the main activity. Could you try this version and send me the log file. Its exactly the same as current version but logs whats going on between the activitites.
[TagMo-Debug.zip](https://github.com/HiddenRambler/TagMo/files/1111171/TagMo-Debug.zip)

Thanks for your help debugging this.

jmh363905, 2017-06-29:
I have the debug version installed and have scanned a couple.  Where are the logs located?  Total Android noob here.  iPhone user that got this phone just for TagMo, but will help any way I can!

HiddenRambler, 2017-06-29:
Sorry I meant LOGCAT.
Do the same thing you did couple of posts ago to get that long log text.

jmh363905, 2017-06-29:
Will do. Want me to scan just one to keep the log shorter (as if logs ever are short) or do a few to get more data?

jmh363905, 2017-06-29:
And do you have a PayPal account for donations?

HiddenRambler, 2017-06-29:
Yes just one tag is enough.

Thanks but this is a hobby no donations needed.

jmh363905, 2017-06-29:
build.board: universal3470
build.bootloader: T399NUVSAQA1
build.brand: samsung
build.cpu_abi: armeabi-v7a
build.cpu_abi2: armeabi
build.device: gardalteMetroPCS
build.display: KOT49H.T399NUVSAQA1
build.fingerprint: samsung/gardalteMetroPCS/gardalteMetroPCS:4.4.2/KOT49H/T399NUVSAQA1:user/release-keys
build.hardware: universal3470
build.host: SWDD3916
build.id: KOT49H
build.manufacturer: samsung
build.model: SGH-T399N
build.product: gardalteMetroPCS
build.radio: unknown
build.serial: 300461d62f164299
build.tags: release-keys
build.time: 1483526291000
build.type: user
build.user: dpi
version.codename: REL
version.incremental: T399NUVSAQA1
version.release: 4.4.2
version.sdk_int: 19

06-29 11:25:30.739 I/InputReader(2488): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.11 ] when=248829199000
06-29 11:25:30.744 I/InputDispatcher(2488): Delivering touch to: action: 0x4, toolType: 1
06-29 11:25:30.744 I/InputDispatcher(2488): Delivering touch to: action: 0x0, toolType: 1
06-29 11:25:30.799 I/InputReader(2488): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=248886546000
06-29 11:25:30.799 I/InputDispatcher(2488): Delivering touch to: action: 0x1, toolType: 1
06-29 11:25:30.864 D/AbsListView(6160): Get MotionRecognitionManager
06-29 11:25:30.869 D/PointerIcon(2488): setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
06-29 11:25:30.869 D/PointerIcon(2488): setMouseCustomIcon IconType is same.101
06-29 11:25:30.904 I/SurfaceFlinger(1775): id=25 createSurf (1x1),1 flag=4, QopupWindow
06-29 11:25:30.904 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:30.904 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:30.914 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:30.914 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:30.919 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:30.919 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:30.979 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:30.979 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:30.999 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:30.999 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:31.264 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:31.264 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:31.689 I/InputReader(2488): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.12 ] when=249777289000
06-29 11:25:31.689 I/InputDispatcher(2488): Delivering touch to: action: 0x4, toolType: 1
06-29 11:25:31.689 I/InputDispatcher(2488): Delivering touch to: action: 0x0, toolType: 1
06-29 11:25:31.744 I/InputReader(2488): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=249833935000
06-29 11:25:31.744 I/InputDispatcher(2488): Delivering touch to: action: 0x1, toolType: 1
06-29 11:25:31.829 D/AbsListView(6160): onDetachedFromWindow
06-29 11:25:31.834 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:31.834 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:31.834 D/PointerIcon(2488): setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
06-29 11:25:31.834 D/PointerIcon(2488): setMouseCustomIcon IconType is same.101
06-29 11:25:31.849 E/ViewRootImpl(6160): sendUserActionEvent() mView == null
06-29 11:25:31.849 W/InputMethodManagerService(2488): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@42a99780 attribute=null, token = android.os.BinderProxy@427fd248
06-29 11:25:32.099 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:32.099 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:32.099 I/SurfaceFlinger(1775): id=25 Removed QopupWindow (9/11)
06-29 11:25:32.104 I/SurfaceFlinger(1775): id=25 Removed QopupWindow (-2/11)
06-29 11:25:32.109 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:32.109 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:33.029 I/InputReader(2488): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.13 ] when=251117876000
06-29 11:25:33.029 I/InputDispatcher(2488): Delivering touch to: action: 0x4, toolType: 1
06-29 11:25:33.029 I/InputDispatcher(2488): Delivering touch to: action: 0x0, toolType: 1
06-29 11:25:33.144 I/InputReader(2488): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=251232943000
06-29 11:25:33.144 I/InputDispatcher(2488): Delivering touch to: action: 0x1, toolType: 1
06-29 11:25:33.419 D/SSRMv2:SIOP(2488): SIOP:: AP = 280, Delta = 10
[... 630 lines cut ...]
06-29 11:25:47.099 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:47.114 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:47.414 D/CustomFrequencyManagerService(2488): releaseDVFSLockLocked : Getting Lock type frm List : DVFS_MIN_LIMIT  frequency : 1200000  uid : 1000  pid : 2488  tag : ACTIVITY_RESUME_BOOSTER@5
06-29 11:25:47.414 D/KeyguardUpdateMonitor(2655): sendKeyguardVisibilityChanged(true)
06-29 11:25:47.414 D/KeyguardUpdateMonitor(2655): handleKeyguardVisibilityChanged(1)
06-29 11:25:47.414 I/SurfaceFlinger(1775): id=34 Removed Mauncher (8/13)
[... cut ...]
06-29 11:25:47.979 I/InputReader(2488): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.20 ] when=266066509000
06-29 11:25:47.979 I/InputDispatcher(2488): Delivering touch to: action: 0x4, toolType: 1
06-29 11:25:47.979 I/InputDispatcher(2488): Delivering touch to: action: 0x0, toolType: 1
06-29 11:25:48.059 I/InputReader(2488): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=266146900000
06-29 11:25:48.059 I/InputDispatcher(2488): Delivering touch to: action: 0x1, toolType: 1

jmh363905, 2017-06-29:
There you go. I deleted all data for the previous installed version and uninstalled it before installing the -debug build, then for good measure restarted device before getting this log.

HiddenRambler, 2017-06-29:
Thanks. It pointed at a possible issue, with how the data is passed back to the parent. I have put in a fix and also included more debug information. Try this new version and see. if it doesnt work could you give me another logcat.

[TagMo-Debug2.zip](https://github.com/HiddenRambler/TagMo/files/1112956/TagMo-Debug2.zip)

jmh363905, 2017-06-29:
Still isn't working.  Logcat output follows:

build.board: universal3470
build.bootloader: T399NUVSAQA1
build.brand: samsung
build.cpu_abi: armeabi-v7a
build.cpu_abi2: armeabi
build.device: gardalteMetroPCS
build.display: KOT49H.T399NUVSAQA1
build.fingerprint: samsung/gardalteMetroPCS/gardalteMetroPCS:4.4.2/KOT49H/T399NUVSAQA1:user/release-keys
build.hardware: universal3470
build.host: SWDD3916
build.id: KOT49H
build.manufacturer: samsung
build.model: SGH-T399N
build.product: gardalteMetroPCS
build.radio: unknown
build.serial: 300461d62f164299
build.tags: release-keys
build.time: 1483526291000
build.type: user
build.user: dpi
version.codename: REL
version.incremental: T399NUVSAQA1
version.release: 4.4.2
version.sdk_int: 19

06-29 12:33:45.229 D/dalvikvm(6205): GC_CONCURRENT freed 1112K, 20% free 7561K/9348K, paused 1ms+1ms, total 16ms
06-29 12:33:45.244 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=FACTORY_TEST_APP
06-29 12:33:45.244 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=FACTORY_TEST_COMMAND
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=MODEL_NAME
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=MODEL_NUMBER
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=MODEL_HARDWARE_REVISION
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=MODEL_COMMUNICATION_MODE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=CHIPSET_MANUFACTURE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=CHIPSET_NAME
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=DEVICE_TYPE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=PANEL_TYPE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SENSOR_NAME_ACCELEROMETER
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SENSOR_NAME_MAGNETIC
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SENSOR_NAME_GYROSCOPE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=REAR_CAMERA_TYPE
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SPEAKER_COUNT
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=MIC_COUNT
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=FTA_HW_VER
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=FTA_SW_VER
06-29 12:33:45.249 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_VIBRATOR
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_LTE
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_QWERTY_KEY
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_FRONT_CAMERA
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_RCV
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=FACTORY_TEST_APO
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_BOOST_MEDIASCAN
06-29 12:33:45.254 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=NEED_LPA_MODE_SET
06-29 12:33:45.259 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SYS_INFO_UPDATE
06-29 12:33:45.259 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SUPPORT_SECOND_MIC_TEST
06-29 12:33:45.259 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=NEED_NOTI_AUDIO_MANAGER
06-29 12:33:45.259 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=CAL_AP_TEMP
06-29 12:33:45.259 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=SETUPWIZARD_INDEX_LIST
06-29 12:33:45.264 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_VOLUME_UP
06-29 12:33:45.279 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_VOLUME_DOWN
06-29 12:33:45.279 D/Launcher.HomeView(2884): onStop
06-29 12:33:45.284 D/Launcher(2884): onTrimMemory. Level: 20
06-29 12:33:45.294 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_POWER
06-29 12:33:45.309 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_MENU
06-29 12:33:45.319 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_HOME
06-29 12:33:45.334 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_BACK
06-29 12:33:45.349 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_SEARCH
06-29 12:33:45.364 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_FOCUS
06-29 12:33:45.374 D/FactoryTestApp(6205): [XMLDataStorage$redefinitionById] id=KEY_TEST_CAMERA
[... 661 lines cut ...]
06-29 12:33:58.704 D/KeyguardUpdateMonitor(2653): handleKeyguardVisibilityChanged(1)
06-29 12:33:58.709 D/KeyguardUpdateMonitor(2653): sendKeyguardVisibilityChanged(true)
06-29 12:33:58.714 D/KeyguardUpdateMonitor(2653): sendKeyguardVisibilityChanged(true)
06-29 12:33:58.719 D/KeyguardUpdateMonitor(2653): handleKeyguardVisibilityChanged(1)
06-29 12:33:58.719 D/KeyguardUpdateMonitor(2653): handleKeyguardVisibilityChanged(1)
06-29 12:33:58.719 D/KeyguardUpdateMonitor(2653): handleKeyguardVisibilityChanged(1)
[... cut ...]
06-29 12:33:59.684 I/InputReader(2486): Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.44 ] when=151008342000
06-29 12:33:59.684 I/InputDispatcher(2486): Delivering touch to: action: 0x4, toolType: 1
06-29 12:33:59.684 I/InputDispatcher(2486): Delivering touch to: action: 0x0, toolType: 1
06-29 12:33:59.779 I/InputReader(2486): Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=151099958000
06-29 12:33:59.779 I/InputDispatcher(2486): Delivering touch to: action: 0x1, toolType: 1

jmh363905, 2017-06-29:
Question: Do I have to send the following command every time I restart my device or is it a once and done thing?  The reason I ask is if I don't have to each restart (as I have been doing), I can take the stuff with me that I need and can be more reactive (times when I'm away from my laptop).

The command in question: adb shell pm grant com.pluscubed.matlog android.permission.READ_LOGS

HiddenRambler, 2017-06-29:
You have to send it ever time you restart it I think. but you don't really need to restart the phone. You can just remove and install the new version without restarting the phone.

I'm trying to figure out why this is hapenning. as far as i can see i'm doing everything as it should be done.

HiddenRambler, 2017-06-29:
Could you try this one?
[TagMo-Debug3.zip](https://github.com/HiddenRambler/TagMo/files/1113130/TagMo-Debug3.zip)

jmh363905, 2017-06-29:
That fixed it! Do you need the log still? I have the recording saved

HiddenRambler, 2017-06-29:
No need for the log, I understand the issue.

It was put in place to fix a different bug. I suppose we should try to find a proper fix for the original issue.

Thanks for you help in tracking this down.

jmh363905, 2017-06-29:
Glad to be of help. Is this build safe to use or should I revert back to 2.3.3. You mentioned the issue was born of another bugs squashing. I didn't know if said bug is back now in this build or if it's a show stopper. Any light to shed on what functions to avoid in this build to prevent said bug from presenting itself?

HiddenRambler, 2017-06-29:
It's safe to use this build. It just reverts one previous commit and adds extra logging.

The problematic commit was: 
https://github.com/HiddenRambler/TagMo/commit/bfaf0e43b8f9480282044bcfa8af774e74407b70

It was to fix a crash when trying to scan amiibo. It was only for a few people so if you didnt get it already you should be ok.

Could you tell me the your android version and phone model. I can figure out the best solution.

jmh363905, 2017-06-29:
SGH-T399N. Version 4.4.2. Kernel 3.4.39-2903595. Build KOT49H.T399NUVSAQA1. 

Logcat device info:

build.board: universal3470
build.bootloader: T399NUVSAQA1
build.brand: samsung
build.cpu_abi: armeabi-v7a
build.cpu_abi2: armeabi
build.device: gardalteMetroPCS
build.display: KOT49H.T399NUVSAQA1
build.fingerprint: samsung/gardalteMetroPCS/gardalteMetroPCS:4.4.2/KOT49H/T399NUVSAQA1:user/release-keys
build.hardware: universal3470
build.host: SWDD3916
build.id: KOT49H
build.manufacturer: samsung
build.model: SGH-T399N
build.product: gardalteMetroPCS
build.radio: unknown
build.serial: 300461d62f164299
build.tags: release-keys
build.time: 1483526291000
build.type: user
build.user: dpi
version.codename: REL
version.incremental: T399NUVSAQA1
version.release: 4.4.2
version.sdk_int: 19

jmh363905, 2017-06-29:
I think it's noble to make this available for free. You should setup a PayPal account for donations. There's folks that appreciate your efforts that much

Bombastisch, 2017-06-29:
Hey! Congratulations! I just saw that U made the app availible for 3Ds! Thats awesome, hopefully this app won't be stopped by this step.

HiddenRambler, 2017-06-29:
I've release  2.4.1 with the fix for this issue.

jmh363905, 2017-06-29:
Works beautifully.  And I love being able to be on current version where all my new AMIIBOs are named something besides "Mario (LOZ)". By chance if any AMIIBOs I scan come up mis-identified or straight up unidentified, is there a page you have set up to get the info needed to implement for subsequent releases for these mis-named AMIIBOs?  (again, none of mine are showing up incorrect now, but as Nintendo releases new ones... you know... ;D )

jmh363905, 2017-06-29:
I don't know if this is something you've thought about, but is there any way to make the app, upon launch, attempt to update that AMIIBO naming data by reaching out to a server where the raw info is stored in a text file or something perhaps that it can then parse, keeping that data updated between releases?  I have a MacOS running the Server app and a totally un-utilized web service.  I could, working with you, set it up to be a backup or even primary server for hosting that.  Just a thought that probably doesn't belong in this thread, so after this post I'll let it die or at very least leave it open to get back on topic.

HiddenRambler, 2017-06-30:
For now if you run across any unknown / incorrectly identified amiibo's raise a ticket here. I do have plans to make the app load the amiibo list seperately so it can maintained seperately. Specially now that i need to implement similar list for Thenaya app.

I'm closing this ticket as it's finished. please open a new one for any other issues/requests.

closed 2017-06-30 by HiddenRambler