HiddenRamblings/TagMo #53: TagId : 2F7868746D6C312F/3C214443545950

Midnax, 2017-07-02:
I have a Samsung Galaxy S3 neo with android 4.4.2. I tried all the Tagmo versions there are. But non of them work. The latest version won't open the amiibo bin files correctly. Instead of giving the name of the file at TagId , with every file I try it'll say TagId : 2F7868746D6C312F/3C214443545950 ...As a result , whenever I click WRITE TAG (AUTO) it won't transfer the file to the ntag215. It will say done before you hold the tag against your phone. Any way I can fix this?

HiddenRambler, 2017-07-02:
Very strange. Could you post the first 12 bytes of one of you files for me? You can see them using a hex editor like HxD https://mh-nexus.de/en/hxd/

Please only post the first 12 bytes as posting whole file might cause copyright issues.

Also an Logcat if you can get one would be good.

Midnax, 2017-07-03:
I don't think the problem has to do the amiibo bin files I downloaded cuz I downloaded them from https://nfc-bank.com/bins.php?s=f09be7d3296182a282b399e9e1bf1ba3 . It's a popular site to get amiibo bin files from and it seems to work for other people. But when I tried multiple files from different people on this site , non work. I think the problem has to do with my android 4.4.2 kitkat. When I load any file in Tagmo it changes the file to something it cannot read and all files get the same TagId : 2F7868746D6C312F/3C214443545950 . My guess is that this some kind of error code.

twokegg, 2017-07-07:
I am having the same issue. I am running on a Droid Maxx 2 on android 5.1.1. the only difference is it simply says  "failedto decrypt tag" when I use write tag (auto) and put the tag up to the phone.

HiddenRambler, 2017-07-08:
@Midnax i think you downloads are corrupt. 3C214443545950 in ascii is "<!DCTYP" and is 2F7868746D6C312F "/xhtml1/" both of which indicate that its an html page. 

I think whatever download link you used gave you an html page rather than an actual bin (perhaps an error occured?) either way your bin files are invalid. I suggest you download them to your pc and transfer them to the phone.

If you still don't believe me rename whatever bin file which is not working to an .html and open it in your browser.

closed 2017-07-08 by HiddenRambler