HiddenRamblings/TagMo #54: Stopped working on Moto Droid Turbo 2 after OS upgrade

jmonty, 2017-07-06:
Was running TagMo 2.3.2 fine on older Android OS (I think it was version 6.1). Then upgraded phone to Android 7.0. TagMo stopped working. Tried a bunch of other versions of TagMo (up to 2.4.1) but they all just can't read/write. None of the versions crash, it just seems TagMo cannot "see" the NFC tags now. OS NFC setting is (still) enabled.

Bombastisch, 2017-07-07:
I guess HiddenRambler could help much more with a LogCat. 

> Could you get a logcat output when the error occurs?
> If you don't know how to take a logcat this would help:
> https://www.xda-developers.com/guide-sending-a-logcat-to-help-debug-your-favorite-app/

Then solving the problem is probably much easier!

HiddenRambler, 2017-07-08:
Indeed a log cat would make life a lot easier for me. I will try to check the app on Anrdoid 7 but that will take a bit of time.

Bombastisch, 2017-07-08:
This issue is phone specific. I use Android 7.1.2 and I have no issues at all!

HiddenRambler, 2017-07-08:
Just tried on LG G6 (android 7.0) and it works fine. Makes me think that it is indeed a problem with your specific phone. Can you post your phone model. Also could you try using an App like NXP TagInfo to see if it can read the tag? https://play.google.com/store/apps/details?id=com.nxp.taginfolite

Ultimately without a logcat i don't think i can do much.

HiddenRambler, 2017-07-14:
Can you try the latest build (2.4.3)

jmonty, 2017-07-15:
I tried 2.4.3 but had the same result. I got a logcat this time (from app startup up to attempting to scan a tag). Seems like a lot to copy/paste here. What is the preferred method for providing such info?

jmonty, 2017-07-15:
Just noticed the file attach feature. Trying that now...

[device_info.txt](https://github.com/HiddenRambler/TagMo/files/1150473/device_info.txt)
[logcat.txt](https://github.com/HiddenRambler/TagMo/files/1150474/logcat.txt)

HiddenRambler, 2017-07-15:
Thanks for the log, but it doesnt have any of the debug messages i would expect to see. What did you use to get this logcat?

There are lots of errors from an app called DT Ignite. which you might want to disable: https://www.androidcentral.com/everything-you-need-know-about-dt-ignite

jmonty, 2017-07-16:
I used the log method recommended by bombastich above (matlog). Is there some other method you'd like me to use? I'm technical and willing to help.

HiddenRambler, 2017-07-16:
That is the method i'd normally recommend, but looks like it doesnt work. I've made a quick version of TagMo with an option to to generate a logcat. Could you install the attached version attempt to scan a tag and then choose "Dump Logcat (for bug reporting)" option from the app menu (top right hand corner icon).

This wil create a file called "tagmo_logcat.txt" in the amiibo folder, which you can attach to a comment here.
[TagMoTest3(SingleTop).zip](https://github.com/HiddenRambler/TagMo/files/1150949/TagMoTest3.SingleTop.zip)

jmonty, 2017-07-16:
Did all that, but didn't seem to log much! What I did:
1) Uninstalled existing TagMo
2) Installed new test version.
3) Ran new one, selected Scan Tag. Tried to scan but nothing happened in the scan view, so after several seconds waiting for the scan, hit the "back" button.
4) Selected the "Dump Logcat" option.
5) Retrieved file and here we are.
[tagmo_logcat.txt](https://github.com/HiddenRambler/TagMo/files/1151418/tagmo_logcat.txt)

HiddenRambler, 2017-07-17:
According to the log nothing is actually working, or rather i get the feeling that the phone is killing the tagmo process all the time. This matches with what i saw with the earlier log but I thought that it was a mistake.

To be honest i don't know what is going on here. This is what i would check if i were you:

1) Make sure there are not task managers running. (Tools which kill idle tasks)
2) Disable Ignite (see my earlier comment) it seems to be doing some task related things.
3) Free up more ram so android wont try allways kill processes to free up memory. (Temporarily disable remove apps like facebook twitter).

I will keep checking to see if i can reproduce this on another phone, but i'm kind of lost at the moment.

jmonty, 2017-07-18:
Not really sure what else I can try. On your three points above:
1) I never installed another task manager, so I don't think something unexpected can be running that kills off idle tasks. Is there some way to check for something else?
2) I did actually disable Ignite after you mentioned it, and so it was already disabled when I tried your latest test version above.
3) I don't have a bunch of stuff running on this phone. I only use it for playing around with Android, not as my main phone, so it doesn't even have apps like Facebook nor Twitter on it.
BTW, I just realized that during the scan attempt, a noise is produced (I might have had the sound off before). So even though the scan is not successful, and gives no indication of success on screen, *something* is noticing the NFC tag and making a sound every time I bring the tag close to the phone while in the scan view. Perhaps this clue helps. BTW, I'm willing to iterate on debug versions if you have interest in trying more. I really like TagMo and want to use it more!

Bombastisch, 2017-07-18:
"Also could you try using an App like NXP TagInfo to see if it can read the tag? https://play.google.com/store/apps/details?id=com.nxp.taginfolite "

As Hiddenrambler wrote before, are you sure your NFC is working? Because I'm not surr if this "sound" you hear is normal. Just try to scan a amiibo with tha app linked before.

jmonty, 2017-07-18:
Ok, Bombastisch's message lead to my finding a "work around" that I can't fully explain! But here's what happened.

I tried to use the "NXP TagInfo" app but it had the same result. But this lead me to looking again at the Android NFC settings. There are two settings (the main NFC on/off and the "NFC Security Reminders" checkbox). I ended up turning them both off and then turning the main NFC back on, and then things started working. So I thought it must be the "NFC Security Reminders" setting because it was still off, right? But then I turned that setting back on and things are still working!?! So all I can say is that a work-around might be to toggle the Android NFC settings to see if that clears things up. 

Thanks for all of the attention and sorry that it ended up being something (apparently) not app-related and weird. Closing this issue.

closed 2017-07-18 by jmonty

Bombastisch, 2017-07-18:
Okay seems like it's solved. This is probably a common issue on samsung phones, because there are more issuse like that. 

I dont even have this option in my NFC settings. 

I'm not sure if this setting  could be turned off by the app itself, but it would probably help some samsung phone users

HiddenRambler, 2017-07-18:
Wow, never seen that option on any android phone.

jmonty, 2017-07-18:
Just to clarify, this is a Motorola Droid Turbo 2, running Android 7.0.  I attached a screenshot of the NFC settings for possible future reference.
[image]