HiddenRamblings/TagMo #55: Samsung Galaxy S6 Edge Plus Wont Write

Murkdeth, 2017-07-13:
Ive tried every build of Tagmo and still havent been able to get past the error message it says done before the scan starts and gives an error message when it tries to write

aruur, 2017-07-13:
Same issue here, with my Nexus 6p

HiddenRambler, 2017-07-14:
Could one of you please give me a logcat output. it would help greatly. Without one i have no way to reproduce this issue.

HiddenRambler, 2017-07-14:
Could you check the latest version (2.4.3) hoping it fixed this issue as well.

Murkdeth, 2017-07-17:
How do I retrieve the logcat output for you? 2.4.3 didn't work either unfortunately I really appreciate your help and effort though thanks!

HiddenRambler, 2017-07-17:
You can try the test version in this comment: https://github.com/HiddenRambler/TagMo/issues/54#issuecomment-315600206

HiddenRambler, 2017-07-18:
Can you check the solution that worked for #54

RRMalone, 2017-07-27:
I have the normal GS6 Edge and it will not work on my phone, I've used every version and every file manager. I tried to just use my PowerSave and unfortunately there's not a clear way to write using that.

HiddenRambler, 2017-07-31:
@RRMalone, What exactly is happenning? do you see an error?

RRMalone, 2017-07-31:
It just says failed, I downloaded the right files and everything but it just won't work.

Sent from my U.S. Cellular® Smartphone

HiddenRambler, 2017-08-01:
Please:
1) Use the latest version (new release as of yesterday)
2) Do a scan tag and scan an existing tag.
3) Generate a logcat (there is a menu option for it)
4) Attach the generated logcat txt file to the ticket.

Murkdeth, 2017-08-02:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/1194453/tagmo_logcat.txt)

Murkdeth, 2017-08-02:
Hi sorry for the late reply I used the most recent version 2.5 to generate the logcat. Im still running into the same issue as before on this build. Thanks for your time I really appreciate this project

HiddenRambler, 2017-08-02:
The error in the log looks similar to what one gets when wrong type of tags are used. 

Could you install https://play.google.com/store/apps/details?id=com.nxp.taginfolite app and scan your tag with that and tell us if it clearly says NTAG215 in the app or something different like NTAG213/NTAG216.

RRMalone, 2017-08-02:
It's definitely not the tags, they're 215's. With this many not working on the galaxy's it has to be the app. I used actual Amiibo, copies and downloaded tags.

Sent from my U.S. Cellular® Smartphone

HiddenRambler, 2017-08-02:
Actually a lot of of complaints on galaxy was resolved. There was a recent ticket where the guy recieved mixed tags from a seller with both 215's and 216.

RRMalone, 2017-08-02:
I didn't say Galaxy's won't work, I said there's a lot of people saying it doesn't work and most the ones I see are Galaxy's!
That would be a problem with the tags though, not the phone or app.
There were a few people that said the newer ones cut out a piece of code but u can't even view mine, nothing on it will work.

Sent from my U.S. Cellular® Smartphone

Murkdeth, 2017-08-02:
[image]

HiddenRambler, 2017-08-02:
That indeed looks like the correct tag type. Just to check one more error condition could you do a full scan on the nxp app and send a screenshot of the hex dump that is shown ("Memory Content")

Or after the full scan you can use the "share" option to export the xml/text of the output.

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
[image]

Murkdeth, 2017-08-02:
Hope that's the right thing

Murkdeth, 2017-08-02:
i bought the tags off aliexpress i guess they sent me bum tags. I only tried to write to one of the 15 but unfortunately the all show the lock and the 0x00 bytes so i guess im screwed.. lol I appreciate your help anyways thanks man

RRMalone, 2017-08-03:
> If you look at that screen the tag is write protected. meaning somehow at somepoint (all pages are +r and *r). This is the reason why tagmo is failing. It could be that a previous write operation failed somehow, but tagmo would never leave an entire tag filled with 0x00 bytes. Regardless of how it happened this particular tag can no longer be used.
>
> Please check and make sure that other tags you have do not have the same issue and try using one of them.

I ordered some through AliExpress too but mine are unlocked and have 540

Sent from my U.S. Cellular® Smartphone

RRMalone, 2017-08-03:
504*

Sent from my U.S. Cellular® Smartphone

HiddenRambler, 2017-08-03:
@RRMalone still waiting for the reply to my previous post.

Please:

    Use the latest version (new release as of yesterday)
    Do a scan tag and scan an existing tag.
    Generate a logcat (there is a menu option for it)
    Attach the generated logcat txt file to the ticket.

Bombastisch, 2017-09-08:
Will we get any response? Help is appreciated ;)

RRMalone, 2017-09-08:
I just used my GS3 and it works fine. My 6 edge won't work. Apparently it cuts out part of the code. I saw done images online and one of the numbers kept not showing up on Android 7.0

Sent from my U.S. Cellular® Smartphone

HiddenRambler, 2017-09-08:
@RRMalone Could you submit a debug log? I'm sure we could get it to work. I've tested with mulitple 7.0 androids and it works.

@Bombastisch Welcome back ;)

Bombastisch, 2017-09-09:
Haha, I am still in Cincinnati, OH for the next to weeks, before I fly back to Germany. But I'll just moderate the issue thread from here! ;)

RRMalone, 2017-09-09:
I tried to a couple times a few weeks ago but I can try again and see if it'll let me.

Sent from my U.S. Cellular® Smartphone

Bombastisch, 2017-09-11:
Help is appreciated! ;)

RRMalone, 2017-11-12:
I finally got the GS6 to work using an old version (4.2) and now I can't get the S8 to load the keys and I tried every single version multiple times.

Bombastisch, 2017-11-12:
This is an old thread now.
Could you please describe your problem again and include a logcat? 
I am glad we finally hear something on this thread again.

Bombastisch, 2017-12-19:
Closing due to a lack of feedback. If your problem still occurs please open a new issue thread. We developers are just simple persons. We don't own every kind of phone on this earth. If something isn't working correctly we need your help to work closely with us together to fix this problem for all of us. Thank you. :)

closed 2017-12-19 by Bombastisch