HiddenRamblings/TagMo #56: Unable to write tags

MercilessOtter, 2017-07-14:


HiddenRambler, 2017-07-14:
Can you confirm that it is a NTAG215 that you're trying to write to? You can use https://play.google.com/store/apps/details?id=com.nxp.taginfolite app to check.

MercilessOtter, 2017-07-14:
I can confirm that it is an NFC Forum Type 2 Tag and it is empty/un-written

HiddenRambler, 2017-07-14:
NFC Forum Type 2 applies to other NTAGs as well. If you used the TagInfo app i linked to it should show something like:

IC type:
NTAG215

and not another NTAG value.

MercilessOtter, 2017-07-14:
IC type NTAG215, 

I had downloaded the NXP writer not reader, remedied and received the correct info

This is the link to where I had purchased the 3M tags:

https://www.amazon.com/TagLabs-NTAG215-Chips-25-Pack-Choice/dp/B071GWLSTL/ref=sr_1_fkmr0_1?ie=UTF8&qid=1500042996&sr=8-1-fkmr0&keywords=taglabs%2Bnfc%2Bnta&th=1

HiddenRambler, 2017-07-14:
Assuming it's not a problem due to phone/tag moving during the write process and losing connection?

Could you attach a dump of the blank tag you have? It doesnt have any private information. You can use the share option in the TagInfo app to generate a text or xml file with the full information.

MercilessOtter, 2017-07-14:
** TagInfo scan (version 4.23) 2017-07-14 10:40:47 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF Message present:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.Ndef]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:44:55:CA:60:57:81
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:44:55 9D (UID0-UID2, BCC0)
[01] *  CA:60:57:81 (UID3-UID6)
[02] .  7C 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  E1:10:3E:00 (OTP0-OTP3)
[04] .  03 00 FE 00 |....|
[05] .  00 00 00 00 |....|
[06] .  00 00 00 00 |....|
[07] .  00 00 00 00 |....|
[08] .  00 00 00 00 |....|
[09] .  00 00 00 00 |....|
[0A] .  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] .  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] .  00 00 00 00 |....|
[11] .  00 00 00 00 |....|
[12] .  00 00 00 00 |....|
[13] .  00 00 00 00 |....|
[14] .  00 00 00 00 |....|
[15] .  00 00 00 00 |....|
[16] .  00 00 00 00 |....|
[17] .  00 00 00 00 |....|
[18] .  00 00 00 00 |....|
[19] .  00 00 00 00 |....|
[1A] .  00 00 00 00 |....|
[1B] .  00 00 00 00 |....|
[1C] .  00 00 00 00 |....|
[1D] .  00 00 00 00 |....|
[1E] .  00 00 00 00 |....|
[1F] .  00 00 00 00 |....|
[20] .  00 00 00 00 |....|
[21] .  00 00 00 00 |....|
[22] .  00 00 00 00 |....|
[23] .  00 00 00 00 |....|
[24] .  00 00 00 00 |....|
[25] .  00 00 00 00 |....|
[26] .  00 00 00 00 |....|
[27] .  00 00 00 00 |....|
[28] .  00 00 00 00 |....|
[29] .  00 00 00 00 |....|
[2A] .  00 00 00 00 |....|
[2B] .  00 00 00 00 |....|
[2C] .  00 00 00 00 |....|
[2D] .  00 00 00 00 |....|
[2E] .  00 00 00 00 |....|
[2F] .  00 00 00 00 |....|
[30] .  00 00 00 00 |....|
[31] .  00 00 00 00 |....|
[32] .  00 00 00 00 |....|
[33] .  00 00 00 00 |....|
[34] .  00 00 00 00 |....|
[35] .  00 00 00 00 |....|
[36] .  00 00 00 00 |....|
[37] .  00 00 00 00 |....|
[38] .  00 00 00 00 |....|
[39] .  00 00 00 00 |....|
[3A] .  00 00 00 00 |....|
[3B] .  00 00 00 00 |....|
[3C] .  00 00 00 00 |....|
[3D] .  00 00 00 00 |....|
[3E] .  00 00 00 00 |....|
[3F] .  00 00 00 00 |....|
[40] .  00 00 00 00 |....|
[41] .  00 00 00 00 |....|
[42] .  00 00 00 00 |....|
[43] .  00 00 00 00 |....|
[44] .  00 00 00 00 |....|
[45] .  00 00 00 00 |....|
[46] .  00 00 00 00 |....|
[47] .  00 00 00 00 |....|
[48] .  00 00 00 00 |....|
[49] .  00 00 00 00 |....|
[4A] .  00 00 00 00 |....|
[4B] .  00 00 00 00 |....|
[4C] .  00 00 00 00 |....|
[4D] .  00 00 00 00 |....|
[4E] .  00 00 00 00 |....|
[4F] .  00 00 00 00 |....|
[50] .  00 00 00 00 |....|
[51] .  00 00 00 00 |....|
[52] .  00 00 00 00 |....|
[53] .  00 00 00 00 |....|
[54] .  00 00 00 00 |....|
[55] .  00 00 00 00 |....|
[56] .  00 00 00 00 |....|
[57] .  00 00 00 00 |....|
[58] .  00 00 00 00 |....|
[59] .  00 00 00 00 |....|
[5A] .  00 00 00 00 |....|
[5B] .  00 00 00 00 |....|
[5C] .  00 00 00 00 |....|
[5D] .  00 00 00 00 |....|
[5E] .  00 00 00 00 |....|
[5F] .  00 00 00 00 |....|
[60] .  00 00 00 00 |....|
[61] .  00 00 00 00 |....|
[62] .  00 00 00 00 |....|
[63] .  00 00 00 00 |....|
[64] .  00 00 00 00 |....|
[65] .  00 00 00 00 |....|
[66] .  00 00 00 00 |....|
[67] .  00 00 00 00 |....|
[68] .  00 00 00 00 |....|
[69] .  00 00 00 00 |....|
[6A] .  00 00 00 00 |....|
[6B] .  00 00 00 00 |....|
[6C] .  00 00 00 00 |....|
[6D] .  00 00 00 00 |....|
[6E] .  00 00 00 00 |....|
[6F] .  00 00 00 00 |....|
[70] .  00 00 00 00 |....|
[71] .  00 00 00 00 |....|
[72] .  00 00 00 00 |....|
[73] .  00 00 00 00 |....|
[74] .  00 00 00 00 |....|
[75] .  00 00 00 00 |....|
[76] .  00 00 00 00 |....|
[77] .  00 00 00 00 |....|
[78] .  00 00 00 00 |....|
[79] .  00 00 00 00 |....|
[7A] .  00 00 00 00 |....|
[7B] .  00 00 00 00 |....|
[7C] .  00 00 00 00 |....|
[7D] .  00 00 00 00 |....|
[7E] .  00 00 00 00 |....|
[7F] .  00 00 00 00 |....|
[80] .  00 00 00 00 |....|
[81] .  00 00 00 00 |....|
[82] .  00 00 00 BD (LOCK2-LOCK4, CHK)
[83] .  04 00 00 FF (CFG, MIRROR, AUTH0)
[84] .  00 05 -- -- (ACCESS)
[85] +P FF FF FF FF (PWD0-PWD3)
[86] +P 00 00 -- -- (PACK0-PACK1)

  *:locked & blocked, x:locked,
  +:blocked, .:un(b)locked, ?:unknown
  r:readable (write-protected),
  p:password protected, -:write-only
  P:password protected write-only

MercilessOtter, 2017-07-14:
I was able to write tags with the 2.3.1 update, but because of the SSB issue, I was unable to configure some BIN files to write to the tags, which is why I upgraded to 2.4.1

HiddenRambler, 2017-07-14:
Could you check 2.3.1 and 2.3.2 to see if both of them work? It would help me narrow down the problem.

I can't see anything wrong with your tag.

MercilessOtter, 2017-07-14:
I have tested 2 tags and yes on each version 3.1 & 3.2, both will load and write tags properly.
But when trying to access the SSB data on a SSB BIN file the app will close

Bombastisch, 2017-07-14:
Thats because there is no SSB data on thr tag.  There is only SSB data on the tag if you created a savefile in SSB (the game) itself. 

The fix from hiddenrambler now isnt correctly working either. It actually offers you to edit SSB data on the tag if there is none. 

The button should actually just lightup if there is really SSB data on it. I would love to do that. But i have to less programming knowledge to do that

HiddenRambler, 2017-07-14:
@MercilessOtter Could i ask you to try 2.3.4 and 2.4.0 as well.

I looked at the code but could not find an obvious change that could cause this bug.

MercilessOtter, 2017-07-14:
2.3.4BETA - I am unable to write any SSB based BIN, it will close when attempting to autowrite, and it will close when trying to access editing SSB Data

2.4.0 - I am able to auto write BINs regardless if it is an SSB based or not, but regardless when trying to access the edit SSB data the app will close

HiddenRambler, 2017-07-14:
Could you try this version:
[TagMoTest.zip](https://github.com/HiddenRambler/TagMo/files/1149218/TagMoTest.zip)

If it works could you retry 2.4.1 release.

MercilessOtter, 2017-07-14:
Hidden, the Test would not allow me to auto write any BIN files regardless of it being SSB or not (tried with 8 bit link, SSB sheik, SSB Ganon, and finally tried with Shovel Knight) I am able to access edit SSB without the app crashing but I am now yielding the original java error

HiddenRambler, 2017-07-14:
Could you try this version? 
[TagMoTest2.zip](https://github.com/HiddenRambler/TagMo/files/1149301/TagMoTest2.zip)

MercilessOtter, 2017-07-14:
I am sorry that took me so long to test Hidden, the second test APK will not auto write any BINs either whether or not it is SSB based. I am able to access editing SSB data. The only issue is writing to the NTAG215 tag

MercilessOtter, 2017-07-14:
I have re-downloaded all of the BIN files I am using to ensure that none of them are corrupted as well

HiddenRambler, 2017-07-14:
did you check the 3rd version?

MercilessOtter, 2017-07-14:
3rd one worked!!! I am able to write all BIN files! Both regular and SSB based BINs, I am also able to access editing SSB data and am able to write after editing the data!!!

Thank you Hidden, you're the man!

HiddenRambler, 2017-07-14:
Glad to hear it. 

Now we need someone with android 4.4 to test it as well.

@jmh363905 could I bother you to test this TagMoTest3(SingleTop).zip file from my previous comment. I want to make sure it doesnt break the functionality on your phone.

jmh363905, 2017-07-14:
You bet, gimme a few.

jmh363905, 2017-07-14:
Scanned and wrote an NTAG without issue.

HiddenRambler, 2017-07-14:
That was fast :) 
So hopefully this fix works for both version. I'll put out a new release.

Many thanks for taking the time to test it.

Bombastisch, 2017-07-14:
I like the way you try to make the app functional for everyone! @HiddenRambler   That's really great!

jmh363905, 2017-07-14:
He's a great dev!

HiddenRambler, 2017-07-14:
I'm a He (last time I checked).

2.4.3 released with the fix. 

Thanks for the help.

closed 2017-07-14 by HiddenRambler