HiddenRamblings/TagMo #57: Got an issue with TagMo which I'd like to potentially get help with.

Pridd, 2017-07-22:
Not sure if there is any other issue solved or open already with this, and my apologies if I'm just re-opening one.

So I have a samsung galaxy J3 and have recently learnt that I -could- do this instead of consistently and constantly buy nonexistant amiibo, so I've started doing this instead, bought 20 NTAG215 tags, and what not.

Problem is, I have not been able to create any amiibo, using the most recent tagmo version, and 2.3.1 (since I saw it was being recommended), and am constantly getting the same error. The error in question is "Error while writing main data (stage 1) java.io.IOException: Transcieve Failed".

Any help would be appreciated, thank you in advance.

HiddenRambler, 2017-07-22:
The latest version is 2.4.3 https://github.com/HiddenRambler/TagMo/releases

Transcieve failed is a common error when you have the wrong type of tag as well (not NTAG215). Easiest way to verify is to use the NXP TagInfo app and to make sure it recognises it as a NTAG215: https://play.google.com/store/apps/details?id=com.nxp.taginfolite

Pridd, 2017-07-22:
Yes, I have tried that one as well. Aparently, 13 of the 20 tags worked just fine after I tried out more extensively right now.

After scanning with that app, aparently they have sent 7 NTAG213 tags. Which aparently is the root of the problem.

Thank you for the help, and my apologies for bothering.

HiddenRambler, 2017-07-22:
Glad it's resolved.

closed 2017-07-22 by HiddenRambler