HiddenRamblings/TagMo #66: 2.5 Lost AMIIBO Scan Ability?

jmh363905, 2017-08-02:
Am I missing it somewhere or have we lost the ability to scan AMIIBOs?  No Scan Tag button and I can't find it in the settings anywhere.

HiddenRambler, 2017-08-02:
its the place as before. screenshot?

jmh363905, 2017-08-02:
[image]

jmh363905, 2017-08-02:
I will note that I was NOT able to upgrade the existing install.  Install failed until I uninstalled 2.4.3 and then installed this one afterward.

HiddenRambler, 2017-08-02:
see if you can "scroll" the screen to the right.

jmh363905, 2017-08-02:
Negative.  Just up and down and there's nothin down there :-/

jmh363905, 2017-08-02:
[image]
I am able to load and write them.  This is a screenshot of a loaded one.  I know it's not pertinent to this issue, but I kinda miss seeing the serial number too.

jmh363905, 2017-08-02:
I trust when you say it should be there.  Is it a GUI scaling issue perhaps (if that's a thing even)?  And I've asked this before and I'll def. say it again (differently this time): send me an email to juliopoe at gmail with your PayPal address and I'll donate to your cause.  I ain't even playin' :)

HiddenRambler, 2017-08-02:
there was a change done to make the main panel scrollable. i'm guessing this is due to that. i can't reproduce the issue. but see if this fixes it.
[TagMoTest (2).zip](https://github.com/HiddenRamblings/TagMo/files/1195419/TagMoTest.2.zip)

jmh363905, 2017-08-02:
You know who you are?  The man.  You the man!  (or woman if that's the case, no offense intended if so :D )
[image]

HiddenRambler, 2017-08-02:
sorry the layout still doesnt look correct (it's all misaligned). can you tell me the phone model and the android version so i can check this further.

jmh363905, 2017-08-02:
Samsung Galaxy Light SGH-T399N (480x800 res -GSM Arena). Android 4.4.2 Kernel 3.4.39-2361166 Build KOT49H.T399NUVUANH1.

(Sorry, I figured since you solved my issue I would close it, but I'll leave that to you :) )

jmh363905, 2017-08-02:
And you know me--feel free to make me a test dummy for various builds.  This issue and any others that you need someone to test with as you've done before.  I know today I had a delay from my post to the time I got the screenshot to you, but I had to go do that adulting thing and my job-job had me tied up for a minute.

North101, 2017-08-02:
@HiddenRambler Looks like the new amiibo info pushed the buttons off the screen. The recent change to the layout should fix it but a proper fix would be to support lower dpi screens

jmh363905, 2017-08-02:
A lot of the info shown is redundant (it seems). Would removing some info that is repetitive help?

jmh363905, 2017-08-02:
[image]
 Like this one.  I see Splatoon and Marie twice. Not sure about the figure thing.  I guess that changes to card for the stuff that isn't a figurine like Pokemon and Mario Sports kinda stuff?

North101, 2017-08-02:
@jmh363905 So from top left to bottom right you're seeing:
Amiibo Name
Amiibo ID (Unique to that Amiibo)
Amiibo Series (Link might show up from Super Smash Bros or Legend of Zelda)
Amiibo Type (Figure, Card, Yarn)
Game Series (Mario will always show up as from Super Mario Bros even though he has a Super Smash Bros Amiibo)
Character (Sheik will show up as Zelda)

Whether its repetitive or not depends on the Amiibo and the game you're using it on. e.g. A game may check if the Amiibo is Zelda, but Sheik also identifies as Zelda.

jmh363905, 2017-08-02:
IMHO, I think some of that is ancillary and could perhaps be tucked away.  Maybe a "More Info" button for those who truly want to see or doesn't know that Shiek is from the Zelda games? Amiibo Name, Series, and Serial are what (I believe) to be the most needed info.  The rest of it kind of gums up the rule of simplicity.  I mean no disrespect at all.

jmh363905, 2017-08-02:
I see where it is all somehow relevant info.  But when you scan Majora's Mask Link you get "Link - Majora's Mask", "Legend of Zelda", "The Legend of Zelda", "Link"  One of those Legend of Zelda's is un-needed IMO.  How the game identifies it is subsidiary info.  And the solitary "Link" is redundant.  Identifying an Inkling Squid as an Inkling is also redundant. There may be instances where all lines of info may have unique identifying information, but from someone who has scanned and written every one ever made, I can assure you--it's very far and few between.  My point is that you could also tell what each character drops in Breath of the Wild, but at what point are you displaying TMI.  I think that threshold got passed after 4 lines of relevant info:  ID (I'm assuming this is the serial I keep referring to), Name, Series, and Figurine/Card/Yard.  I understand the need to want to have more and more info shown, but I'm a proponent of KISS.  I'm not an opponent of a "More Info" button to show this extra info in the rare circumstances that it isn't redundant.

I know, I know.. I'm not a coder and the only thing I'm providing is feedback, but feedback is invaluable.  I open this app at least 4 times a day.  Have written and scanned over 600 tags and counting, and I'm telling you as a fan of this project, TMI.... Just.. TMI.

jmh363905, 2017-08-03:
I don't intend to be hateful and I hope I was not perceived that way.  I guess the main driving behind my words is this: Today myself and every other person who doesn't carry a phablet around had this beloved app break.  If you're looking to fix it, the simplest would be to remove some of the extra info--the thing that caused it to break for all of us who like modestly sized phones.

North101, 2017-08-03:
Its not coming across as hateful (you're just expressing an opinion). I believe it was already fixed yesterday and just needs a new version released.

Showing less info wouldn't actually fix the under lying issue in all cases, but rearranging whats shown or adding constraints to the text boxes would (which is what was done).

jmh363905, 2017-08-03:
I'm also an ideas man. I think I tossed the idea of small GIFs in some of the white space to the lower left. Didn't get traction, but I hope to see it come to fruition some day still ;-)

HiddenRambler, 2017-08-04:
@jmh363905 could you check if the attached file looks ok. it's the build from the latest trunk. I checked on the emulator with your specs and it looked fine.

[TagMoTest (3).zip](https://github.com/HiddenRamblings/TagMo/files/1201356/TagMoTest.3.zip)

jmh363905, 2017-08-04:
This works just fine :)

Bombastisch, 2017-08-06:
Can I close this thread or did you keep it as a reminder @HiddenRambler ? :)

HiddenRambler, 2017-08-06:
I'll close it as it appears to be fixed in trunk.

closed 2017-08-06 by HiddenRambler

jmh363905, 2017-08-07:
I don't know if this is related to the sizing issue or not, but I noticed that I'm missing the ability to edit Twilight Princess data. Is there still a button for that that I'm not seeing off the screen or is that gone now?

HiddenRambler, 2017-08-07:
It should still be there but it only appers to for wolf link amiibo.

jmh363905, 2017-08-07:
That's what I have scanned but it isn't there.