HiddenRamblings/TagMo #67: Min API Level

North101, 2017-08-04:
@HiddenRambler 

The min api level is set to level 14, however NfcAdapter.ACTION_ADAPTER_STATE_CHANGED requires a min api level of 18. Should we increase it? It seems like this would cause a crash on NFC enabled devices < Android ICS (non NFC enabled devices wouldn't even get that far).

Below is the current distribution of android versions. I think we can safely update it to at least 18 though preferably 19.
[image]

HiddenRambler, 2017-08-04:
When the app was first developed there were some users who explicitly asked for <4.4 android support, which is why I chose that api level.

ACTION_ADAPTER_STATE_CHANGED is used solely to warn the user about Nfc enabled state. We could wrap the code that uses it in an exception handler and ignore any exceptions without affecting other functionality.

I will however ask a question on gbatemp asking if anyone is using <4.3 so we can decide what to do.

North101, 2017-08-04:
Ok, thats understandable. We can probably keep the API level as is then. Should probably add an API check to that then. I havent had a proper look at how the NFC functionality works so I was unsure whether it required level 18 or not

closed 2017-08-05 by North101