HiddenRamblings/TagMo #68: Main UI status text box change

HiddenRambler, 2017-08-04:
As we already have a popup which warns of missing keys. I felt that having seperate key warnings was wasting main ui space.

I feel it's better to merge key/nfc errors togther and show one label as below (with appropriate message) only if there is an error.

[image]

 @North101 any thoughts?

North101, 2017-08-04:
Sounds good but I think its useful to have the NFC info on the main screen, might even be nice to have an easy toggle for it (if thats possible).

North101, 2017-08-04:
Hmm, on second thoughts maybe move the NFC toggle to the action bar?

HiddenRambler, 2017-08-04:
I don't think we can toggle NFC directly. But we can open the nfc settings panel: 

https://stackoverflow.com/questions/5945100/android-changing-nfc-settings-on-off-programmatically

We could prompt for it within the NFC activity if NFC is not available.

Also i'm considering moving the Write Tag (Raw) function to the action bar menu it's not needed except for people doing their own amiibo editing/encryption outside of the app, which is rare if at all.

I'm keen to get some extra space at the top. Specially if we implement your amiibo api branch as we could easily display amiibo images.

North101, 2017-08-04:
Ah thats a shame.

I think maybe putting the errors at the bottom instead of the top would be good.

As for the images, we don't actually need the amiibo api branch, all the images can be accessed via:
`https://raw.githubusercontent.com/N3evin/AmiiboAPI/master/images/icon_<head hex>-<tail hex>.png`

HiddenRambler, 2017-08-04:
Not sure how the warning label would look if the panel has to scroll. My UI-fu is not good.

I'll consolidate the error message, trigger the nfc settings openning. I'll leave it to you to better lay out the ui if you have the time ;).

North101, 2017-08-04:
The error message would exist outside the scroll (as it should always be shown). You'd need something like:
`<RelativeLayout
    xmlns:android="http://schemas.android.com/apk/res/android"
    android:layout_width="match_parent"
    android:layout_height="match_parent">
    
    <TextView
        android:id="@+id/txtError"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:layout_alignParentBottom="true"/>

    <ScrollView
        android:id="@+id/container_scroll_view"
        android:layout_above="@id/txtError"
        android:layout_width="match_parent"
        android:layout_height="match_parent">
           [...]
    </ScrollView>
</RelativeLayout>`

HiddenRambler, 2017-08-04:
the layout changes have been pushed. will get the nfc changes working next.

closed 2017-08-05 by HiddenRambler