HiddenRamblings/TagMo #70: 2.5.1 Will not Write - Transceive Failed

jmh363905, 2017-08-07:
Topic says it all.  They are the same tags (blank NTAG215) that I've been using with TagMo the whole time.

HiddenRambler, 2017-08-07:
Were you trying to restore a tag or write a new one?

Could you include a logcat dump.

jmh363905, 2017-08-07:
Write a new one from a loaded bin file.
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/1206345/tagmo_logcat.txt)

jmh363905, 2017-08-07:
Downgrading allows me to write the same bin file to the same tag that 2.5.1 failed on, and the AMIIBOs are working in-game, so I know I have the proper tags.  Not sure what it could be. I perused through the log and saw it threw: "version information error" before the Transceive failed text.

HiddenRambler, 2017-08-07:
It could be that the phone/tag moved while the operation was being carried out. Due to the extra validation it takes longer to do the write operation, and the phone/tag may have gotten out of sync.

If that's not the case could you use NXP TagInfo app to dump the tag details and share it. I'm curious as to why it failed to retrieve the version. I believe you when you say the tags are correct I'd like to see what may be different from the version that I have.

jmh363905, 2017-08-07:
It failed a few times actually.  And I press the tags firmly to the bottom to write them.  Failed over and over in 2.5.1.  The one you posted in a previous thread to work out GUI works perfectly with the same tag that 2.5.1 fails on.  How do I dump tag details.  The thing about these tags (all I've EVER gotten) that always perplexed me is that they self-identify as NTAG203 and NTAG205...until written.. Then the same app that identifies them as NTAG203 or 205 (depending on which NTAG info app you use), identifies them as NTAG215 after they have AMIIBO data written to them.

https://www.aliexpress.com/item/50pcs-Lot-NTAG215-NFC-Stickers-Amiibo-Tag-For-TagMo-Dia-25mm-Lable-Forum-Type2-Sticker-Amiibos/32808465647.html?spm=a2g0s.9042311.0.0.bcPFwh

Those are the tags that I've purchased over and over.  (I have over 200 in stock right now and written over 300 of them successfully with previous builds).

jmh363905, 2017-08-07:
I will upload a proof video if you need it haha.

HiddenRambler, 2017-08-07:
No need to go that far. use this app: https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en_GB

do a full scan with it and use the share option in it to get a xml or a text file (either is fine).

That app is by NXP the manufacturer of NTAG so it should be definitive.

jmh363905, 2017-08-07:
** TagInfo scan (version 4.23) 2017-08-07 17:59:56 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG203(F) (NTAG203(F))

-- NDEF ------------------------------

# No NDEF Message present:

-- EXTRA ------------------------------

# Memory size:
168 bytes total memory
* 42 pages, with 4 bytes per page
* 144 bytes user memory (36 pages)

# IC detailed information:
Full product name:
* NT2H0301G0DUD or NT2H0301F0DTx

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.Ndef]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 24576 ms

# Detailed protocol information:
ID: 04:73:7E:32:61:57:81
ATQA: 0x4400
SAK: 0x00

jmh363905, 2017-08-07:
I'm fixing to make a video and upload to YouTube with this tag getting scanned, showing as incorrect NTAG203, then 2.5.1 failing, then installing the WIP one you told me to try before 2.5.1 was released, scanning tag again to show I haven't done a switch-a-roo, then write it successfully, and will then scan said "amiibo" on a New Nintendo 3DS to show that it reads it as a real thing and lets me change the owner info, reset it, etc...  I've made hundreds for Breath of the Wild for my friends....None have failed.

HiddenRambler, 2017-08-07:
As you can clearly see it's not an NTAG215. Looks like an older NTAG series. likely compatible with 215 for basic operations. Glad I added the version check or I'd never learn about this. Could you do a full hex scan (the last tab on that app) and share the file. 

For now here is a version with the version check disabled, this should work:
[TagMo.2.5.1(NoVersionCheck).zip](https://github.com/HiddenRamblings/TagMo/files/1206412/TagMo.2.5.1.NoVersionCheck.zip)

jmh363905, 2017-08-07:
None that I've EVER found online (I've tried other suppliers) have ever identified as NTAG215 unless written for some reason.

HiddenRambler, 2017-08-07:
That would indicate a phone issue. I'll add a preference to disable version check.

jmh363905, 2017-08-07:
I'm uploading proof videos now showing exactly what I'm describing.  It seems that if it's going to check for compatible tags, NTAG205 and 203 should be allowed because they are indeed compatible with AMIIBO functionality on all 3 systems and automagically show up as NTAG215 after being written by TagMo.  I'm not lying about that at all and I'm fixing to prove it to you and other nay-sayers that read this

HiddenRambler, 2017-08-07:
If it is a phone issue, then the tags you have are probably NTAG215s not NTAG205/203s.

jmh363905, 2017-08-07:
https://youtu.be/SSaxz7fTwao  <-- 2.5.1 failing with a tag
https://youtu.be/nHtZlLRDV20 <-- 2.5.0 working with same tag and then being scanned successfully on 3DS

HiddenRambler, 2017-08-08:
Now this is speculation but I think this is what is happenning:

NTAG20x are a generation older than NTAG21x, however I cannot see a way NTAG203 would work as an amiibo at all due to the difference in memory size. TagMo blindly writes the security bits at what would be the last memory locations of a NTAG215. These memory locations does not even exist on NTAG203/213 so it should by all means fail. The only tag that has a chance of working is NTAG205.

I believe your phone is identifying the tag incorrectly probably due to the chipset not supporting all NFC operations (I read some galaxy phones having such issues, yours might be one of them). This is causing the NFC Info app to identify it incorrectly. 

The validation feature was added to prevent people from reporting bugs due to using the wrong version of tags. But if problems such as yours exists then it becomes pointless. I will remove the code from the next version. **Please let me know if the version I have attached earlier works for you.**

Bombastisch, 2017-08-08:
Actually as shown in the "2.5.1 not working" video,  the two diffrent apps identify the same tag as different types. NFC tools identifys the tag as Mifare Ultralight and NXP Taginfo says it's NTAG 203. So it actually has to be something phone specific, because thats not normal.

I actually liked the idea of checking the type, but apperently older phones have a problem with that.

HiddenRambler, 2017-08-08:
I believe NTAG is a specific type of Ultralight.

jmh363905, 2017-08-08:
The No version check version works fine.  I'm tempted to mail you one of these tags so you can investigate a little further.

jmh363905, 2017-08-08:
What I forgot to show in those videos is that the tag, after written, shows up as an MiFare NTAG215 in NFC Tools app (the other one I used in videos that wasn't the one you recommended)

HiddenRambler, 2017-08-08:
Do you have any friends with a different android phone with NFC? you could get one scanned and see what it says.

jmh363905, 2017-08-08:
All my friends and myself use Fruit Fone.   I purchased this one for the sole purpose of using TagMo.  But I will see what I can do.  May be a several-day delay though.

Bombastisch, 2017-08-08:
@HiddenRambler 

I don't think so. Ultralight is from the Mifare product line and the others are NTags.

HiddenRambler, 2017-08-08:
@Bombastisch both are part of NXP. and NTAG has a compatibility write method for compatibility with mifare classic. also the android api for writing ntag is the same as for mifare:
https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/NfcActivity.java#L151

jmh363905, 2017-08-08:
I have a friend coming over this afternoon who has a Samsung Galaxy S7 at his house and is going to scan one of these tags with the NXP Taginfo you recommended in previous thread.  I'll provide feedback as well as exact model number and android+kernel version info at that time

jmh363905, 2017-08-08:
I don't know if you want a new issue started over this or if this venue is fine:  Very very minor, but when you scan a blank tag, the text turns red to the left, but on subsequent scans and even loading of bin files, the text never goes back black.  I'm sure this was only meant to be red when a blank tag or something was scanned, but it kinda sticks that way until the app is closed and reopened.

jmh363905, 2017-08-09:
OK... Same tag, scanned with a Samsung-SM-G930A, running Android 6.0.1.  Kernel 3.18.20-8575098.  The following is the info dump from a tag that identified as 203 on my phone and 215 on his: 

** TagInfo scan (version 4.23) 2017-08-08 21:15:52 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG215

-- NDEF ------------------------------

# No NDEF Message present:

-- EXTRA ------------------------------

# Memory size:
504 bytes user memory
* 126 pages, with 4 bytes per page

# IC detailed information:
Full product name: NT2H1511G0DUx
Capacitance: 50 pF

# Version information:
Vendor ID: NXP
Type: NTAG
Subtype: 50 pF
Major version: 1
Minor version: V0
Storage size: 504 bytes
Protocol: ISO/IEC 14443-3

# Configuration information:
ASCII mirror disabled
NFC counter: disabled
No limit on wrong password attempts
Strong load modulation enabled

# Originality check:
Signature verified with NXP public key

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.Ndef]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 618 ms

# Detailed protocol information:
ID: 04:A6:29:32:61:57:80
ATQA: 0x4400
SAK: 0x00

# Memory content:
[00] *  04:A6:29 03 (UID0-UID2, BCC0)
[01] *  32:61:57:80 (UID3-UID6)
[02] .  84 48 00 00 (BCC1, INT, LOCK0-LOCK1)
[03] .  E1:10:3E:00 (OTP0-OTP3)
[04] .  03 00 FE 00 |....|
[05] .  00 00 00 00 |....|
[06] .  00 00 00 00 |....|
[07] .  00 00 00 00 |....|
[08] .  00 00 00 00 |....|
[09] .  00 00 00 00 |....|
[0A] .  00 00 00 00 |....|
[0B] .  00 00 00 00 |....|
[0C] .  00 00 00 00 |....|
[0D] .  00 00 00 00 |....|
[0E] .  00 00 00 00 |....|
[0F] .  00 00 00 00 |....|
[10] .  00 00 00 00 |....|
[11] .  00 00 00 00 |....|
[12] .  00 00 00 00 |....|
[13] .  00 00 00 00 |....|
[14] .  00 00 00 00 |....|
[15] .  00 00 00 00 |....|
[16] .  00 00 00 00 |....|
[17] .  00 00 00 00 |....|
[18] .  00 00 00 00 |....|
[19] .  00 00 00 00 |....|
[1A] .  00 00 00 00 |....|
[1B] .  00 00 00 00 |....|
[1C] .  00 00 00 00 |....|
[1D] .  00 00 00 00 |....|
[1E] .  00 00 00 00 |....|
[1F] .  00 00 00 00 |....|
[20] .  00 00 00 00 |....|
[21] .  00 00 00 00 |....|
[22] .  00 00 00 00 |....|
[23] .  00 00 00 00 |....|
[24] .  00 00 00 00 |....|
[25] .  00 00 00 00 |....|
[26] .  00 00 00 00 |....|
[27] .  00 00 00 00 |....|
[28] .  00 00 00 00 |....|
[29] .  00 00 00 00 |....|
[2A] .  00 00 00 00 |....|
[2B] .  00 00 00 00 |....|
[2C] .  00 00 00 00 |....|
[2D] .  00 00 00 00 |....|
[2E] .  00 00 00 00 |....|
[2F] .  00 00 00 00 |....|
[30] .  00 00 00 00 |....|
[31] .  00 00 00 00 |....|
[32] .  00 00 00 00 |....|
[33] .  00 00 00 00 |....|
[34] .  00 00 00 00 |....|
[35] .  00 00 00 00 |....|
[36] .  00 00 00 00 |....|
[37] .  00 00 00 00 |....|
[38] .  00 00 00 00 |....|
[39] .  00 00 00 00 |....|
[3A] .  00 00 00 00 |....|
[3B] .  00 00 00 00 |....|
[3C] .  00 00 00 00 |....|
[3D] .  00 00 00 00 |....|
[3E] .  00 00 00 00 |....|
[3F] .  00 00 00 00 |....|
[40] .  00 00 00 00 |....|
[41] .  00 00 00 00 |....|
[42] .  00 00 00 00 |....|
[43] .  00 00 00 00 |....|
[44] .  00 00 00 00 |....|
[45] .  00 00 00 00 |....|
[46] .  00 00 00 00 |....|
[47] .  00 00 00 00 |....|
[48] .  00 00 00 00 |....|
[49] .  00 00 00 00 |....|
[4A] .  00 00 00 00 |....|
[4B] .  00 00 00 00 |....|
[4C] .  00 00 00 00 |....|
[4D] .  00 00 00 00 |....|
[4E] .  00 00 00 00 |....|
[4F] .  00 00 00 00 |....|
[50] .  00 00 00 00 |....|
[51] .  00 00 00 00 |....|
[52] .  00 00 00 00 |....|
[53] .  00 00 00 00 |....|
[54] .  00 00 00 00 |....|
[55] .  00 00 00 00 |....|
[56] .  00 00 00 00 |....|
[57] .  00 00 00 00 |....|
[58] .  00 00 00 00 |....|
[59] .  00 00 00 00 |....|
[5A] .  00 00 00 00 |....|
[5B] .  00 00 00 00 |....|
[5C] .  00 00 00 00 |....|
[5D] .  00 00 00 00 |....|
[5E] .  00 00 00 00 |....|
[5F] .  00 00 00 00 |....|
[60] .  00 00 00 00 |....|
[61] .  00 00 00 00 |....|
[62] .  00 00 00 00 |....|
[63] .  00 00 00 00 |....|
[64] .  00 00 00 00 |....|
[65] .  00 00 00 00 |....|
[66] .  00 00 00 00 |....|
[67] .  00 00 00 00 |....|
[68] .  00 00 00 00 |....|
[69] .  00 00 00 00 |....|
[6A] .  00 00 00 00 |....|
[6B] .  00 00 00 00 |....|
[6C] .  00 00 00 00 |....|
[6D] .  00 00 00 00 |....|
[6E] .  00 00 00 00 |....|
[6F] .  00 00 00 00 |....|
[70] .  00 00 00 00 |....|
[71] .  00 00 00 00 |....|
[72] .  00 00 00 00 |....|
[73] .  00 00 00 00 |....|
[74] .  00 00 00 00 |....|
[75] .  00 00 00 00 |....|
[76] .  00 00 00 00 |....|
[77] .  00 00 00 00 |....|
[78] .  00 00 00 00 |....|
[79] .  00 00 00 00 |....|
[7A] .  00 00 00 00 |....|
[7B] .  00 00 00 00 |....|
[7C] .  00 00 00 00 |....|
[7D] .  00 00 00 00 |....|
[7E] .  00 00 00 00 |....|
[7F] .  00 00 00 00 |....|
[80] .  00 00 00 00 |....|
[81] .  00 00 00 00 |....|
[82] .  00 00 00 BD (LOCK2-LOCK4, CHK)
[83] .  04 00 00 FF (CFG, MIRROR, AUTH0)
[84] .  00 05 -- -- (ACCESS)
[85] +P FF FF FF FF (PWD0-PWD3)
[86] +P 00 00 -- -- (PACK0-PACK1)

  *:locked & blocked, x:locked,
  +:blocked, .:un(b)locked, ?:unknown
  r:readable (write-protected),
  p:password protected, -:write-only
  P:password protected write-only

--------------------------------------

Bombastisch, 2017-08-09:
I have an idea to fix this:   

Instead of trying to check the NTag before writing you could do it afterwards!   

Look here: if the tag was succsesfully written everything is fine and you don't check for an Ntag215. But if there is an error when writing, then run the Ntag215 check to tell the user that he is probably using the wrong tags.

HiddenRambler, 2017-08-09:
@jmh363905
It looks like defective phone hypothesis was correct. However as the phone is actually usable as long as there is no version check I will disable version checking. Good to know about this issue so it will help others. Thanks for taking the time to test.

@Bombastisch 
If it writes then chances are it's all ok anyway. So doesn't seem like it's worth the effort at the moment. Perhaps a good idea would be to add a new menu item to just show the tag information incase someone needs to check.

North101, 2017-08-09:
@HiddenRambler I recommend adding a settings for it to the menu to enable/disable certain verifications

HiddenRambler, 2017-08-09:
@jmh363905 Sorry when I said defective phone I meant that it's a problem with the phone's chipset (common to all phones of the model) not that you have a broken phone. It could be because the NFC chipset was old but I've seen phones with older chipset than yours working fine. so i'm leaning towards defective/problematic. From the articles which I could find this seems to have been a number of older Samsung models.

BrainBacon, 2017-08-09:
It looks like I'm running into the same issue and I'm using a new Google Pixel.

HiddenRambler, 2017-08-09:
@BrainBacon I find it difficult to believe that the pixel can have a similar defect. could you use the TagInfo app and share the output.

BrainBacon, 2017-08-09:
I just figured out what I was doing wrong. I was trying to overwrite a tag that already has Amiibo data written to it. It works on an empty tag.

HiddenRambler, 2017-08-09:
What version are you using. the new version would warn tell you that.

BrainBacon, 2017-08-09:
Yeah I am not using the latest version. I tried to upgrade, but I get an error when I try to install the APK. I haven't yet tried to uninstall completely and reinstall.

jmh363905, 2017-08-09:
Now THAT is a problem I had. 2.5.0 wouldn't upgrade. Had to remove and reinstall

BrainBacon, 2017-08-09:
Can confirm, I do now get an error on the latest version.

HiddenRambler, 2017-08-09:
The upgrade issue is because somewhere down the line I had mixed up the signing key for the app. meaning android wouldnt allow an upgrade. Unfortunately if i try to go back to the key, people who have installed the new version will have problems... So yeah.

jmh363905, 2017-08-10:
LOL, that makes sense!  Do the least amount of harm that way.  I didn't see it in a release note so, just a suggestion, may want to mention that so folks know to uninstall all versions prior to 2.5.0 before installing that version or higher.  Just a tip.

I did have an etiquette question.  Earlier I posted about some text that stays red and should return to black (I'm assuming that's what you envisioned)... should I have created an ISSUE about that separately, or is there a place for suggestions perhaps?  I just see this as a chat room essentially, but I would assume things are ideally supposed to stay on topic.  So what of those instances when you have a suggestion to voice to the you, the dev (whether you decide to ever implement said suggestions is another matter)

HiddenRambler, 2017-08-10:
@jmh363905 It's always best to create a new issue. It keeps the issues clean and makes it easier to find search them. We get notifications about both new issues and comments so it doesnt really make a difference for responsiveness.

Bombastisch, 2017-08-10:
A commit just fixed that issue.

@jmh363905  for your new issue, please open a new issue thread. :)

closed 2017-08-10 by Bombastisch