HiddenRamblings/TagMo #73: Suggestion- rearrange buttons

jmh363905, 2017-08-10:
Up until 2.5.0 the write tag button was under the load tag button. Muscle memory is causing me to hit wrong button when I go to write. I see that hardware io buttons are on right, but can the write tag button be moved back to original location we've all come to be used to.

jmh363905, 2017-08-12:
[image]
Maybe grouping the QR stuff together to restore a more familiar layout? (Forgive the sloppy Paintbrush job)

Bombastisch, 2017-08-13:
#74

Bombastisch, 2017-08-24:
@North101 just told me why he made the layout change and his idea behind it sounds pretts logical to me.

Here is his explanation: #77 (scroll down a bit)

I guess it's just a matter of time until you got used tto it. ;)

closed 2017-08-24 by Bombastisch

jmh363905, 2017-08-25:
Care to elaborate on the logic so I can perhaps see it too?

jmh363905, 2017-08-25:
Yeah, I saw his logic in it.  Read on one side and write on the other, but I still maintain that it doesn't flow well at all.  Logic and practice are two different ideas.  Logically you could say that sure, I'll put the read on one side and the write on the other, but when you actually start using the app, the flow isn't there.  It isn't PRACTICAL.  It may be logical, but I argue it isn't practical.  Let's not be coy here--we're all using this to load up passed around BIN files onto blank tags.  That's what Tagmo is used for by the bulk of the population.  They wanna load, they wanna write it.  They don't give a crap about QR stuff or the immense amount of info that isn't helpful really. On that topic, I still also maintain that the new extra info is still too overwhelming and repetitive.  For example scanning an Animal Crossing Card (Reese for example), it will tell you the following info: Reese, Animal Crossing, Animal Crossing (again), and Reese (...again).  No Series info or character number though.  THAT is pertinent info, IMO.  Nothing against North101.  He seems like a go-getter, but when did this project get turned over to him?  Since we "went North" I've had scaling issues that rendered the app useless for a time, write issues because of my NFC chipset not reporting NTAGs properly when blank, and the layout changed.  Seems to me North101 needs to make a fork for his customized version if he's intent on changing everything the users have come to know and love about Tagmo: SIMPLICITY and FAMILIARITY.  I suggest we head back South...where the White Walkers haven't quite made it yet.

Bombastisch, 2017-08-25:
I just really think of removing that comment. It is not a fair comment on what work must be done to do such a app. Tagmo has a very small development team and North really pushed the app development forward and added great features like the browser and the QR -codes (yes that is a great feature). There is a lot of time involved in that progress and North was willing to spend his time on this project. And you just said that this all was unuseful. This isnt a fair way to talk to a dev as a simple user, who actually just complained about button layout before.

Please milden your language in your comment and be sure to treat the devs fairly or I will remove the comment, because it is hateful and hasn't anything to do with ideas or solutions. (With most of the problems already fixed btw.)

Also if you want to use a older version just go for it. The downloads are still available. ;)

jmh363905, 2017-08-25:
I'm not saying great features weren't added.  The browser.  The communication to the AMIIBO database to keep current as new ones are released. But no system lets you scan an AMIIBO as a QR code, so unless it's 2 people sitting next to each other swapping a QR code around on their phone screens, I, MYSELF, do not see the practicality or usability of the QR aspect of it.  I didn't intend for my comment to be hateful.  And I'm sorry it was perceived that way.  I stand by my feedback as a user though: 1. Too much info is shown in places that is often redundant. 2. QR Codes are a neat parlor trick but have no true application when dealing with AMIIBOs, which are NFC and not scanned in with a camera.  3.  The layout, while logical is not practical--such as the original layout was where input (load and scan) was on the first line, output (write and save) were on the 2nd line, and extras at the bottom (hex info, etc). This is all my humble opinion.  The only reason I don't revert to an older version is because of the updated AMIIBO database.  Period.  The updated database is literally the only thing I keep current for.  And I know it seems like I stand alone on this, but honestly--I would wager that I speak for a larger group that simply doesn't have the desire or motivation to create an account to say so.  I mean no disrespect.

jmh363905, 2017-08-25:
You'll have to forgive me.  I should simply get used to the design changes. I get the impression you guys aren't from the States.  I am.. from the south at that.  And since about January I've been on edge what with an orange colored, white supremacist holding power and me being surrounded by other supremacists that feel vindicated and able to come out of the woodwork.  So yeah, I've been on defensive mode for 8 months, and a little over 3 years to go and it has crept into even the smallest aspects of my life.  I'm sorry about that...  You guys are doing a bang up job.  Keep up the awesome work!

HiddenRambler, 2017-08-28:
@jmh363905, the old UI was the way it was because the features kept getting added more or less in that order. The recent release was the first time someone thought about the UI logically and changed it to structured layout.
I understand your point that some information seems superfluous, but some of that information is valuable for people who make a habit of keeping a full collection of Amiibos and care about the slightest difference.
The QR code feature is something I hope that the other Amiibo website will implement. It would allow users to skip the whole download step, hopefully this will come true someday.
All the new features are things different people care about, and unfortunately, it’s not possible to make them in a way to make everyone happy. The hope is that when all its users are taken into consideration that the new changes make a bigger positive change than the inconveniences.

P.S. Hope Harvey isn't causing you any issues.

jmh363905, 2017-08-28:
I understand all that. I didn't mean to lash out and sincerely apologize. And Harvey is one state down from me. Luckily not getting much besides some heavy rains being pushed in. I do have family down there in a costal town where it made landfall and they got evacuated in plenty of time. Thanks for asking!

HiddenRambler, 2017-08-28:
@jmh363905 looks like @North101 heard you: https://github.com/HiddenRamblings/TagMo/pull/82

jmh363905, 2017-08-28:
That's so awesome! I thought that was a good idea with that white space. You guys really do rock