HiddenRamblings/TagMo #76: Request for New Feature

skizdidlyidler, 2017-08-14:
Hello! First I'd like to say I really love this app, and I really appreciate your work!

I wanted to request that a feature be added, I'm sure people will agree!

Is there a way you could implement the file name of the loaded tag onto the main screen?
I have fat fingers, and when I select a tag for writing, it's hard to see which one I selected...
This way, I can see which one was selected, prior to writing an incorrect tag! It sucks.. haha!

I'm not sure which version I'm currently running, it doesn't say anywhere, but I'm on a LG v20.
I am kind of scared to update because it works so smoothly as it is. But this would be a deal-maker. 

Thanks for your time, and all your work!

~ Ben

Bombastisch, 2017-08-14:
If you install the latest version called "Fixes" , the name of the amiibo you selected stands on the top right corner.

skizdidlyidler, 2017-08-14:
Awesome, good to know! Thanks a bunch!

closed 2017-08-14 by skizdidlyidler