HiddenRamblings/TagMo #78: No Play Store release?

r2DoesInc, 2017-08-19:
Just wondering why there hasnt been a play store release?

Bombastisch, 2017-08-19:
The app is not finished yet, so a release would have no benefits.

Secondly I don't think Nintendo would like it and so Google won't release it.

r2DoesInc, 2017-08-20:
Finished is a relative term :P 
I dont think Nintendo would be be a problem other than using Nintendo or Amiibo in the name.

Bombastisch, 2017-08-20:
But this app is all about Amiibos and how to make your own ones! I guesss Nintendo isn't a fan of this project as we are! xD

Bombastisch, 2017-08-20:
I guess we can close. I don't see any reason __yet__ to ask Google to release it on their app store. But it's the desicion of the devs. Maybe on day it will be available on Google Play. (The app I am using doesn't allow me close the issue, because of third party restrictions of Github) Please close it by yourself. 

Thank you :)

closed 2017-08-20 by HiddenRambler