HiddenRamblings/TagMo #87: Browser Folder Navigation

North101, 2017-09-04:
https://github.com/HiddenRamblings/TagMo/tree/folder_navigation

Features:
1. Default dir /tagmo
2. Remember which folder
3. Option for finding files from current folder or every subfolder

Preview:
[image]
[image]

Bombastisch, 2017-09-07:
Love it, great idea! I am mainly using the search option, but i guess with that you can filter out certain amiibos more easily!

North101, 2017-09-07:
@Bombastisch its mainly to let people select where files are found from. Its also good UX as it shows which root folder is being read from

residentevil789, 2017-09-09:
It works well though it somehow reads internal memory instead of external

North101, 2017-09-09:
@residentevil789 are you talking about my unreleased branch?
Can you explain what you mean by "though it somehow reads internal memory instead of external"

residentevil789, 2017-09-10:
Well I assume it would read the external memory card for the .bin files but instead decided to read the internal memory of my phone. I'm using the 2.5.2 version and weirdly enough i have no idea how to make it show pictures as well

North101, 2017-09-10:
@residentevil789 external memory is in the works. The pictures will be in the next release

residentevil789, 2017-09-10:
ah. Can't wait for the next release then 👍

HiddenRambler, 2017-09-11:
@North101  this is good. I usualy end up files in the download folder and have to copy them to tagmo before i can use it. This will make life easier.

I take it this isnt ready to merge yet?

North101, 2017-09-11:
@HiddenRambler its like 95% ready. Need to add keep settings when rotating

Bombastisch, 2017-09-12:
can be closed?

closed 2017-09-15 by North101