HiddenRamblings/TagMo #89: Save Icon on ImageView doesnt work

Bombastisch, 2017-09-12:
The save icon on imageview actually doesnt save the picture of the amiibo.

Bombastisch, 2017-09-12:
(version 2.6)

North101, 2017-09-12:
Well its fixed in the dev/next version. Wasn't expecting a release before the last change was merged (but also didn't know something was commented out)

HiddenRambler, 2017-09-12:
@North101 apoloies. I wasn't aware tha t merge was pending. I can do another update if you let me know when. 

perhaps we should come release process. Create a new Issue for atleast a day when a new release is needed/about to be released?

North101, 2017-09-12:
@HiddenRambler no problem, was my mistake anyway.

I'm fine with the current build now for my changes.

HiddenRambler, 2017-09-12:
I'll do a new release tonight then.

Bombastisch, 2017-09-12:
Thank you for the quick responses! ;)

closed 2017-09-12 by Bombastisch

HiddenRambler, 2017-09-12:
Release 2.6.1 done.