HiddenRamblings/TagMo #94: "Once per day" error

purpldevl, 2017-09-23:
It appears that the Tagmo app might be assigning the same serial number or code for each Amiibo that it writes to the Powertag token- trying to use different Amiibo in Breath of the Wild results in "You can only use this Amiibo once per day" notifications.

jmh363905, 2017-10-12:
The ability to randomize the serial numbers would fix this.

draconispax, 2018-01-29:
I'm having the same problem as OP, a problem I wasn't experiencing with the powersaves program

maryjodibella, 2019-02-14:
Are there any planned updates for this?  I am sad...I finally found a used android tablet that would write tags, and Tagmo runs fine, but I need to be able to randomize serial numbers to get past the 'once a day' problem.

TheForeignHunter, 2023-05-15:
I know this is an old issue but I'm having the same issue again. Tears of the Kingdom thinks it's the same amiibo and I'm getting the "once per day" message. I have power tag support turned on and I tried editing the properties and randomizing the serial number but that didn't seem to help. NFC Tools is reading the same serial number, 04:07:08:09:10:12:13 which I think is the one used to reset the powertag.