HiddenRamblings/TagMo #96: Powertag

jmh363905, 2017-10-09:
I know it's still very much early in the development but I can't find info in the wiki as suggested by release notes. If there's anything I can do to help with this add-on let me know. In the mean time what functionality is available and can the wiki be updated to reflect any info on the topic? You guys rock

Bombastisch, 2017-10-09:
https://github.com/HiddenRamblings/TagMo/wiki/PowerTags---PowerSaves-Support   this is the URL to the wiki.  There is still one bug which @HiddenRambler will fix, when he's back from vacation. ;)

jmh363905, 2017-10-09:
Well I feel stupid lol. I just clicked on the wiki tab. Didn't even see the sections to the right side. If it was a snake it would have bit me

jmh363905, 2017-10-09:
One thing that would be cool to see implemented in tagmo that you can do with the power saves app is the ability to randomize the serial number of the AMIIBO currently on the tag. I know this is a brand new feature but are any plans to implement that on the board for the future?

purpldevl, 2017-10-10:
I had made a separate topic for that a few weeks ago- HiddenRamblings tagged it as "enhancement" so I believe that means that it's being looked at.

Bombastisch, 2017-10-12:
>  randomize the serial number of the AMIIBO currently on the tag

For feature requests, always open a new thread. Otherwise it will get messy.
I also added a issue tag called "feature request" so you can mark your threads better. :)

Bombastisch, 2017-12-25:
Can be closed right?

manobon, 2018-01-29:
Just wondering, before making any potential mistakes - to Rewrite a new bin file to the PowerTag, what steps are needed?

That is, 1st bin file written to the PowerTag, with no interaction with the PowerSaves software/item 

Now, wanting to write a new bin to the PowerTag

- I couldn't tell from the instructions if the PowerTag needs to be written to via PowerSaves with the Reset file - but maybe I'm just missing something obvious

Bombastisch, 2018-01-29:
I am not sure, but I think you only have to enable powertag support in the settings and then just write a new .bin to it.

jmh363905, 2018-01-29:
I've never been able to get it to work but i just wrote it off as a work in progress. I tried it on a brand new one. I tried writing the specially crafted file to another one then trying. Nothing works for me so I've been just waiting on an update

jmh363905, 2018-01-30:
Haven't been able to since first unveiled. Says it writes but they don't work on consoles

purpldevl, 2018-01-30:
You have to write the "reset" bin to it using PowerSaves before it works on TagMo.

purpldevl, 2018-01-30:
So sorry, responded to this through email so it didn't show me the full chain of messages.

jmh363905, 2018-01-30:
I did that

Bombastisch, 2018-01-30:
Seems like this is on hold for now, atleast I don't know what to do about it, until @HiddenRambler returns from nirvana or so. ;)

HiddenRambler, 2018-01-30:
Powersaves should just work provided the tag was never used with an actual powersaves device AND powersaves support was enabled in the options the first time writing to it. 

If it was not enabled prior to first write there is a chance that the tag would be in an inconsistent state now. You can use the reset bit via a powersaves device to reset it or I can calculate the key for your current tag and include it in the app.

Please post a full dumps of the powertags using NXP TagInfo app (https://play.google.com/store/apps/details?id=com.nxp.taginfolite) and i'll put out a quick build out for you two.

Sorry about the afk for a while. Real life intervened a lot and I still don't have much free time. Also splatoon 2 happened. Will try to go through some of the comments this weekend but no promises :).

Thx for keeping this alive Basti. [Do you play splatoon 2. i saw a player called Basti some time back] :)

Bombastisch, 2018-01-30:
OMGGGGG. HE'S ALIVE!  Ehm, I have splatoon 2 but I didn't play it too much and not recently, because I am busy in school and you know how it is. If you got time and would like to play with me (you probably need to carry me) I would send you my Nintendo ID over Discord ;) looking forward to it :D. We will have to check on time zones tho xD

jmh363905, 2018-01-30:
Do you want it dumped when Reading the "Blank" tag?  Right now when I try to write it says PowerTag keys not loaded.  Others it will say transceive failed.  Mostly, I just get "Unfortunately TagMo has stopped responding".  So what exactly do you want dumped?  Me scanning a blank one?  Trying to write to one? (remember, that i've got one of those wonky NFC chipsets that the "ignore NTAG215 validation" is required for)

HiddenRambler, 2018-01-31:
@jmh363905 If you could dump the "broken" powertag using the NXP and app post the full data full information from it, I can add a way to recover it.

The "PowerTag keys not loaded" is a strange error, is the option to enable power tag support checked in he appliaction?

jmh363905, 2018-02-01:
Yes, the option is enabled (I've actually noticed that I can't even use the app with the option enabled.  Loading tags crashes, scanning tags crashes, etc...)  When I disable it, things work fine.  And do you want the Full Scan info from NXP? There's not much to it (and remember, my NXP misidentifies stuff too, so...)  I'll post what you need, just need to know where to get what you're looking for from NXP.

HiddenRambler, 2018-02-02:
in the NXP app there is an option to share the scanned data as a xml file. you can use it to email the file to yourself and then download it from your email.

I just need the first 8 bytes of the tag (first two pages) to generate the key.

manobon, 2018-02-06:
Sorry for being afk as well - glad the discussion re-started! 

I was able to write the first bin, only using Tagmo with the Powersaves option selected - worked perfectly on console, across different games, etc.

After reading the responses, just wrote to the Powersave with Tagmo/Powersaves option enabled again - worked again! On console, across games, etc...

Third time, all same options - only tried with one game, but it said the same-as-previous amiibo was being used. Tagmo scan indicated the new/a different amiibo. 

Will try with a different game - If I get the same result (i.e., game recognizes Previous amiibo), I'll post the dump using the app linked above

jmh363905, 2018-02-07:
I'll post two different dumps.  One as empty PowerTag and the other with the TagMo Reset file flashed to it. Same power tag for both scans.

First:  Empty PowerTag:

** TagInfo scan (version 4.23) 2018-02-06 18:38:53 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG203(F) (NTAG203(F))

-- NDEF ------------------------------

# No NDEF data storage populated:

-- EXTRA ------------------------------

# Memory size:
168 bytes total memory
* 42 pages, with 4 bytes per page
* 144 bytes user memory (36 pages)

# IC detailed information:
Full product name:
* NT2H0301G0DUD or NT2H0301F0DTx

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 24576 ms
MIFARE Classic support present in Android

# Detailed protocol information:
ID: 04:DF:5F:5A:F9:3D:80
ATQA: 0x4400
SAK: 0x00

XML contents: 

<?xml version="1.0" encoding="UTF-8"?>
<scan>
	<version>4.23</version>
	<date>2018-02-06 18:38:53</date>
	<title>NXP Semiconductors NTAG203(F) (NTAG203(F)) tag</title>
	<uid nxp="true">04:DF:5F:5A:F9:3D:80</uid>
	<hasndef>false</hasndef>
	<section>
		<subsection title="IC manufacturer">
			<block type="text">
	<content>NXP Semiconductors</content>
</block>
		</subsection>
		<subsection title="IC type">
			<block type="text">
	<content>NTAG203(F) (NTAG203(F))</content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="No NDEF data storage populated">
			<block type="text">
	<content></content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="Memory size">
			<block type="text">
	<content>168 bytes total memory
► 42 pages, with 4 bytes per page
► 144 bytes user memory (36 pages)</content>
</block>
		</subsection>
		<subsection title="IC detailed information">
			<block type="text">
	<content>Full product name:
► NT2H0301G0DUD or NT2H0301F0DTx</content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="Technologies supported">
			<block type="text">
	<content>ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible</content>
</block>
		</subsection>
		<subsection title="Android technology information">
			<block type="text">
	<content>Tag description:
► TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
► Maximum transceive length: 253 bytes
► Default maximum transceive time-out: 24576 ms
<hexoutput>MIFARE Classic support present in Android</hexoutput></content>
</block>
		</subsection>
		<subsection title="Detailed protocol information">
			<block type="text">
	<content>ID: 04:DF:5F:5A:F9:3D:80
ATQA: 0x4400
SAK: 0x00</content>
</block>
		</subsection>
	</section>
</scan>

jmh363905, 2018-02-07:
TagMo Reset flashed to same tag (2nd dump:

** TagInfo scan (version 4.23) 2018-02-06 18:39:53 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
NTAG203(F) (NTAG203(F))

-- NDEF ------------------------------

# No NDEF data storage populated:

-- EXTRA ------------------------------

# Memory size:
168 bytes total memory
* 42 pages, with 4 bytes per page
* 144 bytes user memory (36 pages)

# IC detailed information:
Full product name:
* NT2H0301G0DUD or NT2H0301F0DTx

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
* Default maximum transceive time-out: 24576 ms
MIFARE Classic support present in Android

# Detailed protocol information:
ID: 04:07:08:09:10:12:13
ATQA: 0x4400
SAK: 0x00

XML contents:
<?xml version="1.0" encoding="UTF-8"?>
<scan>
	<version>4.23</version>
	<date>2018-02-06 18:39:53</date>
	<title>NXP Semiconductors NTAG203(F) (NTAG203(F)) tag</title>
	<uid nxp="true">04:07:08:09:10:12:13</uid>
	<hasndef>false</hasndef>
	<section>
		<subsection title="IC manufacturer">
			<block type="text">
	<content>NXP Semiconductors</content>
</block>
		</subsection>
		<subsection title="IC type">
			<block type="text">
	<content>NTAG203(F) (NTAG203(F))</content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="No NDEF data storage populated">
			<block type="text">
	<content></content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="Memory size">
			<block type="text">
	<content>168 bytes total memory
► 42 pages, with 4 bytes per page
► 144 bytes user memory (36 pages)</content>
</block>
		</subsection>
		<subsection title="IC detailed information">
			<block type="text">
	<content>Full product name:
► NT2H0301G0DUD or NT2H0301F0DTx</content>
</block>
		</subsection>
	</section>
	<section>
		<subsection title="Technologies supported">
			<block type="text">
	<content>ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible</content>
</block>
		</subsection>
		<subsection title="Android technology information">
			<block type="text">
	<content>Tag description:
► TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
► Maximum transceive length: 253 bytes
► Default maximum transceive time-out: 24576 ms
<hexoutput>MIFARE Classic support present in Android</hexoutput></content>
</block>
		</subsection>
		<subsection title="Detailed protocol information">
			<block type="text">
	<content>ID: 04:07:08:09:10:12:13
ATQA: 0x4400
SAK: 0x00</content>
</block>
		</subsection>
	</section>
</scan>

HiddenRambler, 2018-02-07:
@manobon what game are you using? Currently tagmo will use the same unique id for a powertag which will cause some games like BOTW (Which enforce UID restrictions for a 24 hour period) to think it's the same amiibo. I will be fixing it soon.

manobon, 2018-02-07:
@hiddenrambler ah! That's exactly it - I was using it with BOTW and Odyssey, so that's what was confusing me - so, it Is working exactly as designed (so far)!

DaCrazyGamer, 2020-04-03:
Can i please confirm Tagmo is working with Powertags

syserr0r, 2020-04-03:
PowerTags work for me, but the serial number is not changed, as mentioned above

DaCrazyGamer, 2020-04-03:
Thanks. Have you had any issues with Power Tag burning out.

DaCrazyGamer, 2020-04-12:
> PowerTags work for me, but the serial number is not changed, as mentioned above

I believe there i a way to change serial number now

syserr0r, 2020-04-12:
> 
> 
> > PowerTags work for me, but the serial number is not changed, as mentioned above
> 
> I believe there i a way to change serial number now

Not from within TagMo with current release

DaCrazyGamer, 2020-04-13:
> > > PowerTags work for me, but the serial number is not changed, as mentioned above
> > 
> > 
> > I believe there i a way to change serial number now
> 
> Not from within TagMo with current release

Powersaves software's allows you to