HiddenRamblings/TagMo #100: Rethinking the MainActivity

North101, 2017-11-26:
I've never felt like the MainActivity flowed right, it felt wasted design wise. I can't think of a better way to describe it.

I think the solution is to promote the BrowserActivity to be the main activity and have what was the main activity be apart of the BrowserActivity. As shown in the rough mockup below:
[image]

@HiddenRambler @Bombastisch thoughts?

Bombastisch, 2017-11-26:
I am actually not sure about it.

We as the users will be able to very easily adapt to something like this, but for newcomers this as the front screen will be very crowded.

I'll just do a very short comparision:

Now: 

If you start the app for the first time, you see the information to load the keys and enable NFC. Then the only buttons lighting up are "load tag" and "scan tag". This is like the perfect tutorial, because it learns you step-wise what to do in chronological order to copy a tag.

-In this PR: 
The user starts of with a empty list and there a somebuttons in the corner and but nothing more to it.
Additionally many people might look up a tutorial on YouTube and so they might be confused to face a completly new start screen with no text based description.

Please don't het me wrong, I am **not**   against rethinking the app ;)
But my concern is the user friendlyness, as the simplistic an text based  mainmenu is maybe more easier.

Bombastisch, 2017-11-28:
I actually have changed my mind about it. I like it but I think of some improvements, like the text based menu

Bombastisch, 2017-11-28:
I will do some photoshop magic and show you my ideas, but I am really busy right now so I think I can send them next weekend.

Bombastisch, 2017-11-28:
[image]

I surprisingly got to finish my stuff earlier so I was able draw this thing. ;)

This is kind of the a mix of both layouts. 
I also included a button for the editing of all relevant Amiibo data, like you said in #104 
and the settings button at the top.

But the only feature missing right now is "SCAN QR" I wouldn't drop that feature in any case, so we need to rethink this round button in the bottom right corner.

Please share your thoughts over it. :)

Bombastisch, 2018-01-07:
Wrote my thoughts about it in #104 because you shared the .apk there. I am 
very happy with the changes. Wondering what @HiddenRambler  thinks about it. He's on vacation for quiet a time now^^ ;)  Hope he's having a good time.

HiddenRambler, 2018-02-07:
Sorry for not commenting on this earlier but I didn't have time to put a proper response.

I agree that the main activity is not very elegant or efficient. I wouldn't mind promoting the browser as the main UI. Only we'd need a nice way to allow scanning an existing amiibo. perhaps via toolbar button?

Bombastisch, 2018-02-07:
I am not sure if you have seen it, but in the bottom left there is this NFC button to scan. Maybe you also want to checkout #104 @HiddenRambler

HiddenRambler, 2018-02-07:
Aha! I'm just so unused to those floating buttons. That ought to do the trick.

I did indeed see #104 and I want to merge it, but didn't want to start on it till I knew I had a few hours to play with it.

North101, 2018-02-07:
@hiddenrambler there's still some work to do to it. I can't remember exactly what, should have made notes (been off sick, had some RSI issues and been swamped with work).

I think the main thing is validation. E.g. what happens when NFC is off or the decryption keys aren't loaded etc. I think I also wanted to rewrite how the tag data code

Shadowghost, 2018-03-30:
@North101 any news on this, #99, #103 and #104 ? Would be nice to have a status update. :)

AbandonedCart, 2021-04-07:
@North101 If you are interested in seeing your design updated, check out https://github.com/8BitDream/TagMo

AbandonedCart, 2021-09-11:
This has now become part of the released experimental build. Thanks once again @North101

closed 2021-09-11 by AbandonedCart